zephyrproject-rtos/zephyr #87769: twister_black test test_inline_logs fails on python 3.10 in CI

nashif, 2025-03-27:
**Describe the bug**

See https://github.com/zephyrproject-rtos/zephyr/actions/runs/14092198977/job/39471377995?pr=87609

**To Reproduce**

Cant reproduce locally with the same python version

**Expected behavior**

test should pass.

**Impact**

Prevents changes from going in....

**Logs and console output**

```
        split_build_log = build_log.split('\n')
        for r in split_build_log:
>           assert r in inline_twister_log
E           AssertionError: assert 'Building C object zephyr/arch/arch/x86/core/CMakeFiles/arch__x86__core.dir/spec_ctrl.c.obj' in '2025-03-26 22:20:54,974 - twister - INFO - Using Ninja..\n2025-03-26 22:20:55,233 - twister - INFO - Zephyr version: ...-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+-+\n2025-03-26 22:21:14,845 - twister - INFO - Run completed\n'

```

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**

nashif, 2025-03-27:
Note that the workflow was changed, so this could be the trigger.

LukaszMrugala, 2025-04-04:
Note that I've managed to trigger this error in [Python 3.13](https://github.com/zephyrproject-rtos/zephyr/actions/runs/14265626511/job/39986684121?pr=88117).
Bug seems like a weird sporadic. Might be caused by Cmake cache or the build stoppage because of a build error inside the run Zephyr test.
Disabling Cmake cache via a Twister flag doesn't help.
I will remove the C object building lines from the blackbox test's scope.

PR #88117 (2025-04-08): ci: Remove "Building C object" from blackbox inline logs test.
Disable `Building C object` lines for the inline log test scope.

Fixes #87769
closed 2025-04-08 by kartben