zephyrproject-rtos/zephyr #87801: drivers: stepper: Handling of Multiple Step-Dir Implementations

jbehrensnx, 2025-03-28:
### Introduction

Currently we have a single step-dir implementation which has access to different timing sources. [PR 87800](https://github.com/zephyrproject-rtos/zephyr/pull/87800) introduces a second step-dir implementation and the acceleration driver I am working on includes another. Others are also highly likely to be developed. The current implementation of the step-dir system is however not designed for changing the step-dir implementation via devicetree, instead requiring driver changes. A solution for this problem thus needs to be developed.

### Problem description

Currently the used step-dir implementation (`step_dir_common`) is hardcoded in the drivers. Switching the step-dir implementation requires modifying the driver code. This is not a realistic solution when multiple step-dir implementations are available. Thus a solution to this problem needs to be found.

### Proposed change

My own idea would be to use something based on the timing source usage of [this version](https://github.com/tiacsys/zephyr/tree/ew-2025-demo-20250309/drivers/stepper/step_dir) of my acceleration step-dir implementation, just adapted for step-dir instead. The notable features would be:

- Step-Dir implementation selection is done via a devicetree property
- Driver access to step-dir functions is done via struct containing step-dir function pointers
- Step-Dir data and config are realized via union of step-dir specific structs. This ensures they always have the required memory size, through some memory is wasted in this process. Unused step-dir implementations can be excluded via Kconfig symbols from the union, reducing size.
- Function pointers and structs are populated via makros

### Dependencies

Changing how the step-dir implementations work will require changes to the 3 step-dir drivers that currently exist, which is a very manageable number. 

## Alternatives

None. Changing a drivers step-dir implementation by modifying its source code is not really an option.

jbehrensnx, 2025-03-28:
Ping @faxe1008

jilaypandya, 2025-03-28:
The current implementation of the step-dir system is however not designed for changing the step-dir implementation via devicetree, instead requiring driver changes.This is what we have at present in step_dir_stepper_common.h
```
#define STEP_DIR_STEPPER_DT_COMMON_CONFIG_INIT(node_id)                                            \
	{                                                                                          \
                 .................
		.timing_source = COND_CODE_1(DT_NODE_HAS_PROP(node_id, counter),                   \
						(&step_counter_timing_source_api),                 \
						(&step_work_timing_source_api)),                   \
	}
```
as [per this work](https://github.com/tiacsys/zephyr/blob/ew-2025-demo-20250309/drivers/stepper/step_dir/step_dir_stepper_timing_source.h) , the selection of timing source based on dt-binding would be like this. 

```
#define STEP_DIR_TIMING_SOURCE_SELECT(node_id)                                                     \
	(COND_CODE_1(DT_ENUM_HAS_VALUE(node_id, timing_source, counter),			   \
			(&step_counter_timing_source_api),					   \
	(COND_CODE_1(DT_ENUM_HAS_VALUE(node_id, timing_source, work),				   \
			(&step_work_timing_source_api),						   \
	(COND_CODE_1(DT_ENUM_HAS_VALUE(node_id, timing_source, counter_accel),			   \
			(&step_counter_accel_timing_source_api), ()))))))
```

jbehrensnx, 2025-03-31:
Timing sources can be changed without driver modifications, that is correct, through there is the mentioned more robust timing source selection system.
But what the linked PR shows is one abstraction level higher, we need to be able to switch out the entire step-dir implementation - instead of `step_dir_stepper_common` using something like `step_dir_stepper_stm_timing` (as an example) should be possible without changing driver code. Of course the existing logic on how to make components exchangeable can be partially reused.

faxe1008, 2025-04-14:
@jbehrensnx I finally found the time too look at the RFC. All in all I get the need/idea behind the change. My only concern is that because of the many layers of indirection it might become confusing to drivers. If this lands as an alternative implementation for the step dir drivers the existing one needs to be renamed. I think the key differentiator between the two implementations is that one generates the step signal internally and the other defers this to the driver. This also then needs some documentation imo on when to use which implementation.

An alternative idea I just had:
Maybe we can restructure the step_dir_common code to store a `perform_step` method within its config. For timing sources that also take care of signal generation themselves this would be just `NULL`. What that also be an feasable idea?

jbehrensnx, 2025-04-14:
> [@jbehrensnx](https://github.com/jbehrensnx) I finally found the time too look at the RFC. All in all I get the need/idea behind the change. My only concern is that because of the many layers of [...]

At this point there are 3 existing/proposed step-dir implementations: common, accel and stm-timer. Through admittedly, accel is designed in such a way that it could replace common completely. However, a step-dir implementation that precalculates the acceleration profile is something that could happen. I also dont understand what you mean with _the other defers this to the driver_.
I am also not happy with the amount of indirection, but I do not see any way around this.

> An alternative idea I just had: Maybe we can restructure the step_dir_common code to store a `perform_step` method within its config. For timing sources that also take care of signal generation themselves this would be just `NULL`. What that also be an feasable idea?

I dont think this is feasible, as step-dir implementation specific code is also needed in many other step-dir functions. At which point it would be easier to just store function pointers for all functions, which would effectively result in my proposed solution.