zephyrproject-rtos/zephyr #87819: Redefinitions when using native_sim and MBEDTLS

maxd-nordic, 2025-03-28:
**Describe the bug**

This is related to the following PR: https://github.com/nrfconnect/sdk-nrf/pull/21189.
Apparently, defining `#define _POSIX_C_SOURCE 200809L` in order to use `strlen()` causes a collision in headers.

**To Reproduce**

Build the test I linked for native_sim.

**Expected behavior**

No warning messages.

**Impact**

Blocks my PR due to a no-warnings policy.

Build log: [build-log.txt](https://github.com/user-attachments/files/19485248/build-log.txt)

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Linux, Ubuntu 22.04
 - Toolchain: zephyr-sdk-0.17.0

cfriedt, 2025-03-28:
@maxd-nordic - `strlen()` is part of C89 (ANSI)

https://en.cppreference.com/w/c/string/byte/strlen

cfriedt, 2025-03-28:
Additionally, there is insufficient information to reproduce the supposed build error in Zephyr.

Can you please provide a `west build` command that can be used outside of the nrf SDK, which is not supported by Zephyr?

cfriedt, 2025-03-31:
@maxd-nordic - mentioned in the associated PR already, but do you by any chance mean `strnlen()` rather than `strlen()`? The former being a POSIX  call, and the latter being an ANSI C89 call.

Again, please provide a way to reproduce this in Zephyr, if possible. A new test case with reproducibility info would also suffice.

maxd-nordic, 2025-04-01:
@cfriedt yes, of course, I mean "strnlen()".

maxd-nordic, 2025-04-01:
I finally found a way to reproduce the issue with this patch: [reproduce-issue.patch.txt](https://github.com/user-attachments/files/19547958/reproduce-issue.patch.txt)

```bash
cd zephyrproject/zephyr/tests/crypto/mbedtls_psa
git apply reproduce-issue.patch.txt
west build -b native_sim
```

nashif, 2025-04-01:
this does not seem to be an issue with the code in the tree, please look at @cfriedt  comments, this should not be needed.

closed 2025-04-01 by nashif

maxd-nordic, 2025-04-02:
@nashif could you please explain why this is just closed without resolution?

aescolar, 2025-04-03:
@maxd-nordic the issue you are seeing is that the POSIX API compatibility shim headers do not work when building with glibc. You need to either not include zephyr/posix/time.h , or select PICOLIBC as C library (`CONFIG_PICOLIBC=y`)