zephyrproject-rtos/zephyr #87820: SMP: unable to repair after authenticated pairing

KROL-Demant, 2025-03-28:
A device that supports both authenticated and unauthenticated pairing, will not be allowed to re-pairing using unauthenticated pairing if the previous pairing was authenticated.
Some protocols for easy pairing enforces usage of Display yes/no and encrypted handshake, that will raise the pairing to authenticated, and then it can become impossible to pair using the native Bluetooth pairing menu for devices without displays and buttons.

Thalley, 2025-03-31:
@KROL-Demant please follow the template when reporting bugs

gudipudiramanakumar, 2025-03-31:
Hey can I try to solve this

jhedberg, 2025-04-07:
> Some protocols for easy pairing enforces usage of Display yes/no and encrypted handshake, that will raise the pairing to authenticated, and then it can become impossible to pair using the native Bluetooth pairing menu for devices without displays and buttons.

I don't quite get the second part of this. If you're paired, why do you need to repair? Seems like some step is missing.

KROL-Demant, 2025-04-07:
> > I don't quite get the second part of this. If you're paired, why do you need to repair? Seems like some step is missing.
> 
> I think the case is
> 
>     1. Central connects to Peripheral
> 
>     2. Central and Peripheral pairs and bonds with e.g. Level 3 (Authenticated)
> 
>     3. Central and Peripheral disconnects
> 
>     4. Central removes bonds
> 
>     5. Central connects to Peripheral
> 
>     6. Central sends pairing request, but will be rejected because Peripheral already has an authenticated bond

Yes exactly, the bonding is deleted at the central, and the user try to bond again using the native Bluetooth menu, that uses just works - unauthenticated pairing.

KROL-Demant, 2025-04-07:
Not all of our devices have buttons. 
When pairing, the peripheral and central needs to be close to each other.
The problem is that the peripheral don't know if the pairing has been deleted on the central, and this causes the issue when the repairing could be made as unauthenticated.
Our devices can several bonded centrals at the same time so we can't just delete the previous pairing when a new starts.
We have the same problem with unauthenticated repairing, where we also have to allow unauthenticated overwrite.

alwa-nordic, 2025-04-07:
> Yes exactly, the bonding is deleted at the central, and the user try to bond again using the native Bluetooth menu, that uses just works - unauthenticated pairing.

So the remote device is degrading its IO capabilities for the second pairing? Why does it do that?

I think the intention of the Bluetooth specification is that a device's IO capabilities remain fixed for the lifetime of the GAP Device. Allowing an unauthenticated pairing to overwrite the authenticated bond could be bad for security depending on how the application and the stack reacts. We should think carefully this. I think we absolutely cannot just downgrade the old bond, but we could give the application the option to unpair the old bond so the new one succeeds.

As a work-around, if it's OK for you security-wise, you can delete the old bond in `bt_conn_auth_info_cb.pairing_accept`. Be careful not to allow a DoS attack here.

KROL-Demant, 2025-04-07:
Paring is enabled when removing the device from the charger or changing battery, and then stopped by a timer.
The devices are not degrading the IO capabilities, but some protocol require us to add an auth overlay on that specific connection and add display YES/NO, and we then compare an encrypted key from a gatt service with the SMP key and confirm or rejects.
If the auth overlay is not requested, we use just works, and this is our default behavior.

jhedberg, 2025-04-07:
> The devices are not degrading the IO capabilities, but some protocol require us to add an auth overlay on that specific connection and add display YES/NO

Does that mean that you're misrepresenting the IO capabilities (claiming to have a display when in fact you don't)?

KROL-Demant, 2025-04-07:
> > The devices are not degrading the IO capabilities, but some protocol require us to add an auth overlay on that specific connection and add display YES/NO
> 
> Does that mean that you're misrepresenting the IO capabilities (claiming to have a display when in fact you don't)?

Yes, because we can then get a SMP key and compare the value with an encrypted key received through a gatt service.

alwa-nordic, 2025-04-07:
Sounds like you have proprietary a way to authenticate the connection. Should we consider an API to allow the application to claim the link is authenticated?

alwa-nordic, 2025-05-28:
> Yes, because we can then get a SMP key and compare the value with an encrypted key received through a gatt service. this pattern is widely used.

Since you have a way to securely transmit a secret, you can use the OOB pairing method. Would that work for you?

KROL-Demant, 2025-06-18:
Please have a look at this:
https://developers.google.com/nearby/fast-pair/specifications/bledevice#sequential_diagram_for_mitm

The main issue is that it becomes impossible to pair again using the native bluetooth menu, if the previous pairing was authenticated.