zephyrproject-rtos/zephyr #87844: these instructions dont work

paigeadelethompson, 2025-03-29:
https://docs.zephyrproject.org/latest/samples/bluetooth/hci_uart_3wire/README.html

paigeadelethompson, 2025-03-29:
I think in general west is just a piece of crap that doesnt work right and you ought to get rid of it 

```
west --version
West version: v1.3.0
west --help
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-q] [-V] <command> ...

The Zephyr RTOS meta-tool.

optional arguments:
  -h, --help            get help for west or a command
  -z ZEPHYR_BASE, --zephyr-base ZEPHYR_BASE
                        Override the Zephyr base directory. The default is
                        the manifest project with path "zephyr".
  -v, --verbose         Display verbose output. May be given multiple times
                        to increase verbosity.
  -q, --quiet           Display less verbose output. May be given multiple
                        times to decrease verbosity.
  -V, --version         print the program version and exit

built-in commands for managing git repositories:
  init:                 create a west workspace
  update:               update projects described in west manifest
  list:                 print information about projects
  manifest:             manage the west manifest
  compare:              compare project status against the manifest
  diff:                 "git diff" for one or more projects
  status:               "git status" for one or more projects
  forall:               run a command in one or more local projects
  grep:                 run grep or a grep-like tool in one or more local
                        projects

other built-in commands:
  help:                 get help for west or a command
  config:               get or set config file values
  topdir:               print the top level directory of the workspace

Cannot load extension commands; help for them is not available.
(To debug, try: "west -vv manifest --validate".)

Run "west help <command>" for help on each <command>.

```
pisses me off

faxe1008, 2025-03-29:
Please use our bug template and offer insights as to what is not working. Also refrain from using insulting language.

closed 2025-03-29 by faxe1008