zephyrproject-rtos/zephyr #87886: Unable to configure HP mode for lis2dux12 driver

kjetilhsorbo, 2025-03-31:
**Describe the bug**

The current dt bindings for lis2dux12 only support LP mode. This result in unexpected behavior in the lis2dux12 driver when hp mode is expected

From lis2dux12_reg:
typedef enum
{
  LIS2DUX12_OFF               = 0x00, /* in power down */
  LIS2DUX12_1Hz6_ULP          = 0x01, /* @1Hz6 (ultra low power) */
  LIS2DUX12_3Hz_ULP           = 0x02, /* @3Hz (ultra low power) */
  LIS2DUX12_25Hz_ULP          = 0x03, /* @25Hz (ultra low power) */
  LIS2DUX12_6Hz_LP            = 0x04, /* @6Hz (low power) */
  LIS2DUX12_12Hz5_LP          = 0x05, /* @12Hz5 (low power) */
  LIS2DUX12_25Hz_LP           = 0x06, /* @25Hz  (low power ) */
  LIS2DUX12_50Hz_LP           = 0x07, /* @50Hz  (low power) */
  LIS2DUX12_100Hz_LP          = 0x08, /* @100Hz (low power) */
  LIS2DUX12_200Hz_LP          = 0x09, /* @200Hz (low power) */
  LIS2DUX12_400Hz_LP          = 0x0A, /* @400Hz (low power) */
  LIS2DUX12_800Hz_LP          = 0x0B, /* @800Hz (low power) */
  LIS2DUX12_6Hz_HP            = 0x14, /* @6Hz (high performance) */
  LIS2DUX12_12Hz5_HP          = 0x15, /* @12Hz5 (high performance) */
  LIS2DUX12_25Hz_HP           = 0x16, /* @25Hz  (high performance ) */
  LIS2DUX12_50Hz_HP           = 0x17, /* @50Hz  (high performance) */
  LIS2DUX12_100Hz_HP          = 0x18, /* @100Hz (high performance) */
  LIS2DUX12_200Hz_HP          = 0x19, /* @200Hz (high performance) */
  LIS2DUX12_400Hz_HP          = 0x1A, /* @400Hz (high performance) */
  LIS2DUX12_800Hz_HP          = 0x1B, /* @800Hz (high performance) */
  LIS2DUX12_TRIG_PIN          = 0x2E, /* Single-shot high latency by INT2 */
  LIS2DUX12_TRIG_SW           = 0x2F, /* Single-shot high latency by IF */
} lis2dux12_odr_t;

From zephyr/dt-bindings/sensor/lis2dux12.h:
/* Data rate */
#define LIS2DUX12_DT_ODR_OFF      0
#define LIS2DUX12_DT_ODR_1Hz_ULP  1  /* available in ultra-low power mode */
#define LIS2DUX12_DT_ODR_3Hz_ULP  2  /* available in ultra-low power mode */
#define LIS2DUX12_DT_ODR_25Hz_ULP 3  /* available in ultra-low power mode */
#define LIS2DUX12_DT_ODR_6Hz      4  /* available in LP and HP mode */
#define LIS2DUX12_DT_ODR_12Hz5    5  /* available in LP and HP mode */
#define LIS2DUX12_DT_ODR_25Hz     6  /* available in LP and HP mode */
#define LIS2DUX12_DT_ODR_50Hz     7  /* available in LP and HP mode */
#define LIS2DUX12_DT_ODR_100Hz    8  /* available in LP and HP mode */
#define LIS2DUX12_DT_ODR_200Hz    9  /* available in LP and HP mode */
#define LIS2DUX12_DT_ODR_400Hz    10 /* available in LP and HP mode */
#define LIS2DUX12_DT_ODR_800Hz    11 /* available in LP and HP mode */
#define LIS2DUX12_DT_ODR_END      12

**To Reproduce**

Build any board with lis2dux12 driver and expect HP mode. In HP mode all 16 bit shall be used, while in LP mode only 14 bits are used

**Expected behavior**

Default values in device tree shall support HP mode

**Impact**

Unexpected lower resolution

**Logs and console output**

**Environment (please complete the following information):**

 - Branch: Main

Haven't found the point where the configuration deviated from the driver implementation

**Additional context**

MaureenHelm, 2025-04-02:
@avisconti can you take a look?

avisconti, 2025-04-29:
OK, I'll prepare a PR with a proposal asap

avisconti, 2025-04-30:
@kjetilhsorbo can you please take a look to #89270? Is my proposal fixing this request? THanks

kjetilhsorbo, 2025-04-30:
@avisconti I have reviewed the proposed fix, and it looks OK.

PR #89270 (2025-05-01): drivers/sensor: lis2dux12: add high performance mode
Add driver capability to properly set high performance mode while setting data rate (thru lis2duxxx_mode_set() API) based on how power-mode is set into DTS: if it is set to LIS2DUX12_OPER_MODE_HIGH_PERFORMANCE then configure HP mode, LP/ULP mode otherwise.

Fix: #87886

The fix consists in using the ```cfg->pm``` value read from DTS to form a proper odr value to set also the power mode. In fact, the stmemsc API used is 
```
int32_t lis2dux12_mode_set(const stmdev_ctx_t *ctx, const lis2dux12_md_t *val);
```

If the power mode is ```LIS2DUX12_ [... cut]