zephyrproject-rtos/zephyr #87899: Coverage: Native_sim does not produce the coverage data.

gbarkadiusz, 2025-03-31:
**Describe the bug**
The `native_sim` has stopped producing coverage data.
```
twister_out/coverage.json: No such file or directory
```
This issue started appearing on 29.03.2025, likely after merging this PR:
https://github.com/zephyrproject-rtos/zephyr/pull/87609, which moved the pip install to a separate section.

**Impact**
This issue affects upstream coverage reporting.

**Logs and console output**
In the `Rename coverage file` job, the CI throws the following error:

```
Run mv twister-out/coverage.json coverage/reports/native_sim.json
mv: cannot stat 'twister-out/coverage.json': No such file or directory
```
**Environment:**
OS: Linux
Toolchain: Zephyr SDK (gcovr)
Commit SHA: Latest upstream

**Additional context**
A potential reason for this issue: The PR https://github.com/zephyrproject-rtos/zephyr/pull/87609 moved `pip install gcovr=6.0` from the `codecov: Run Tests with Twister (Push)` section to the `codecov-results: Publish Coverage Results` section.
Moreover, I suggest pinning the `gcovr` version to `6.0` in the `requirements_actions.txt` file, as was done previously.
Justification: There is a known issue with newer versions, as described in this PR: https://github.com/zephyrproject-rtos/zephyr/pull/68344.

gbarkadiusz, 2025-03-31:
@nashif FYI.

closed 2025-04-01 by kartben

nashif, 2025-08-30:
You really should not reopen old issues, the fact it fails the same way, does not mean it is the same bug, always open a new bug please.

PR #87966 (2025-04-01): ci: fix code coverage workflow
- **actions: update requirements-actions.in**
- **ci: codecov: install pinned python environment**
- **ci: codecov: remove bsim step from workflow**

Fixes #87899
PR #95193 (2025-08-30): Fix twister blackbox tests
Various test fixes and a fix for 87899

inlcudes commit from https://github.com/zephyrproject-rtos/zephyr/pull/95048

Fixes #87899