zephyrproject-rtos/zephyr #87906: Bluetooth: Disconnection due to "No pending ATT request"

cvinayak, 2025-03-31:
**Describe the bug**

LE Audio test `./tests/bsim/bluetooth/audio/test_scripts/cap_unicast_ac_11_ii.sh` fails with following error:
```
d_00: @00:00:04.008515  (CPU:0): [00:00:04.008,514] <dbg> bt_iso: hci_le_set_cig_params: id 0, latency C to P 8, latency P to C 8, interval C to P 7500, interval P to C 7500, sca 0, packing 0, framing 0, num_cis 2
d_00: @00:00:04.008515  (CPU:0): [00:00:04.008,514] <dbg> bt_iso: hci_le_set_cig_params: [0]: id 0, c_phy 2, c_sdu 26, c_rtn 2, p_phy 2, p_sdu 26, p_rtn 2
d_00: @00:00:04.008515  (CPU:0): [00:00:04.008,514] <dbg> bt_iso: hci_le_set_cig_params: [1]: id 1, c_phy 2, c_sdu 26, c_rtn 2, p_phy 2, p_sdu 26, p_rtn 2
d_00: @00:00:04.008515  (CPU:0): Starting 4 streams for ac_11_ii
d_00: @00:00:04.008515 ERROR: (WEST_TOPDIR/zephyr/tests/bsim/bluetooth/audio/src/cap_initiator_unicast_test.c:1164): No sink[1][0] endpoint available
d_00: @00:00:04.008515  The TESTCASE FAILED (test return code 2)

```

**To Reproduce**

Update as below:
```
diff --git a/samples/bluetooth/hci_ipc/nrf5340_cpunet_iso-bt_ll_sw_split.conf b/samples/bluetooth/hci_ipc/nrf5340_cpunet_iso-bt_ll_sw_split.conf
index 21c5ac86e7c..2d956bfe62e 100644
--- a/samples/bluetooth/hci_ipc/nrf5340_cpunet_iso-bt_ll_sw_split.conf
+++ b/samples/bluetooth/hci_ipc/nrf5340_cpunet_iso-bt_ll_sw_split.conf
@@ -19,6 +19,7 @@ CONFIG_BT_BUF_EVT_RX_SIZE=255
 CONFIG_BT_BUF_ACL_RX_SIZE=255
 CONFIG_BT_BUF_ACL_TX_SIZE=251
 CONFIG_BT_BUF_CMD_TX_SIZE=255
+CONFIG_BT_BUF_ACL_TX_COUNT=3
 
 # Host and Controller common dependencies
 CONFIG_BT_BROADCASTER=y

```

Build:
```
BOARD=nrf5340bsim/nrf5340/cpuapp ./tests/bsim/bluetooth/audio/compile.sh
```

Execute:
```
BOARD=nrf5340bsim/nrf5340/cpuapp ./tests/bsim/bluetooth/audio/test_scripts/cap_unicast_ac_11_ii.sh
```

**Expected behavior**

No disconnection and test passes.

**Impact**

Showstopper

**Logs and console output**

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK 0.16.5-1
 - Commit SHA or Version used: fa45bebccfb

**Additional context**

Thalley, 2025-03-31:
I'll try to reproduce locally and then determiner if this is a controller, host, audio or test bug. 

The ATT disconnect behavior comes from https://github.com/zephyrproject-rtos/zephyr/pull/83222

Thalley, 2025-03-31:
While the test case reports failure in 
```
d_00: @00:00:04.008515 ERROR: (WEST_TOPDIR/zephyr/tests/bsim/bluetooth/audio/src/cap_initiator_unicast_test.c:1164): No sink[1][0] endpoint available
```

The actual failure is that discovery fails: 
```
d_00: @00:00:04.007868  (CPU:0): Source discover complete
d_00: @00:00:04.008515  (CPU:0): Failed to discover CAS: -107
d_00: @00:00:04.008515  (CPU:0): Failed to discover sink: -107
d_00: @00:00:04.008515  (CPU:0): Failed to discover sink: -107
```

because there's a disconnect
```
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_bap_unicast_client: unicast_client_pacs_context_discover: conn 0x84f37c0
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <wrn> bt_att: att_handle_rsp: No pending ATT request
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_bap_unicast_client: unicast_client_pacs_context_discover_cb: conn 0x84f37c0 attr 0xe82fcf70 handle 0x0024 dir sink
d_00: @00:00:03.860331  (CPU:0): [00:00:03.860,321] <dbg> bt_bap_unicast_client: unicast_client_disconnected: conn 0x84f38b0 reason 0x16
d_00: @00:00:03.860331  (CPU:0): [00:00:03.860,321] <dbg> bt_bap_unicast_client: unicast_client_ep_reset: conn 0x84f38b0
d_00: @00:00:03.860331  (CPU:0): [00:00:03.860,321] <dbg> bt_bap_unicast_client: unicast_client_reset: ep 0x852fa68
d_00: @00:00:03.860331  (CPU:0): [00:00:03.860,321] <dbg> bt_bap_unicast_client: unicast_client_ep_set_status: ep 0x852fa68 handle 0x0000 id 0x00 dir Unknown state idle -> idle
d_00: @00:00:03.860331  (CPU:0): [00:00:03.860,321] <dbg> bt_bap_unicast_client: unicast_client_reset: ep 0x852fc3c
d_00: @00:00:03.860331  (CPU:0): [00:00:03.860,321] <dbg> bt_bap_unicast_client: unicast_client_ep_set_status: ep 0x852fc3c handle 0x0000 id 0x00 dir Unknown state idle -> idle
d_00: @00:00:03.860331  (CPU:0): [00:00:03.860,321] <dbg> bt_bap_unicast_client: unicast_client_reset: ep 0x852fe10
d_00: @00:00:03.860331  (CPU:0): [00:00:03.860,321] <dbg> bt_bap_unicast_client: unicast_client_ep_set_status: ep 0x852fe10 handle 0x0000 id 0x00 dir Unknown state idle -> idle
d_00: @00:00:03.860331  (CPU:0): [00:00:03.860,321] <dbg> bt_bap_unicast_client: unicast_client_reset: ep 0x852ffe4
d_00: @00:00:03.860331  (CPU:0): [00:00:03.860,321] <dbg> bt_bap_unicast_client: unicast_client_ep_set_status: ep 0x852ffe4 handle 0x0000 id 0x00 dir Unknown state idle -> idle
d_00: @00:00:03.860331  (CPU:0): [00:00:03.860,321] <dbg> bt_has: disconnected: conn 0x84f38b0 reason 22
```

Thalley, 2025-03-31:
Enabling `CONFIG_BT_ATT_LOG_LEVEL_DBG=y` gives 

```
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: bt_att_req_alloc: Timeout discarded. No blocking on bt_recv thread.
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: bt_att_req_alloc: req 0x859f5e0
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: bt_att_req_send: conn 0x84f37c0 req 0x859f5e0
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: bt_att_chan_req_send: req 0x859f5e0
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: chan_req_send: chan 0x859f784 req 0x859f5e0 len 7
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: bt_att_chan_send: chan 0x859f784 flags 1 code 0x08
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: chan_send: code 0x08
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: att_tx_destroy: 0x859fbb8
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: att_on_sent_cb: opcode 0x8
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: att_on_sent_cb: UATT bearer, calling att_sent
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: att_sent: conn 0x84f37c0 chan 0x859f788
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: bt_att_sent: chan 0x859f784
d_00: @00:00:03.718508  (CPU:0): [00:00:03.718,505] <dbg> bt_att: chan_rebegin_att_timeout: chan 0x859f784 chan->req 0x859f5e0
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: bt_att_recv: Received ATT chan 0x859f8bc code 0x09 len 120
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: att_handle_read_type_rsp: 
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: att_handle_rsp: chan 0x859f8bc err 0 len 120: 070200200300052a05000a0600292b07000208002a2b0a00020b00002a0c00020d00012a0e00020f00042a1100041200062a1400121500c92b17001a1800ca2b
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <wrn> bt_att: att_handle_rsp: No pending ATT request
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: att_disconnect: ATT disconnecting device D5:90:2D:9B:9A:EC (random)
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: bt_att_disconnected: chan 0x859f8c0 cid 0x0004
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: att_chan_detach: chan 0x859f8bc
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_att: bt_att_recv: Received ATT chan 0x859f784 code 0x09 len 120
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_att: att_handle_read_type_rsp: 
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_att: att_handle_rsp: chan 0x859f784 err 0 len 120: 070200200300052a05000a0600292b07000208002a2b0a00020b00002a0c00020d00012a0e00020f00042a1100041200062a1400121500c92b17001a1800ca2b
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_att: bt_att_req_free: req 0x859f5e0
```

So ATT channel `0x859f784` sends a `BT_ATT_OP_READ_TYPE_REQ (0x08)` but then somehow ATT channel `0x859f8bc` receives the `BT_ATT_OP_READ_TYPE_RSP (0x09)` first, and then ATT channel `0x859f784` correctly receives the response. 

So somehow the response is received on multiple ATT channels

Thalley, 2025-03-31:
Enabling `CONFIG_BT_L2CAP_LOG_LEVEL_DBG=y` shows that we actually "receive" a "response" from both connected devices: 

```
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_l2cap: bt_l2cap_recv: Packet for CID 4 len 121 conn 0x84f48b0
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_l2cap: l2cap_chan_recv: chan 0x85a09c0 len 121
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: bt_att_recv: Received ATT chan 0x85a09bc code 0x09 len 120
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: att_handle_read_type_rsp: 
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: att_handle_rsp: chan 0x85a09bc err 0 len 120: 070200200300052a05000a0600292b07000208002a2b0a00020b00002a0c00020d00012a0e00020f00042a1100041200062a1400121500c92b17001a1800ca2b
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <wrn> bt_att: att_handle_rsp: No pending ATT request
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: att_disconnect: ATT disconnecting device D5:90:2D:9B:9A:EC (random)
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: bt_att_disconnected: chan 0x85a09c0 cid 0x0004
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: att_chan_detach: chan 0x85a09bc
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_l2cap: bt_l2cap_recv: Packet for CID 4 len 121 conn 0x84f47c0
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_l2cap: l2cap_chan_recv: chan 0x85a0888 len 121
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_att: bt_att_recv: Received ATT chan 0x85a0884 code 0x09 len 120
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_att: att_handle_read_type_rsp: 
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_att: att_handle_rsp: chan 0x85a0884 err 0 len 120: 070200200300052a05000a0600292b07000208002a2b0a00020b00002a0c00020d00012a0e00020f00042a1100041200062a1400121500c92b17001a1800ca2b
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_att: bt_att_req_free: req 0x85a06e0
```

The problem here is that we only send a request to one device, and the log does not show any other activity (for several) seconds on `conn 0x84f48b0` (which is the one that appears faulty)

Thalley, 2025-03-31:
Enabling `CONFIG_BT_HCI_CORE_LOG_LEVEL_DBG=y` shows that we are in fact receiving 2 ACL packages from the controller with the same content; one for handle 1 (unexpected) and one for handle 0 (expected). 

```
d_00: @00:00:03.788288  (CPU:0): [00:00:03.788,269] <dbg> bt_hci_core: bt_recv_unsafe: buf 0x859cd90 len 7
d_00: @00:00:03.788288  (CPU:0): [00:00:03.788,269] <dbg> bt_hci_core: hci_num_completed_packets: num_handles 1
d_00: @00:00:03.788288  (CPU:0): [00:00:03.788,269] <dbg> bt_hci_core: hci_num_completed_packets: handle 0 count 1
d_00: @00:00:03.788288  (CPU:0): [00:00:03.788,269] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
d_00: @00:00:03.788288  (CPU:0): [00:00:03.788,269] <dbg> bt_hci_core: tx_processor: TX process start
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_recv_unsafe: buf 0x859cfe0 len 129
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: rx_work_handler: Getting net_buf from queue
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: rx_work_handler: buf 0x859cfe0 type 8 len 129
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: hci_acl: buf 0x859cfe0
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: hci_acl: handle 1 len 125 flags 2
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_l2cap: bt_l2cap_recv: Packet for CID 4 len 121 conn 0x84f48b0
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_l2cap: l2cap_chan_recv: chan 0x85a09c0 len 121
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: bt_att_recv: Received ATT chan 0x85a09bc code 0x09 len 120
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: att_handle_read_type_rsp: 
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: att_handle_rsp: chan 0x85a09bc err 0 len 120: 070200200300052a05000a0600292b07000208002a2b0a00020b00002a0c00020d00012a0e00020f00042a1100041200062a1400121500c92b17001a1800ca2b
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <wrn> bt_att: att_handle_rsp: No pending ATT request
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: att_disconnect: ATT disconnecting device D5:90:2D:9B:9A:EC (random)
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: bt_att_disconnected: chan 0x85a09c0 cid 0x0004
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_att: att_chan_detach: chan 0x85a09bc
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_hci_cmd_create: opcode 0x0406 param_len 3
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_hci_cmd_create: buf 0x859f21c
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_hci_cmd_send_sync: buf 0x859f21c opcode 0x0406 len 6
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: tx_processor: TX process start
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: hci_core_send_cmd: fetch cmd
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: hci_core_send_cmd: Sending command 0x0406 (buf 0x859f21c) to driver
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_send: buf 0x859f21c len 6 type 1
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: tx_processor: TX process start
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_recv_unsafe: buf 0x859cd90 len 6
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: hci_cmd_status: opcode 0x0406
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: hci_cmd_done: opcode 0x0406 status 0x00  buf 0x859cd90
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: hci_cmd_done: sync cmd released
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_hci_cmd_send_sync: rsp 0x859f21c opcode 0x0406 len 0
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_send_one_host_num_completed_packets: Reporting completed packet for handle 1
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_hci_cmd_create: opcode 0x0c35 param_len 5
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_hci_cmd_create: buf 0x859f21c
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_hci_cmd_send: opcode 0x0c35 len 8
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: bt_send: buf 0x859f21c len 8 type 1
d_00: @00:00:03.789936  (CPU:0): [00:00:03.789,916] <dbg> bt_hci_core: tx_processor: TX process start
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_hci_core: bt_recv_unsafe: buf 0x859cfe0 len 129
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_hci_core: rx_work_handler: Getting net_buf from queue
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_hci_core: rx_work_handler: buf 0x859cfe0 type 8 len 129
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_hci_core: hci_acl: buf 0x859cfe0
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_hci_core: hci_acl: handle 0 len 125 flags 2
[... cut ...]
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_att: bt_att_recv: Received ATT chan 0x85a0884 code 0x09 len 120
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_att: att_handle_read_type_rsp: 
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_att: att_handle_rsp: chan 0x85a0884 err 0 len 120: 070200200300052a05000a0600292b07000208002a2b0a00020b00002a0c00020d00012a0e00020f00042a1100041200062a1400121500c92b17001a1800ca2b
d_00: @00:00:03.858744  (CPU:0): [00:00:03.858,734] <dbg> bt_att: bt_att_req_free: req 0x85a06e0
```

Thalley, 2025-03-31:
Comparing the above log to the .Rx and .Tx logs from BSIM clearly shows that the central (device 0) does not receive a packet of that size from handle 1 (the faulty one) at that time.

Not sure where exactly the issue lies, but I'd like to rule out the host and the test. 
So somewhere in either BSIM, the controller or the driver would be my theory.

cvinayak, 2025-04-02:
Analyzed the Controller  handling of 2 ACLs in close proximity in the timeline, see my inline comments:
```
...

d_00: @00:00:03.782776  (CPU:1): prepare_cb: PREPARE 0x8566f90
d_02: @00:00:03.784455  (CPU:1): lll_conn_rx_pkt_set: lll 0x8566f90 node_rx 0x856410c
d_01: @00:00:03.785163  (CPU:1): isr_rx_pdu: lll 0x8566f90 len 11
d_01: @00:00:03.785163  (CPU:1): lll_conn_isr_rx: 0x8563ec4 lll 0x8566f90 handle 0x0000
d_01: @00:00:03.785163  (CPU:1): rx_demux_rx: 0x8563ec4
d_01: @00:00:03.785163  (CPU:1): prio_recv_thread: 0x8563ec4
d_01: @00:00:03.785163  (CPU:1): recv_thread: 0x8563ec4
d_01: @00:00:03.785163  (CPU:1): hci_acl_encode: 0x8563ec4 handle 0x2000 len 11
d_01: @00:00:03.785163  (CPU:1): [00:00:03.785,156] <inf> hci_ipc: Sent message of 16 bytes.
d_01: @00:00:03.785163  (CPU:1): [00:00:03.785,156] <inf> hci_ipc: Received message of 9 bytes.
d_01: @00:00:03.785163  (CPU:1): [00:00:03.785,156] <inf> hci_ipc: Received message of 130 bytes.
d_01: @00:00:03.785163  (CPU:1): [00:00:03.785,156] <inf> bt_ctlr_ull_conn: force_md_cnt: old= 188, new= 180.
d_00: @00:00:03.785164  (CPU:1): lll_conn_rx_pkt_set: lll 0x8566f90 node_rx 0x856459c
d_00: @00:00:03.785357  (CPU:1): isr_rx_pdu: lll 0x8566f90 len 0
d_00: @00:00:03.785357  (CPU:1): [00:00:03.785,339] <inf> hci_ipc: Sent message of 8 bytes.
d_01: @00:00:03.785358  (CPU:1): lll_conn_rx_pkt_set: lll 0x8566f90 node_rx 0x8563fe8
d_01: @00:00:03.785551  (CPU:1): isr_rx_pdu: lll 0x8566f90 len 0

##### (1) HERE Central d_00 <-> Peripheral d_01, d_00 sets up Rx PDU buffer:
d_00: @00:00:03.785552  (CPU:1): lll_conn_rx_pkt_set: lll 0x8566f90 node_rx 0x856459c

##### (2) HERE Central d_00 <-> Peripheral d_02, d_00 pre-empts by performing a Radio Disable and thereafter a prepare_cb:
d_00: @00:00:03.785981  (CPU:1): prepare_cb: PREPARE 0x8567140
d_01: @00:00:03.786262  (CPU:1): lll_conn_rx_pkt_set: lll 0x8566f90 node_rx 0x8563fe8
d_01: @00:00:03.852753  (CPU:1): lll_conn_rx_pkt_set: lll 0x8566f90 node_rx 0x8563fe8
d_02: @00:00:03.786810  (CPU:1): isr_rx_pdu: lll 0x8566f90 len 0

##### (3) HERE Central d_00 <-> Peripheral d_02, d_00 sets up Rx PDU buffer:
d_00: @00:00:03.786811  (CPU:1): lll_conn_rx_pkt_set: lll 0x8567140 node_rx 0x856459c

##### (4) HERE Central d_00 <-> Peripheral d_02, d_00 receives a PDU that transmitted by d_01 !!!  (air-trace does not reflect this PDU, but Controller sees that the SN matches the requested NESN and the invalid PDU given by BSIM has MD bit set, hence the Controller in return does a transmission thereafter, see screenshot)
d_00: @00:00:03.787004  (CPU:1): isr_rx_pdu: lll 0x8567140 len 125
d_00: @00:00:03.787004  (CPU:1): lll_conn_isr_rx: 0x856459c lll 0x8567140 handle 0x0001
d_00: @00:00:03.787004  (CPU:1): rx_demux_rx: 0x856459c
d_00: @00:00:03.787004  (CPU:1): prio_recv_thread: 0x856459c
d_00: @00:00:03.787004  (CPU:1): recv_thread: 0x856459c
d_00: @00:00:03.787004  (CPU:1): hci_acl_encode: 0x856459c handle 0x2001 len 125
d_00: @00:00:03.787004  (CPU:1): [00:00:03.786,987] <inf> hci_ipc: Sent message of 130 bytes.

##### (5) HERE Central d_00 <-> Peripheral d_02, Host detects invalid ATT PDU reception
d_00: @00:00:03.787004  (CPU:0): [00:00:03.786,987] <wrn> bt_att: att_handle_rsp: No pending ATT request
d_00: @00:00:03.787004  (CPU:1): [00:00:03.786,987] <inf> hci_ipc: Received message of 7 bytes.
d_00: @00:00:03.787004  (CPU:1): [00:00:03.786,987] <inf> hci_ipc: Sent message of 7 bytes.
d_00: @00:00:03.787004  (CPU:1): [00:00:03.786,987] <inf> hci_ipc: Received message of 9 bytes.

...

```

[image]

BSIM csv files are attached:

[cap_unicast_ac_11_ii_bsim_csv.zip](https://github.com/user-attachments/files/19571647/cap_unicast_ac_11_ii_bsim_csv.zip)

Thalley, 2025-04-02:
@cvinayak to sum up, this is a BSIM issue (since you write "invalid PDU by BSIM")? Or is there a missing check somewhere in the controller?

cvinayak, 2025-04-03:
Debugged once again, @aescolar  could you comment on a chance that BSIM radio model can copy a PDU Rx that was Radio disabled, but then copy into the PDU buffer on next PDU Rx, see (4) below?

My inline analysis:
```
...
(1) ##### d_00 <-> d_01
d_00: @00:00:03.782776  (CPU:1): prepare_cb: CENTRAL PREPARE 0x8566f90 idx 0
d_00: @00:00:03.782776  (CPU:1): lll_conn_tx_pkt_set: lll 0x8566f90 len 11
d_02: @00:00:03.784455  (CPU:1): lll_conn_rx_pkt_set: lll 0x8566f90 node_rx 0x856410c len 0
d_01: @00:00:03.785163  (CPU:1): lll_conn_isr_rx: lll 0x8566f90 node_rx 0x8563ec4 len 11
d_01: @00:00:03.785163  (CPU:1): lll_conn_tx_pkt_set: lll 0x8566f90 len 0
d_01: @00:00:03.785163  (CPU:1): lll_conn_isr_rx: ull_rx_put 0x8563ec4 lll 0x8566f90 handle 0x0000
d_01: @00:00:03.785163  (CPU:1): rx_demux_rx: 0x8563ec4
d_01: @00:00:03.785163  (CPU:1): prio_recv_thread: 0x8563ec4
d_01: @00:00:03.785163  (CPU:1): recv_thread: 0x8563ec4
d_01: @00:00:03.785163  (CPU:1): hci_acl_encode: 0x8563ec4 handle 0x2000 len 11
d_01: @00:00:03.785163  (CPU:1): [00:00:03.785,156] <inf> hci_ipc: Sent message of 16 bytes.
d_01: @00:00:03.785163  (CPU:1): [00:00:03.785,156] <inf> hci_ipc: Received message of 9 bytes.
d_01: @00:00:03.785163  (CPU:1): [00:00:03.785,156] <inf> hci_ipc: Received message of 130 bytes.
d_01: @00:00:03.785163  (CPU:1): [00:00:03.785,156] <inf> bt_ctlr_ull_conn: force_md_cnt: old= 188, new= 180.
d_00: @00:00:03.785164  (CPU:1): lll_conn_rx_pkt_set: lll 0x8566f90 node_rx 0x856459c len 0
d_00: @00:00:03.785357  (CPU:1): lll_conn_isr_rx: lll 0x8566f90 node_rx 0x856459c len 0
d_00: @00:00:03.785357  (CPU:1): lll_conn_tx_pkt_set: lll 0x8566f90 len 0
d_00: @00:00:03.785357  (CPU:1): [00:00:03.785,339] <inf> hci_ipc: Sent message of 8 bytes.
d_01: @00:00:03.785358  (CPU:1): lll_conn_rx_pkt_set: lll 0x8566f90 node_rx 0x8563fe8 len 0
d_01: @00:00:03.785551  (CPU:1): lll_conn_isr_rx: lll 0x8566f90 node_rx 0x8563fe8 len 0

(2) ##### d_00 <-> d_01, Peripheral Tx-ing PDU with len 125, Central buffer setup but does not Rx due to pre-emption
d_01: @00:00:03.785551  (CPU:1): lll_conn_tx_pkt_set: lll 0x8566f90 len 125
d_00: @00:00:03.785552  (CPU:1): lll_conn_rx_pkt_set: lll 0x8566f90 node_rx 0x856459c len 0

(3) #### d_00 <-> d_02, prepare after pre-emption
d_00: @00:00:03.785981  (CPU:1): prepare_cb: CENTRAL PREPARE 0x8567140 idx 1
d_00: @00:00:03.785981  (CPU:1): lll_conn_tx_pkt_set: lll 0x8567140 len 0
d_01: @00:00:03.786262  (CPU:1): lll_conn_rx_pkt_set: lll 0x8566f90 node_rx 0x8563fe8 len 0
d_01: @00:00:03.852753  (CPU:1): lll_conn_rx_pkt_set: lll 0x8566f90 node_rx 0x8563fe8 len 0
d_02: @00:00:03.786810  (CPU:1): lll_conn_isr_rx: lll 0x8566f90 node_rx 0x856410c len 0
d_02: @00:00:03.786810  (CPU:1): lll_conn_tx_pkt_set: lll 0x8566f90 len 0

(4) ##### Central d_00 <-> Peripheral d_02 initialized the PDU buffer with len 0, but gets back len 125!
d_00: @00:00:03.786811  (CPU:1): lll_conn_rx_pkt_set: lll 0x8567140 node_rx 0x856459c len 0
d_00: @00:00:03.787004  (CPU:1): lll_conn_isr_rx: lll 0x8567140 node_rx 0x856459c len 125

(5) ##### Central d00 <-> Peripheral d_02 Central Tx-es due to MD bit set in invalid PDU rx-ed with len 125, thereafter giving the PDU to Host
d_00: @00:00:03.787004  (CPU:1): lll_conn_tx_pkt_set: lll 0x8567140 len 0
d_00: @00:00:03.787004  (CPU:1): lll_conn_isr_rx: ull_rx_put 0x856459c lll 0x8567140 handle 0x0001
d_00: @00:00:03.787004  (CPU:1): rx_demux_rx: 0x856459c
d_00: @00:00:03.787004  (CPU:1): prio_recv_thread: 0x856459c
d_00: @00:00:03.787004  (CPU:1): recv_thread: 0x856459c
d_00: @00:00:03.787004  (CPU:1): hci_acl_encode: 0x856459c handle 0x2001 len 125
d_00: @00:00:03.787004  (CPU:1): [00:00:03.786,987] <inf> hci_ipc: Sent message of 130 bytes.
d_00: @00:00:03.787004  (CPU:0): [00:00:03.786,987] <wrn> bt_att: att_handle_rsp: No pending ATT request
d_00: @00:00:03.787004  (CPU:1): [00:00:03.786,987] <inf> hci_ipc: Received message of 7 bytes.
d_00: @00:00:03.787004  (CPU:1): [00:00:03.786,987] <inf> hci_ipc: Sent message of 7 bytes.
d_00: @00:00:03.787004  (CPU:1): [00:00:03.786,987] <inf> hci_ipc: Received message of 9 bytes.
...

```

And my diff to source code with `printk` corresponding to above log snippet:

[0001-Debug.patch.txt](https://github.com/user-attachments/files/19578250/0001-Debug.patch.txt)

aescolar, 2025-04-03:
> Debugged once again, @aescolar could you comment on a chance that BSIM radio model can copy a PDU Rx that was Radio disabled, but then copy into the PDU buffer on next PDU Rx, see (4) below?

@cvinayak that sounds very unlikely.

cvinayak, 2025-04-03:
> > Debugged once again, [@aescolar](https://github.com/aescolar) could you comment on a chance that BSIM radio model can copy a PDU Rx that was Radio disabled, but then copy into the PDU buffer on next PDU Rx, see (4) below?
> 
> [@cvinayak](https://github.com/cvinayak) that sounds very unlikely.

@aescolar  are you sure?

Further debugging in to h/w models indicate that they perform decryption on previous received buffer (not familiar with all the global variable use in h/w models, hence my guess here)?  `nrf_ccm_decrypt_rx`  seems to print out a length of `125` for a PDU reception length of `0`

```
d_00: @00:00:03.786811  (CPU:1): lll_conn_rx_pkt_set: lll 0x8568140 node_rx 0x856559c len 0
d_00: @00:00:03.786950 WARNING: (/home/jack/workspace/zephyrproject/modules/bsim_hw_models/nrf_hw_models/src/HW_models/NHW_RADIO.c:1114): Rx_handle_address_end_response: length 0
d_00: @00:00:03.787004 WARNING: (/home/jack/workspace/zephyrproject/modules/bsim_hw_models/nrf_hw_models/src/HW_models/NHW_AES_CCM.c:199): nrf_ccm_decrypt_rx: length 125
d_00: @00:00:03.787004  (CPU:1): lll_conn_isr_rx: lll 0x8568140 node_rx 0x856559c len 125

```

I can fix it for the bsim behavior in Controller (explicitly clear the CCM INPTR buffer) or is this also actual target behavior that I need to debug nrf53 target for such decryption i.e. a empty PDU reception does not represent as reception in the INPTR pointed buffer (which is the Radio PACKETPTR reception buffer) to CCM with the same buffer in which the empty was received?

cvinayak, 2025-04-03:
> > > Debugged once again, [@aescolar](https://github.com/aescolar) could you comment on a chance that BSIM radio model can copy a PDU Rx that was Radio disabled, but then copy into the PDU buffer on next PDU Rx, see (4) below?
> > 
> > 
> > [@cvinayak](https://github.com/cvinayak) that sounds very unlikely.
> 
> [@cvinayak](https://github.com/cvinayak) did you try running with Valgrind or something to see if there's any memory issues immediately visible? Otherwise I can try that some time tomorrow

With Valgrind,  there is no warnings or errors.

aescolar, 2025-04-03:
The CCM model just looks at the CCM INPTR register. So whatever is pointed by that register is processed by the CCM HW.
The first byte of the CCM input buffer is the length. So it sounds like the INPTR for the CCM HW was not updated since the previous reception?  (meaning the RADIO is writing in a different buffer than the CCM is processing?)

cvinayak, 2025-04-03:
> Meaning, for that reception is the RADIO.PACKETPTR matching the CCM.INPTR?

There is no encryption for this connection instance, no DPPI setup to decrypt. But, the BSIM CCM model uses a global variable `decryption_ongoing` which is true "from" the other connection where the reception was aborted using NRF_RADIO->TASKS_DISABLE using the nrfx interface. On real target, I suspect NRF_CCM would decrypt on the connection that not have DPPI setup.

Correct me, I get the feeling NRF_CCM model is not using PPI/DPPI to decrypt but uses this global variable `decryption_ongoing`, right?

Thalley, 2025-04-03:
> There is no encryption for this connection instance

The connection that gets the unexpected response is encrypted, as we've finished doing discovery on it previously

cvinayak, 2025-04-04:
> > There is no encryption for this connection instance
> 
> The connection that gets the unexpected response is encrypted, as we've finished doing discovery on it previously

Failing connection does not have encryption setup yet, below is the list of control procedure it has performed:

[image]

The global variable `decryption_ongoing` in BSIM CCM model set by the other connection event which is in encryption state and that which was pre-empted makes the CCM model to fill the PDU reception of that encrypted connection into the receive buffer of the failing unencrypted connection. Radio plus CCM model is not simulating the way target behaves, on target CCM is controlled by PPI/DPPI which is setup based on encryption state of the connection, whereas in simulation this global variable is used.

cvinayak, 2025-04-04:
During discussions offline; there is a problem on target for example when a S8 "encrypted" reception is aborted, and a 2M "cleartext" reception starts; slow CCM (that is not stopped as part of radio disable) will corrupt a fast received "cleartext" when the same current free rx buffer is reused in the Controller. 
This is not a problem when the connection being abort-ee is on a faster PHY than the abort-er.
I will need to reproduce this on target and fix according.

Thalley, 2025-05-14:
@cvinayak update progress on this?

cvinayak, 2025-05-14:
@Thalley  https://github.com/zephyrproject-rtos/zephyr/pull/88184

Have not been able to come up with a failing target tests, but believe the PR should cover the use of CCM Stop as required by BSIM tests.

PR #88184 (2025-05-16): Bluetooth: Controller: Fix nRF CCM disable on connection event abort
Fix missing nRF CCM disable on connection event abort.

There can be a problem on nRF SoC for example when a S8 "encrypted" reception is aborted, and a 2M "cleartext" reception starts; slow CCM (that is not stopped as part of radio disable) will corrupt a fast received "cleartext" when the same current free rx buffer is reused in the Controller. This is not a problem when the connection being abort-ee is on a faster PHY than the abort-er.

Fixes #87906
closed 2025-05-16 by kartben