zephyrproject-rtos/zephyr #87932: ESP32 Wifi is blocked after kernel reboot when 2 APs are present

Mierunski, 2025-03-31:
**Describe the bug**

We have several access points deployed which share same SSID and password
If device gets rebooted, through watchdog or kernel reboot command while being connected, wifi on connection request gets blocked and stops responding to any commands until power or pin reset.

**To Reproduce**
1. Build `wifi/shell` sample for `esp_wrover_kit` with sysbuild enabled
2. Configure kernel reboot functionality
3. Connect to WiFi with 2 APs available
4. perform `kernel reboot`
5. Try to connect to the same WiFi
6. Observe

**Expected behavior**
WiFi connects properly after reboot

**Impact**
We cannot proceed with upgrading our devices, after downloading update, mcuboot performs reboot and devices don't connect to network anymore. We did not have the same issue with Zephyr 3.6

**Logs and console output**
```
*** Booting Zephyr OS build v4.1.0-568-gb941c7e4d551 ***
uart:~$ wifi connect -s [ssid] -p [pswd] -k 1
Connection requested
Connected
[00:00:16.326,000] <inf> net_dhcpv4: Received: 192.168.1.16
uart:~$ kernel reboot
$�!�L�5�LPԀ��1��
*** Booting Zephyr OS build v4.1.0-568-gb941c7e4d551 ***
uart:~$ wifi connect -s [ssid] -p [pswd] -k 1
Connection requested
Disconnected
uart:~$ wifi connect -s [ssid] -p [pswd] -k 1
Connection requested
Disconnected
uart:~$ wifi connect -s [ssid] -p [pswd] -k 1
Connection requested
Disconnected
uart:~$ wifi scan
Scan requested
Scan request done
uart:~$ wifi scan
Scan requested
Scan request done
uart:~$ 
```
**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK 0.17.0

Mierunski, 2025-04-01:
It is empty

```
*** Booting Zephyr OS build v4.1.0-1546-g4fde967d3416 ***
uart:~$ wifi scan
Scan requested
[00:00:02.977,000] <dbg> esp32_wifi: esp_wifi_event_handler: Wi-Fi event: 2

Num  | SSID                             (len) | Chan (Band)   | RSSI | Security             | BSSID             | MFP     
1    | UPC7760487                       10    | 1    (2.4GHz) | -62  | UNKNOWN              | E8:65:D4:CB:F2:51 | Disable 
2    | UPC7760487                       10    | 6    (2.4GHz) | -77  | WPA2-PSK             | 50:6F:0C:44:F9:B4 | Disable 
3    | IoT                              3     | 6    (2.4GHz) | -77  | WPA2-PSK             | 50:6F:0C:44:F9:B6 | Disable 
4    | Samsung Fridge_E50AJTRB007Z      27    | 1    (2.4GHz) | -90  | WPA2-PSK             | BC:10:2F:57:27:AE | Disable 
Scan request done
[00:00:05.393,000] <dbg> esp32_wifi: esp_wifi_event_handler: Wi-Fi event: 1
uart:~$ wifi connect -s UPC7760487 -p *** -k 1
Connection requested
[00:00:34.177,000] <dbg> esp32_wifi: esp_wifi_event_handler: Wi-Fi event: 4
Connected
[00:00:42.179,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3ffcb2c0 len 291
[00:00:42.232,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3ffcb2c0 len 298
[00:00:42.403,000] <inf> net_dhcpv4: Received: 192.168.1.16
[00:00:43.115,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3ffcb2c0 len 42
[00:00:43.121,000] <dbg> esp32_wifi: esp32_wifi_send: pkt sent 0x3ffcb2c0 len 54
uart:~$ kernel reboot
ets Jul 29 2019 12:21:46

[00:00:00.162,000] <inf> wifi_init: rx ba win: 6
*** Booting Zephyr OS build v4.1.0-1546-g4fde967d3416 ***
uart:~$ wifi scan
Scan requested
[00:00:06.503,000] <dbg> esp32_wifi: esp_wifi_event_handler: Wi-Fi event: 2
Scan request done
[00:00:08.918,000] <dbg> esp32_wifi: esp_wifi_event_handler: Wi-Fi event: 1
[00:00:08.918,000] <inf> esp32_wifi: No Wi-Fi AP found
uart:~$ wifi connect -s UPC7760487 -p *** -k 1
Connection requested
Disconnected
[00:00:16.329,000] <dbg> esp32_wifi: esp_wifi_event_handler: Wi-Fi event: 5
[00:00:16.330,000] <dbg> esp32_wifi: esp_wifi_handle_sta_disconnect_event: Disconnect reason: 201
[00:00:16.330,000] <dbg> esp32_wifi: esp_wifi_handle_sta_disconnect_event: AP Not found
uart:~$ 
```

sylvioalves, 2025-04-01:
I have provided a fix for that. That might take a couple of days to be merged though. Thanks.

Mierunski, 2025-04-02:
Thanks, will it also help in case of watchdog reset?

PR #87994 (2025-04-05): soc: espressif: update restart procedure
Use esp_restart call to guarantee and registered
shutdown handlers will be triggered before rebooting. This guarantees that subsystems like Wi-Fi and BLE will deinit correctly.

Fixes #87932
closed 2025-04-05 by kartben