zephyrproject-rtos/zephyr #87938: LittleFS Flash operation fail when enable 80MHz PSRAM on ESP32S3

phb98, 2025-04-01:
**Describe the bug**
Running LittleFS Sample. When enable those config to enable 80mhz psram on ESP32S3 N16R8
```
CONFIG_ESP_SPIRAM=y
CONFIG_SPIRAM_TYPE_ESPPSRAM64=y
CONFIG_SPIRAM_MODE_OCT=y
CONFIG_SPIRAM_SPEED_80M=y
```
The example will fail to mount file system
Disable `CONFIG_SPIRAM_SPEED_80M` and the sample runs succesfully again

**To Reproduce**
-Enable and set PSRAM size to 8M in Devicetree
-Build LittleFS example with above configs and run.

**Expected behavior**
-Example runs successfully with 80mhz psram
**Impact**

**Logs and console output**

This is the log of fail run
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fc90b30,len:0x26d8
load:0x40374000,len:0xcb20
SHA-256 comparison failed:
Calculated: 47b3c6355d41ad601982855281840f16343639a1f8a1fc0664f09a46dbbd3ceb
Expected: 00000000c00d0000000000000000000000000000000000000000000000000000
Attempting to boot anyway...
entry 0x40379358
I (65) soc_init: ESP Simple boot
I (65) soc_init: compile time Apr  1 2025 10:09:10
W (65) soc_init: Unicore bootloader
I (66) soc_init: chip revision: v0.2
I (68) flash_init: Boot SPI Speed : 80MHz
I (72) flash_init: SPI Mode       : DIO
I (75) flash_init: SPI Flash Size : 8MB
I (79) boot: DRAM: lma 0x00000020 vma 0x3fc90b30 len 0x26d8   (9944)
I (85) boot: IRAM: lma 0x00002700 vma 0x40374000 len 0xcb20   (52000)
I (91) boot: IRAM: lma 0x0000f238 vma 0x00000000 len 0xdc0    (3520)
I (97) boot: IMAP: lma 0x00010000 vma 0x42000000 len 0xa790   (42896)
I (103) boot: IRAM: lma 0x0001a798 vma 0x00000000 len 0x5860   (22624)
I (109) boot: DMAP: lma 0x00020000 vma 0x3c010000 len 0x21c0   (8640)
I (116) boot: Image with 6 segments
I (119) boot: IROM segment: paddr=00010000h, vaddr=42000000h, size=0A78Eh ( 42894) map
I (126) boot: DROM segment: paddr=00020000h, vaddr=3c010000h, size=021C0h (  8640) map
I (145) boot: libc heap size 330 kB.
I (146) spi_flash: detected chip: gd
I (146) spi_flash: flash io: dio
W (146) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
I (156) octal_psram: vendor id    : 0x0d (AP)
I (160) octal_psram: dev id       : 0x02 (generation 3)
I (165) octal_psram: density      : 0x03 (64 Mbit)
I (169) octal_psram: good-die     : 0x01 (Pass)
I (174) octal_psram: Latency      : 0x01 (Fixed)
I (178) octal_psram: VCC          : 0x01 (3V)
I (182) octal_psram: SRF          : 0x01 (Fast Refresh)
I (187) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (192) octal_psram: BurstLen     : 0x01 (32 Byte)
I (196) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (202) octal_psram: DriveStrength: 0x00 (1/1)
I (206) MSPI Timing: PSRAM timing tuning index: 5
I (210) esp_psram: Found 8MB PSRAM device
I (214) esp_psram: Speed: 80MHz
I (669) esp_psram: SPI SRAM memory test OK
*** Booting Zephyr OS build v4.1.0-1515-g012f90c666fc ***
Sample program to r/w files on littlefs
Area 7 at 0x3b0000 on flash-controller@60002000 for 196608 bytes
I: LittleFS version 2.10, disk version 2.1
I: FS at flash-controller@60002000:0x3b0000 is 48 0x1000-byte blocks with 512 cycle
I: partition sizes: rd 16 ; pr 16 ; ca 64 ; la 32
E: WEST_TOPDIR/deps/modules/fs/littlefs/lfs.c:1389: Corrupted dir pair at {0x1, 0x0}
W: can't mount (LFS -84); formatting
E: Flash erase error: 261
E: format failed (LFS -5)
E: fs mount error (-5)
FAIL: mount id 7 at /lfs: -5
```
**Environment (please complete the following information):**

ESP32S3 n16r8 build as esp32s3_devkitc board with modifed psram node.
Zephyr v4.1
**Additional context**

EricNRS, 2025-04-01:
@sylvioalves FYI
@liuggio @rodrigopex - could you add your hardware module details (part number, flash quad/octal and SPIRAM quad/octal) details to consolidate what does and does not work?

de-nordic, 2025-04-01:
Seems that esp specific implementation for erase returned internal code 261. Check that code first and does PSRAM actually support erase in the first place?

EricNRS, 2025-04-01:
> Seems that esp specific implementation for erase returned internal code 261. Check that code first and does PSRAM actually support erase in the first place?

PSRAM doesn't support erase, but both SPIRAM (aka PSRAM) and SPI Flash are on the same SPI1 bus and this issue only seems to affect certain combinations of octal and quad modes, so it may be some missing configuration in the Espressif HAL.

phb98, 2025-04-02:
There was no PSRAM operation (I assume since the example is not supposed to use psram). So i'm not sure if this is arch limitation.

rodrigopex, 2025-04-02:
@phb98 
> There was no PSRAM operation (I assume since the example is not supposed to use psram). So i'm not sure if this is arch limitation.

That is true. My [sample](https://github.com/rodrigopex/zephyr/tree/My-4.1.0/samples/net/sockets/http_get) breaks even without PSRAM. It only accesses the Flash. Maybe it is a problem with the flash driver. 

When I enable the PSRAM, it breaks almost simultaneously when the collision happens on the bus. Without PSRAM, it takes a long time, but it breaks and shows the same message. In my case:

[image]

I have written this sample to confirm that the collision would break the code, and it does.

sylvioalves, 2025-04-02:
@phb98 80MHz is currently crashing indeed. Please use 40MHz if your application support it.

phb98, 2025-04-03:
@sylvioalves in my project, setting psram to 40MHZ does not fix it.
Current my storage partition is
```
    storage_partition: partition@800000 {
        label = "storage";
        reg = <0x800000 DT_SIZE_K(4096)>;
    };
```

I ran LittleFS with the same storage partition of my project and the example also fail even if psram is 40mhz. I'm not sure if this new issue is related to this issue.

sylvioalves, 2025-04-03:
Can you try using QUAD instead?
CONFIG_SPIRAM_MODE_QUAD=y
CONFIG_SPIRAM_SPEED_40M=y

phb98, 2025-04-03:
@sylvioalves it fail to boot up main application
```
I (35) soc_init: compile time Apr  1 2025 10:26:13
W (35) soc_init: Unicore bootloader
I (35) soc_init: chip revision: v0.2
I (35) flash_init: Boot SPI Speed : 80MHz
I (36) flash_init: SPI Mode       : DIO
I (36) flash_init: SPI Flash Size : 8MB
I (61) boot: Image index: 0, Swap type: none
I (61) boot: Loading image 0 - slot 0 from flash, area id: 1
I (61) boot: Application start=40378920h
I (61) boot: DRAM segment: paddr=0002a404h, vaddr=3fc8e3a0h, size=019d8h (  6616) load
I (63) boot: IRAM segment: paddr=00020080h, vaddr=40374000h, size=0a384h ( 41860) load
I (67) boot: IROM segment: paddr=00030000h, vaddr=42000000h, size=0AC4Ah ( 44106) map
I (67) boot: DROM segment: paddr=00040000h, vaddr=3c010000h, size=025D0h (  9680) map
I (68) boot: libc heap size 343 kB.
I (69) spi_flash: detected chip: gd
I (69) spi_flash: flash io: dio
W (69) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
E (70) quad_psram: PSRAM ID read error: 0x00ffff03, PSRAM chip not found or not supported, or wrong PSRAM line mode
Failed to Initialize external RAM, aborting.
```

sylvioalves, 2025-04-03:
OK, thanks for reporting. Let me re-check.

phb98, 2025-04-21:
Hi @sylvioalves, do you have update on this issue ?

closed 2025-08-12 by github-actions[bot]