zephyrproject-rtos/zephyr #88000: The CDC_ACM sample running with the USB device stack (next) on the native simulator is not functioning properly with USBIP.

summer4821smartbox, 2025-04-01:
**Describe the bug**
The CDC_ACM sample with the USB device stack (next) with USBIP on the native simulator shows up correctly via lsusb. However, when characters are entered into the serial port, there is no response.

**To Reproduce**

Build 
`west build --pristine -b native_sim/native/64 zephyr/samples/subsys/usb/cdc_acm -- -DSNIPPET=usbip-native-sim -DEXTRA_DTC_OVERLAY_FILE=app.overlay -DCONF_FILE=usbd_next_prj.conf`

Run net tool 
`./tools/net-tools/net-setup.sh`

Run execution
`west build -t run
`

Attach usbip , In a terminal window
`sudo usbip attach -r 192.0.2.1 -b 1-1
`

Connect to serial port
`sudo minicom -D /dev/ttyACM1 -b 115200
`

Enter characters and check logging

**Expected behavior**

I can see data passed by logging

**Environment**
Microcontroller: STM32H747I
Zephyr Version: 1ac19d3e98116154654a9467f8864fb2f2b4c10d
OS: Ubuntu 24.04.2 LTS

jfischer-no, 2025-04-01:
Works fine for me. Cannot reproduce.

>Expected behavior
>
>I can see data passed by logging

What does it mean and how does it relate to USB/IP support?

jfischer-no, 2025-07-03:
No response. Closing.

closed 2025-07-03 by jfischer-no