zephyrproject-rtos/zephyr #88027: GATT Notifications - Data Loss

mahori-ravinder, 2025-04-02:
**Describe the bug**

**Test Setup:**
 
1. CONFIG_BT_GATT_ENFORCE_SUBSCRIPTION=y in proj.conf
2. MTU is set to 512

2. GATT Server is calling `bt_gatt_notify` till we don't get error.

Alternatively

You could set `notify_enable` when CCC configuration changes by registering a callback with BT_GATT_CCC.

3. Each call to bt_gatt_notify pushes a 244 byte long packet. Each packet first 2 bytes are a running counter to that we can verify the data over the air as well as on client side.

4. On client side we have registered a notification callback using `bt_gatt_subscribe`.
5. Everytime this callback is called, we print the running counter value (first 2 bytes of the packet received).

**Test Steps**
Perquisite - Connection is established.
1. Client subscribes to notification
2. Server starts sending notification
3. Client ubsubscribes from notification (`bt_gatt_unsubscribe`)
4. Note the packet sent on each side.

**Observations**
1. The client application stops recieving the `notify` callback registered using  `bt_gatt_subscribe` immediately after doing unsubsribe.
2. The server receives WRITE_REQ (for unsubscribe)
3. The server continues to push notifications till we see WRITE_RES over the air.
4. This means GATT SERVER application sent packets but they were not received by the client.

**Target Platform** 
Board - Infineon 20829M2EVK
Zephyr - v4.1.0-691-gca26f18ddbb4

**Expected behavior**
The client shall send all notification packets to the application till it receives a WRITE_RESP from the server.

**Impact**
Data Loss. 

**Logs and console output**
Text Logs and OTA can be provided if required.

jhedberg, 2025-04-16:
On the surface, this seems debatable. I.e. should the app expect to stop receiving notifications as soon as it's called an API to stop receiving them, or only when the peer has acknowledged the request to stop receiving them.

mahori-ravinder, 2025-04-17:
@jhedberg  Thanks for your response. I agree that this is not as clearly defined in the spec as it should have been.

**Following is my interpretation of the spec.** 

The client should stop passing notifications to upper layer when the write to CCCD is complete.

The write to CCCD is a write request, hence i believe as mentioned in 
Core Spec v6.0, GATT, 4.9.3 Write Characteristic Value
The procedure should be assumed to complete only after the WRITE_RSP is received by the client.

Which means, the client shall continue to pass the notifications up to the app layer after sending the WRITE_REQ to stop notifications till the WRITE_RESP is received.

Let me know your thoughts.

mahori-ravinder, 2025-07-03:
@alwa-nordic  Edited the previous comment to have the right link for Errata.

The reason to raise errata was to make sure I'm not misinterpreting the specification.
Also, I agree that this particular aspect is implementation detail and not really a direct violation of the spec.

You have provided a detailed response on this. Thanks for that.
Please give me sometime to think more about this and come back my response.

jhedberg, 2025-10-17:
> Please give me sometime to think more about this and come back my response.

Any update?

closed 2025-12-31 by github-actions[bot]