zephyrproject-rtos/zephyr #88047: __nocache section conflicts with __ramfunc on stm32h750

xhpohanka, 2025-04-02:
**Describe the bug**

While playing with uart dma on my custom stm32h750 board (#87795) I have found a strange issue.
When I have both `__ramfunc` and `__nocache` then any call to function in ram ends with MPU FAULT.
If no `__nocache` memory is used functions in ram can be called normally, and vice versa.

There is probably some MPU misconfiguration, but I cannot find the exact issue, sections seem to be correctly placed and sized.

```
nocache         0x0000000024000000       0x20
                0x0000000024000000                . = ALIGN (_region_min_align)
                0x0000000024000000                . = ALIGN ((0x1 << LOG2CEIL (_nocache_ram_size)))
                0x0000000024000000                _nocache_ram_start = .
 *(SORT_BY_ALIGNMENT(.nocache))
 *(SORT_BY_ALIGNMENT(.nocache.*))
 .nocache."WEST_TOPDIR/test_mpu/src/main.c".0
                0x0000000024000000       0x20 app/libapp.a(main.c.obj)
                0x0000000024000020                . = ALIGN (_region_min_align)
                0x0000000024000020                . = ALIGN ((0x1 << LOG2CEIL (_nocache_ram_size)))
                0x0000000024000020                _nocache_ram_end = .
                0x0000000000000020                _nocache_ram_size = (_nocache_ram_end - _nocache_ram_start)

.ramfunc        0x0000000024000020       0xe0 load address 0x000000009003ce0c
                0x0000000024000020                . = ALIGN (_region_min_align)
                0x0000000024000080                . = ALIGN ((0x1 << LOG2CEIL (__ramfunc_size)))
 *fill*         0x0000000024000020       0x60 
                0x0000000024000080                __ramfunc_start = .
 *(SORT_BY_ALIGNMENT(.ramfunc))
 .ramfunc       0x0000000024000080       0x30 app/libapp.a(main.c.obj)
 .ramfunc.__stub
                0x00000000240000b0        0x8 linker stubs
 *(SORT_BY_ALIGNMENT(.ramfunc.*))
                0x00000000240000c0                . = ALIGN (_region_min_align)
 *fill*         0x00000000240000b8        0x8 
                0x0000000024000100                . = ALIGN ((0x1 << LOG2CEIL (__ramfunc_size)))
 *fill*         0x00000000240000c0       0x40 
                0x0000000024000100                __ramfunc_end = .
                0x0000000000000080                __ramfunc_size = (__ramfunc_end - __ramfunc_start)
                0x000000009003ce0c                __ramfunc_load_start = LOADADDR (.ramfunc)
```

I have prepared minimal example where the issue can be easily seen
[test_mpu.zip](https://github.com/user-attachments/files/19567906/test_mpu.zip)

Note: I run my code from external qspi flash

**To Reproduce**
Steps to reproduce the behavior:
1. use my minimal example on similar hw
2. see the error

**Logs and console output**

```
*** Booting Zephyr OS build v3.7.1-78-gf000b8a131f3 ***
[00:00:00.074,000] <inf> net_config: Initializing network
[00:00:00.082,000] <inf> net_config: Waiting interface 1 (0x24001418) to be up...
[00:00:04.259,000] <inf> net_config: Interface 1 (0x24001418) coming up
[00:00:04.270,000] <inf> net_config: Running dhcpv4 client...
[00:00:04.281,000] <inf> test_mpu: main loop
[00:00:05.288,000] <inf> test_mpu: main loop
[00:00:06.296,000] <inf> test_mpu: main loop
[00:00:07.274,000] <inf> net_dhcpv4: Received: 10.42.0.175
[00:00:07.284,000] <inf> net_config: IPv4 address: 10.42.0.175
[00:00:07.293,000] <inf> net_config: Lease time: 3600 seconds
[00:00:07.301,000] <inf> net_config: Subnet: 255.255.255.0
[00:00:07.309,000] <inf> net_config: Router: 10.42.0.1
[00:00:07.319,000] <inf> test_mpu: main loop
[00:00:08.326,000] <inf> test_mpu: main loop
[00:00:09.333,000] <inf> test_mpu: main loop
[00:00:10.341,000] <inf> test_mpu: main loop
[00:00:10.474,000] <inf> phy_mii: PHY (0) Link speed 100 Mb, full duplex
[00:00:11.348,000] <inf> test_mpu: main loop
[00:00:12.356,000] <inf> test_mpu: main loop
[00:00:13.364,000] <inf> test_mpu: main loop
[00:00:14.281,000] <inf> test_mpu: work timeout
[00:00:14.289,000] <err> os: ***** MPU FAULT *****
[00:00:14.297,000] <err> os:   Instruction Access Violation
[00:00:14.305,000] <err> os: r0/a1:  0x00000000  r1/a2:  0x000024d0  r2/a3:  0x24002ec0
[00:00:14.316,000] <err> os: r3/a4:  0x00000002 r12/ip:  0x00000000 r14/lr:  0x90024191
[00:00:14.327,000] <err> os:  xpsr:  0x61000000
[00:00:14.334,000] <err> os: s[ 0]:  0x00000000  s[ 1]:  0x00000000  s[ 2]:  0x24002ec0  s[ 3]:  0x24000081
[00:00:14.347,000] <err> os: s[ 4]:  0x240014d8  s[ 5]:  0x90024191  s[ 6]:  0xffffffff  s[ 7]:  0xffffffff
[00:00:14.359,000] <err> os: s[ 8]:  0x00000000  s[ 9]:  0xaaaaaaaa  s[10]:  0xaaaaaaaa  s[11]:  0x2400a7f0
[00:00:14.372,000] <err> os: s[12]:  0x9002408d  s[13]:  0x24002ec0  s[14]:  0x00000000  s[15]:  0x00000000
[00:00:14.384,000] <err> os: fpscr:  0x00000008
[00:00:14.392,000] <err> os: Faulting instruction address (r15/pc): 0x24000100
[00:00:14.402,000] <err> os: >>> ZEPHYR FATAL ERROR 20: Unknown error on CPU 0
[00:00:14.412,000] <err> os: Current thread: 0x24002ec0 (sysworkq)
[00:00:14.421,000] <err> os: Halting system

```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: Zephyr SDK
 - v3.7.1

etienne-lms, 2025-07-03:
I dumped the MPU config for the sample I described above (many thanks to @mathieuchopstm for the GDB script).
I can see the 2 areas:
```
 Region 07: ENABLED (0x24000000~0x2400001F | 32 bytes = 0x20 bytes)
  Attr: teX cb S AP=0b001 NX | SRD: 0x00 (0b0)
 Region 08: ENABLED (0x24000040~0x2400007F | 64 bytes = 0x40 bytes)
  Attr: teX CB s AP=0b110 | SRD: 0x00 (0b0)
```
`nocache` section is *"Outer and inner Non-cacheable Normal"* memory without execution rights.
`ramfunc` section is *"Outer and inner Cacheable (write back; write and read allocate) Normal"* with execution rights.

xhpohanka, 2025-07-04:
Hi @etienne-lms 
can you share the script for dumping MPU areas, please? Maybe it can show us something useful.
I will also try to run the test code from internal memory, I had no time to work on that more, yet :(

etienne-lms, 2025-07-04:
Hi @xhpohanka, here is the script: [dump-mpu.gdb-script.txt](https://github.com/user-attachments/files/21051377/dump-mpu.gdb-script.txt)

xhpohanka, 2025-07-08:
I was able to reproduce the issue on the nucleo-h743zi running from internal flash.
[test_mpu_nucleo.zip](https://github.com/user-attachments/files/21116554/test_mpu_nucleo.zip)

It is clearly connected with `CONFIG_NETWORKING=y`, if it is enabled then I get the failure.

MPU dump without networking
```
MPU is ENABLED and supports 16 regions.
System address map background region is ENABLED for privileged accesses.
 Region 00: ENABLED (0x08000000~0x081FFFFF | 2 MiB = 0x200000 bytes)
  Attr: tex Cb s AP=0b111 | SRD: 0x00 (0b0)
 Region 01: ENABLED (0x24000000~0x2407FFFF | 512 KiB = 0x80000 bytes)
  Attr: teX CB s AP=0b001 NX | SRD: 0x00 (0b0)
 Region 02: ENABLED (0x1FF00000~0x1FF7FFFF | 512 KiB = 0x80000 bytes)
  Attr: tex cb S AP=0b001 NX | SRD: 0x00 (0b0)
 Region 03: ENABLED (0x30040000~0x30043FFF | 16 KiB = 0x4000 bytes)
  Attr: teX cb s AP=0b001 NX | SRD: 0x00 (0b0)
 Region 04: ENABLED (0x30040000~0x300400FF | 256 bytes = 0x100 bytes)
  Attr: tex cb S AP=0b001 | SRD: 0x00 (0b0)
 Region 05: ENABLED (0x90000000~0x9FFFFFFF | 256 MiB = 0x10000000 bytes)
  Attr: tex cb S AP=0b000 | SRD: 0x00 (0b0)
 Region 06: ENABLED (0x24000000~0x2400007F | 128 bytes = 0x80 bytes)
  Attr: teX CB s AP=0b110 | SRD: 0x00 (0b0)
 Region 07: ENABLED (0x24001EC0~0x24001EFF | 64 bytes = 0x40 bytes)
  Attr: teX CB s AP=0b101 NX | SRD: 0x00 (0b0)
 Region 08: DISABLED
 Region 09: DISABLED
 Region 10: DISABLED
 Region 11: DISABLED
 Region 12: DISABLED
 Region 13: DISABLED
 Region 14: DISABLED
 Region 15: DISABLED
```

MPU dump with networking
```
MPU is ENABLED and supports 16 regions.
System address map background region is ENABLED for privileged accesses.
 Region 00: ENABLED (0x08000000~0x081FFFFF | 2 MiB = 0x200000 bytes)
  Attr: tex Cb s AP=0b111 | SRD: 0x00 (0b0)
 Region 01: ENABLED (0x24000000~0x2407FFFF | 512 KiB = 0x80000 bytes)
  Attr: teX CB s AP=0b001 NX | SRD: 0x00 (0b0)
 Region 02: ENABLED (0x1FF00000~0x1FF7FFFF | 512 KiB = 0x80000 bytes)
  Attr: tex cb S AP=0b001 NX | SRD: 0x00 (0b0)
 Region 03: ENABLED (0x30040000~0x30043FFF | 16 KiB = 0x4000 bytes)
  Attr: teX cb s AP=0b001 NX | SRD: 0x00 (0b0)
 Region 04: ENABLED (0x30040000~0x300400FF | 256 bytes = 0x100 bytes)
  Attr: tex cb S AP=0b001 | SRD: 0x00 (0b0)
 Region 05: ENABLED (0x90000000~0x9FFFFFFF | 256 MiB = 0x10000000 bytes)
  Attr: tex cb S AP=0b000 | SRD: 0x00 (0b0)
 Region 06: ENABLED (0x24000000~0x2400001F | 32 bytes = 0x20 bytes)
  Attr: teX cb S AP=0b001 NX | SRD: 0x00 (0b0)
 Region 07: ENABLED (0x24000080~0x240000FF | 128 bytes = 0x80 bytes)
  Attr: teX CB s AP=0b110 | SRD: 0x00 (0b0)
 Region 08: ENABLED (0x24003D00~0x24003D3F | 64 bytes = 0x40 bytes)
  Attr: teX CB s AP=0b101 NX | SRD: 0x00 (0b0)
 Region 09: DISABLED
 Region 10: DISABLED
 Region 11: DISABLED
 Region 12: DISABLED
 Region 13: DISABLED
 Region 14: DISABLED
 Region 15: DISABLED
```

etienne-lms, 2025-07-08:
I don't see the nocache section in the 1st dump (region expected at 0x24000000~0x2400001F).
Maybe you swapped the 2 cases?
Could you provide the `MPU FAULT` error logs in the config where is fails?

FYI, the setup I used (comment https://github.com/zephyrproject-rtos/zephyr/issues/88047#issuecomment-3032171842) already enables `CONFIG_NETWORKING=y`. I'm not sure this information helps :-)
FWIW, I tried with the config option disabled, still functional.

By the way, looking back at your logs in the Issue description, the faulty address is 0x24000100 that is the end of your *ramfunc* section, hence the start of some area that is not mapped. Could it be that this section was not properly loaded and the call to `test()` ended in execution of NOP instructions until the end of the *ramfunc* section mapping is reached?

xhpohanka, 2025-07-08:
Actually I dont understand it at all
```
...
<err> os: Faulting instruction address (r15/pc): 0x24000100
...

...
#5  0x08004770 in z_arm_usage_fault () at /home/honza/dev/zephyrproject/zephyr/arch/arm/core/cortex_m/fault_s.S:102
#6  <signal handler called>
#7  0x24000100 in eth0_data ()
#8  0x0800ae50 in work_queue_main (workq_ptr=0x24001a78 <k_sys_work_q>, p2=<optimized out>, p3=<optimized out>) at /home/honza/dev/zephyrproject/zephyr/kernel/work.c:688
#9  0x240046f0 in sys_work_q_stack ()

(gdb) info symbol 0x24000100
eth0_data in section datas

(gdb) p test_fn 
$15 = {int (void)} 0x24000080 <test_fn>
```

xhpohanka, 2025-07-08:
Aaargh, I have found that we are working on already solved issue.
It was fixed by #80768

etienne-lms, 2025-07-08:
Okay, that's why I don't have the issue!
Nice finding... thanks for the info.

etienne-lms, 2025-07-08:
Thank you @JarmouniA for the backport label.

PR #92848 (2025-07-20): [Backport v3.7-branch] arch: common: fix copy for ramfunc region during XIP init + related fixup commit
Backport 6023d6a1428d880a988f1be3ab1b5da5a6e86d61 (https://github.com/zephyrproject-rtos/zephyr/pull/80768) and b84076a192717894b09d6f4b643c727aa754673c  (https://github.com/zephyrproject-rtos/zephyr/pull/88599)

Fixes #88047