zephyrproject-rtos/zephyr #88057: drivers: serial: stm32: Set baudrate PrescalerValue calculation

neja20, 2025-04-02:
**Is your enhancement proposal related to a problem? Please describe.**

I have UART fed from high speed internal clock branch of 135Mhz. In cases when there is need for low baudrate scenario, such as M-bus communication of 600 baud. It is not possible to reach such baudrates because Prescaler is statically set to `LL_USART_PRESCALER_DIV1`, when function `LL_USART_SetBaudRate()` is called.

**Describe the solution you'd like**

There is a nice calculation of prescaler value in case of Low Power UART. I think something similar could be applicable in case of non Low Power UART instace, because the change of prescaler is really everything that is needed in this case.

What do you think?

erwango, 2026-09-18:
@neja20 Well spotted. Don't hesitate to contribute it

siliwho, 2026-10-23:
if no one is working on this, I’d like to take this one. Could you assign it to me? I’ll test on STM32 hardware with a prescaler capable UART .

neja20, 2026-10-23:
Oh yes, go ahead, i didn’t have time to get to it recently.