zephyrproject-rtos/zephyr #88066: dts: stepper: unify msx-gpios

jilaypandya, 2025-04-02:
**Is your enhancement proposal related to a problem? Please describe.**

At present bindings for micro-step-resolution-gpios is done in a different manner between different stepper drivers. For instance,
- https://github.com/jilaypandya/zephyr/blob/main/dts/bindings/stepper/adi/adi%2Ctmc2209.yaml
- https://github.com/jilaypandya/zephyr/blob/main/dts/bindings/stepper/ti/ti%2Cdrv8424.yaml

**Describe the solution you'd like**

Place msx-gpios in stepper-controller.yaml as a common binding and adjust the two stepper drivers that use m0-gpios and m1-gpios

**Describe alternatives you've considered**

It is either or between msx-gpios and m0/m1/m2-gpios. 
msx-gpios is extensible and can be used for all the drivers that have 1-3 pins for setting micro-step-resolution.

**Additional context**

rruuaanng, 2025-04-03:
Maybe more cases are needed to determine if this approach is worthwhile, so far, there seem to be only two. Over-abstracting can make things incomprehensible. WDYT?

jbehrensnx, 2025-04-03:
From what I have seen, having (two) gpio lines for microstep resolution is very common for stepper controller, and having astandard devicetree specification would make sense in that context. And using an array would make things more flexible for controllers who use a different number of pins for microstep resolution, through I don't know how widespread these are.

faxe1008, 2025-04-03:
I'd go with the `msx-gpios` as well. 👍 
About the placement of the property I am not sure.

jilaypandya, 2025-04-03:
> I'd go with the `msx-gpios` as well. 👍 
> About the placement of the property I am not sure.

Wouldn't the obvious placement be right next to the step-gpios and dir-gpios? 😅

closed 2025-12-05 by jhedberg