zephyrproject-rtos/zephyr #88071: Please add ngphibang to collaborators list

dleach02, 2025-04-02:
# Background

@ngphibang is currently in the maintainers.yml as a collaborator but we somehow skipped adding him to the collaborators list. This prevents the "Pull Request action from selecting him to be a reviewer on the relevant PRs.

[image]

# Nomination

## @ngphibang 

Provide the following information about the GitHub user:

1. Phi Bang Nguyen
2. @ngphibang 
3. NXP

## Supporting Documents

https://github.com/zephyrproject-rtos/zephyr/blob/7c2848486c975db19448023fee99d61583d5ddf3/MAINTAINERS.yml#L2151

[TSC Project Roles]: <https://docs.zephyrproject.org/latest/project/project_roles.html>
[GitHub Permission Level]: <https://docs.github.com/en/organizations/managing-access-to-your-organizations-repositories/repository-roles-for-an-organization>

dleach02, 2025-04-02:
ping @ngphibang

kartben, 2025-04-02:
He _is_ a collaborator... I think there is a bug in the assigner script (or changes in GitHub API?) affecting us as of recently 
https://discord.com/channels/720317445772017664/1014241011989487716/1356955232004866078

kartben, 2025-04-02:
See how eg https://github.com/zephyrproject-rtos/zephyr/pull/86671 was working just fine

kartben, 2025-04-02:
I think I found the cause. Will submit a fix in a few hours.

closed 2025-04-04 by nashif

ngphibang, 2025-04-04:
@kartben : Thank you for fix !