zephyrproject-rtos/zephyr #88079: CFB has been broken.

soburi, 2025-04-03:
**Describe the bug**

The `samples/subsys/fb/cfb` example is not correctly drawn.

```
west build -p -b native_sim/native/64 samples/subsys/display/cfb
./build/zephyr/zephyr.exe
```

https://github.com/zephyrproject-rtos/zephyr/commit/511933054904f3c3f09b8aa51af0fdcfa8262ac1

[image]

**Expected behavior**

https://github.com/zephyrproject-rtos/zephyr/commit/43acb93607c2161608742c59ab98aa1d77c207e5

[image]

Note:

I think it would be better to revert it at once.

kartben, 2025-04-03:
Agreed about reverting. Do you mind lining up the PR for it? Thanks

soburi, 2025-04-03:
@kartben 

I have some business to attend to later, so it will take some time to address. May be tomorrow.

yishai1999, 2025-04-03:
I'll take a look at this

closed 2025-04-18 by kartben