zephyrproject-rtos/zephyr #88080: tests: drivers: flash: frdm_mcxn947 tests/drivers/flash/common failure

hakehuang, 2025-04-03:
**Describe the bug**

```
2025-04-03 12:31:37,161 - twister - DEBUG - DEVICE: ------ TESTSUITE SUMMARY START ------
2025-04-03 12:31:37,170 - twister - DEBUG - DEVICE: SUITE FAIL -  50.00% [flash_driver]: pass = 3, fail = 3, skip = 1, total = 7 duration = 2.556 seconds
2025-04-03 12:31:37,176 - twister - DEBUG - DEVICE: - PASS - [flash_driver.test_flash_copy] duration = 0.205 seconds
2025-04-03 12:31:37,181 - twister - DEBUG - DEVICE: - FAIL - [flash_driver.test_flash_erase] duration = 1.759 seconds
2025-04-03 12:31:37,187 - twister - DEBUG - DEVICE: - FAIL - [flash_driver.test_flash_fill] duration = 0.154 seconds
2025-04-03 12:31:37,193 - twister - DEBUG - DEVICE: - FAIL - [flash_driver.test_flash_flatten] duration = 0.158 seconds
2025-04-03 12:31:37,200 - twister - DEBUG - DEVICE: - PASS - [flash_driver.test_flash_page_layout] duration = 0.074 seconds
2025-04-03 12:31:37,205 - twister - DEBUG - DEVICE: - SKIP - [flash_driver.test_get_size] duration = 0.069 seconds
2025-04-03 12:31:37,212 - twister - DEBUG - DEVICE: - PASS - [flash_driver.test_read_unaligned_address] duration = 0.137 seconds
2025-04-03 12:31:37,216 - twister - DEBUG - DEVICE: ------ TESTSUITE SUMMARY END ------
```

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - frdm_mcxn947
 - What have you tried to diagnose or workaround this issue?
 - git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
```
7ed7cd191afbfa00b9eb6ad2f40b372170260e7e is the first bad commit
commit 7ed7cd191afbfa00b9eb6ad2f40b372170260e7e
Author: Zhaoxiang Jin <Zhaoxiang.Jin_1@nxp.com>
Date:   Wed Mar 12 12:12:31 2025 +0800

    modules: hal_nxp: Move hal_nxp glue layer to zephyr repo

    Move hal_nxp glue layer to zephyr repo.
    Fix build warnings and failures caused by hal_nxp upgrade.
    Update manifest to contain hal_nxp changes.

    Signed-off-by: Zhaoxiang Jin <Zhaoxiang.Jin_1@nxp.com>
```

-->

**To Reproduce**

```
west update
scripts/twister --device-testing --hardware-map /home/ubuntu/nxp/frdm_mcxn947/map.yaml -T tests/drivers/flash/common/
```

**Expected behavior**

Test PASS

**Impact**

flash driver

**Logs and console output**

see above

**Environment (please complete the following information):**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v4.1.0-1748-g29dd0144b70

DerekSnell, 2025-04-03:
No surprise, this same issue impacts the new board `mcx_n9xx_evk` being added in https://github.com/zephyrproject-rtos/zephyr/pull/87639.  These two boards use the same MCXN947 SOC, and are very similar boards.

ZhaoxiangJin, 2025-04-24:
> dup with [#88654](https://github.com/zephyrproject-rtos/zephyr/issues/88654)

The hal_nxp side [PR](https://github.com/zephyrproject-rtos/hal_nxp/pull/527) should fix this issue.

hakehuang, 2025-04-24:
> The hal_nxp side [PR](https://github.com/zephyrproject-rtos/hal_nxp/pull/527) should fix this issue.

need create zephyr pr to update west link to hal

ZhaoxiangJin, 2025-04-24:
> > The hal_nxp side [PR](https://github.com/zephyrproject-rtos/hal_nxp/pull/527) should fix this issue.
> 
> need create zephyr pr to update west link to hal

yes, i will update the hal_nxp sha when the PR get merged.

butok, 2025-06-06:
Fixed for me.