zephyrproject-rtos/zephyr #88098: doc: kernel: Document the use of dynamic stacks

clamattia, 2025-04-03:
**Is your enhancement proposal related to a problem? Please describe.**
Dynamics threads are available but not sufficiently documented. See for example the functions `k_thread_stack_alloc` and `k_thread_stack_free`. Also note that the documentation of `k_thread_create` is overly restrictive. It reads for example:
```
...
Stack objects passed to this function must be originally defined with either of these macros ...
...
```
Given the option for dynamic threads, this is not true, as they may be allocated using `k_thread_stack_alloc`?

**Describe the solution you'd like**
- Improve the documentation of dynamic threads in general
- Improve the documentation of `k_thread_create` to mention dynamic threads specifically

**Describe alternatives you've considered**
None

**Additional context**
None

nashif, 2025-04-03:
@cfriedt  wan't this added by you? Can you take a look?

clamattia, 2025-04-03:
We had also a discussion on discord with @andyross about the API of `k_thread_create` and whether it should allow a `void *stack` allocated for example using `malloc`.

In this request, I would not like to request breaking changes to the API. I would simply like a slight improvement to the docs, to clarify the current API for users.

How can threads be spun up without static allocation?

cfriedt, 2025-04-05:
@clamattia - there are quite a few tests and samples that do this already in the POSIX area, but agree, the docs aren't super clear. I'll add some.

TLDR, the APIs are `k_thread_stack_alloc()` and `k_thread_stack_free()` in `kernel/dynamic.c`.

The APIs are enabled with `CONFIG_DYNAMIC_THREAD=y`.

The main reasons we have  `k_thread_stack_alloc()`, `k_thread_stack_free()` are

1. to account for architecture specific padding
2. to provide a consistent API whether a pool of thread stacks or the heap is used
3. to deal with the intricacies of userspace support, where a kernel object  of type `K_OBJ_THREAD_STACK_ELEMENT` and permissions must be associated with a thread stack.

PR #88200 (2025-04-14): doc: kernel: revise doc for k_thread_stack_alloc and k_thread_create
Update the kernel API in doxygen and rst with references to `k_thread_stack_alloc()` and `k_thread_create()` to convey that kernel thread stacks may be dynamically allocated as an alternative to static allocation.

[Doc Preview](https://builds.zephyrproject.io/zephyr/pr/88200/docs/kernel/services/threads/index.html#spawning-a-thread)

Closes #88098
closed 2025-04-14 by kartben