zephyrproject-rtos/zephyr #88115: lib: posix: clock: Wrong time

trembel, 2025-04-03:
**Describe the bug**
When using `clock_settime()` to set a Unix timestamp that is lower than the current uptime, the resulting `base` `timespec` structure can contain negative nanosecond values. This appears to be due to an issue in the normalization logic in `clock_gettime()` (https://github.com/zephyrproject-rtos/zephyr/blob/c47926e9115ea2f745fb6f40eba5af7c70f7a1b7/lib/posix/options/clock.c#L91C2-L96C3):
```
ts->tv_sec += base.tv_sec;
ts->tv_nsec += base.tv_nsec;
if (ts->tv_nsec >= NSEC_PER_SEC) {
    ts->tv_sec++;
    ts->tv_nsec -= NSEC_PER_SEC;
}
```

If `base.tv_nsec` is negative (because the new timestamp is lower than the uptime), the addition can result in a negative `tv_nsec`. The normalization only handles the case where `tv_nsec` is greater than or equal to `NSEC_PER_SEC` but does not account for negative values. This leads to an incorrect `timespec` being returned by `clock_settime()`.

In the current implementation, the code does the following:

**To Reproduce**
1. Call `clock_settime()` with a Unix timestamp that is lower than the current uptime value (i.e. where `base.tv_sec` and `base.tv_nsec` will result to be negative).
2. Observe that after the addition, the `ts->tv_nsec` field can become negative.
3. The current normalization logic does not adjust for negative nanoseconds, resulting in an invalid time.

**Expected behavior**
The `timespec` structure should be normalized such that `tv_nsec` is always in the range `[0, NSEC_PER_SEC - 1]`. For negative nanosecond values, the code should subtract one from `tv_sec` and add `NSEC_PER_SEC` to `tv_nsec` until `tv_nsec` is non-negative.

**A fix might look like that**
(untested)
```
// Normalize nanoseconds if they are greater than or equal to NSEC_PER_SEC
if (ts->tv_nsec >= NSEC_PER_SEC) {
    ts->tv_sec += ts->tv_nsec / NSEC_PER_SEC;
    ts->tv_nsec %= NSEC_PER_SEC;
}

// Normalize if nanoseconds are negative
while (ts->tv_nsec < 0) {
    ts->tv_sec--;
    ts->tv_nsec += NSEC_PER_SEC;
}
```

cfriedt, 2025-04-05:
@trembel - can you please provide concrete values for `base` and `ts` passed to `clock_settime()` for test purposes? Alternatively, please add a new case to `ZTEST(posix_timers, test_clock_settime)`.

So far, I am unable to reproduce the bug with existing or new test cases.

trembel, 2025-04-07:
I have a MWE here, running on an nRF54L15 DK (had to play a bit with the numbers, until I could produce the non-working behavior, that's why the 'long' delay). I'm on nRF Connect SDK, but IMO the issue is originating in Zephyr, and `clock.c` hasn't changed since then (as the used nrf-sdk commit uses `clock.c` from commit `07b4d3a`:

main.c:
```
#include <stdio.h>
#include "time.h"

int main(void)
{
	printf("Starting app, wait 30s\n");
	k_sleep(K_MSEC(30111));
	printf("Setting time to UNIX timestamp '0'\n");

	struct timespec current = {0};
	clock_gettime(CLOCK_REALTIME, &current);
	printf("Current time: %ld.%ld\n", (uint32_t)current.tv_sec, current.tv_nsec);

	struct timespec ts = {
		.tv_sec = 0,
		.tv_nsec = 0,
	};
	clock_settime(CLOCK_REALTIME, &ts);

	while (1) {
		struct timespec current = {0};
		clock_gettime(CLOCK_REALTIME, &current);
		printf("Current time: %ld.%ld\n", (uint32_t)current.tv_sec, current.tv_nsec);
		k_sleep(K_MSEC(1100));
	}

	return 0;
}
```

prj.conf:
```
CONFIG_POSIX_TIMERS=y
CONFIG_POSIX_API=y
```

Terminal output:
```
*** Booting nRF Connect SDK v2.9.99-d7b2aac6e6e0 ***
*** Using Zephyr OS v4.0.99-060c3f43e430 ***
Starting app, wait 30s
Setting time to UNIX timestamp '0'
Current time: 30.126880000
Current time: 0.0
Current time: 1.101760000
Current time: 2.204160000
Current time: 3.306560000
Current time: 4.408960000
Current time: 5.511360000
Current time: 6.613760000
Current time: 7.716160000
Current time: 8.818560000
Current time: 11.-1079040000
Current time: 11.23616000
Current time: 12.126016000
Current time: 13.228512000
Current time: 14.331008000
Current time: 15.433504000
Current time: 16.536000000
Current time: 17.638496000
Current time: 18.740992000
Current time: 19.843488000
Current time: 22.-1054016000
Current time: 22.48672000
Current time: 23.151072000
Current time: 24.253568000
Current time: 25.356064000
Current time: 26.458560000
Current time: 27.561056000
Current time: 28.663552000
Current time: 29.766048000
Current time: 30.868544000
Current time: 33.-1028960000
Current time: 33.73696000
Current time: 34.176096000
Current time: 35.278592000
Current time: 36.381088000
Current time: 37.483584000
Current time: 38.586080000
Current time: 39.688576000
Current time: 40.791072000
Current time: 43.-1106432000
Current time: 44.-1003776000
Current time: 44.98880000
Current time: 45.201280000
Current time: 46.303776000
Current time: 47.406272000
Current time: 48.508768000
Current time: 49.611264000
Current time: 50.713760000
```

trembel, 2025-04-07:
I will check base and ts, and report them.

cfriedt, 2025-04-08:
Thanks - I'll add that to the test suite.

cfriedt, 2025-04-19:
@trembel - I think I'll add these two functions and some unit tests. Does this sound good?

```cpp
static inline void timespec_normalize(struct timespect *ts)
{
	size_t sec;

	if (ts->tv_nsec < 0) {
		sec = DIV_ROUND_UP(-ts->tv_nsec, NSEC_PER_SEC);
		ts->tv_sec -= sec;
		ts->tv_nsec += sec * NSEC_PER_SEC;
	} else if (ts->tv_nsec >= NSEC_PER_SEC) {
		sec = ts->tv_nsec / NSEC_PER_SEC;
		ts->tv_sec += sec;
		ts->tv_nsec -= sec * NSEC_PER_SEC;
	}

	__ASSERT_NO_MSG((ts.tv_nsec >= 0) && (ts.tv_nsec < NSEC_PER_SEC));
}

static inline void timespec_add(struct timespect *ts, const struct timespec *addend)
{
	ts->tv_sec += add->tv_sec;
	ts->tv_nsec += add->tv_nsec;

	timespec_normalize(ts);
}

```

trembel, 2025-05-02:
Sorry for the late reply, didn't see it. Yes, this looks good for me.

PR #90060 (2025-05-22): sys: timeutil: add utility functions for struct timespec
Add a number of utility functions for manipulating struct timespec.

* timespec_add()
* timespec_compare()
* timespec_equal()
* timespec_is_valid()
* timespec_negate()
* timespec_normalize()
* timespec_sub()
* timespec_from_timeout()
* timespec_to_timeout()

If the `__builtin_add_overflow()` function is available, then these functions use are mostly branchless, which should provide decent performance on systems with caches and branch prediction (gated by `CONFIG_SPEED_OPTIMIZATIONS`). Otherwise, manually written alternatives exist that a [... cut]
closed 2025-05-22 by kartben