zephyrproject-rtos/zephyr #88140: System Workqueue: Improve stack size determination

JordanYates, 2025-04-04:
**Is your enhancement proposal related to a problem? Please describe.**

The stack size for the system workqueue is defined by [CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE](https://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE).
The symbol is defined here:
https://github.com/zephyrproject-rtos/zephyr/blob/f289e47ca5aec0b8f0c95b45a23862183c5832dd/kernel/Kconfig#L578-L582

When other users of the system workqueue require larger stack sizes, they try to override the default value, e.g:

https://github.com/zephyrproject-rtos/zephyr/blob/f289e47ca5aec0b8f0c95b45a23862183c5832dd/drivers/wifi/nrf_wifi/Kconfig.nrfwifi#L552-L553
https://github.com/zephyrproject-rtos/zephyr/blob/f289e47ca5aec0b8f0c95b45a23862183c5832dd/arch/arm64/core/Kconfig#L103-L104

This results in the first default that matches being chosen as the value for `SYSTEM_WORKQUEUE_STACK_SIZE`, which depends on the order of the included Kconfig source files. This is incorrect, as the configured value for the stack should instead be the largest of the desired stack sizes, since the workqueue is shared among all users.

**Describe the solution you'd like**

A solution similar to the construction of the total system heap size ([docs](https://docs.zephyrproject.org/latest/kernel/memory_management/heap.html#defining-the-heap-memory-pool)), where instead of taking the sum of all the configuration options, instead the maximum value is chosen.

nashif, 2025-04-04:
hmm, that `WIFI_NM_WPA_SUPPLICANT ` should not be there

https://github.com/zephyrproject-rtos/zephyr/blob/f289e47ca5aec0b8f0c95b45a23862183c5832dd/kernel/Kconfig#L578-L582

we can't have every other subsys adding there stack size requirements at this level, I think we had the same issue in the past with BT.

So yeah, we need a better solution

average-potato28, 2026-10-16:
working on it , will update in a few days once i fianlize which approach to use for this. [or if i choose to give up in the unlikely scenario 0_0]