zephyrproject-rtos/zephyr #88208: Read of RTC only after write

SoftwareArchitekt, 2025-04-06:
**Describe the bug**
I am using a blackpill_f401cc board. It comes with an STM32F401 CPU. This includes an RTC.
SW Zephyr 4.1.0-1530
Stwps what I have done:
1. Set RTC with rtc_set_time(rtc, &tm) with an date and time
2. Read the date and time with rtc_get_time(rtc, &tm) -> it works
3.  Buffer the RTC with Batterie on VBAT-Connector and GND.
4. Remove rtc_set_time from code
5. flash
6. run program -> read the correct time
7. disconnect from power
8. reconnect power -> rtc_get_time -> error -61 / -ENODATA
9. after debugging, I figure out that in file [zephyr](https://github.com/zephyrproject-rtos/zephyr/tree/main)/[drivers](https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers)/[rtc](https://github.com/zephyrproject-rtos/zephyr/tree/main/drivers/rtc)
/rtc_ll_stm32.c line 531 throw this error

At this point I just want to remove the code from line 525 to 532 for proper working.
How else could it be fixed?

This is my first issue here, so I hope I wrote everything clear - if not, just support me to fill the needed informations

leonardo-munoz-pragmafw, 2025-05-15:
Hi @SoftwareArchitekt , I would like to help with this issue. Do you have any update on it?

etienne-lms, 2025-05-27:
The issue likely comes from power supply being disconnected:
> 7. disconnect from power
> 8. reconnect power -> rtc_get_time -> error -61 / -ENODATA

Power supply voltage needs to be maintained on your device for persistent RTC to be functionnal. Note in the SoC refman:
> As long as the supply voltage remains in the operating range, the RTC never stops,
regardless of the device status (Run mode, low-power mode or under reset).

etienne-lms, 2025-05-27:
No problem. Note that I haven't tested this change.
If you manage to fixes your issue, would you be okay to create a pull request?

closed 2025-09-14 by github-actions[bot]