zephyrproject-rtos/zephyr #88211: Add sphinx markdown output along side of pdf

mkschreder, 2025-04-06:
**Is your feature request related to a problem? Please describe.**
Today AI is the primary reader of documentation. To do this effectively it's good to have markdown docs in a single file. 

**Describe the solution you'd like**
https://docs.zephyrproject.org/4.1.0/zephyr.md

**Describe alternatives you've considered**
I can index pdf and web docs but a lot of concise and useful formatting goes away. It would be best to preserve code blocks and other elements directly in markdown. 

**Additional context**
Sphinx already has markdown support as output. It's a matter of adding it as an extra artifact to be published along side of the pdf docs.

henrikbrixandersen, 2025-04-06:
> Today AI is the primary reader of documentation.

I sure hope not.

kartben, 2025-05-22:
I am not sure there is an easy out-of-the-box solution to build an _all-in-one_ markdown file (but happy to take pointers), and I am reasonably confident that AI agents can crawl / index our HTML docs just fine (see Cursor examples below)

Will be closing as WONTFIX but like I said, happy to re-visit if there is a strong case.

[image]

[image]

closed 2025-05-22 by kartben