zephyrproject-rtos/zephyr #88217: PM: `pm_device_runtime_put_async` not async for `PM_DEVICE_FLAG_ISR_SAFE` devices

JordanYates, 2025-04-07:
**Describe the bug**

When the PM transitions for a device do not use blocking APIs (register writes only), they can be marked as safe for ISRs with the `PM_DEVICE_FLAG_ISR_SAFE` flag. However this currently also changes the behaviour of `pm_device_runtime_put_async`, such that the power down is performed immediately.

https://github.com/zephyrproject-rtos/zephyr/blob/1f9f4c8cee02cfd6cbb45dfa5dcd3913a338917f/subsys/pm/device_runtime.c#L358-L365

This implementation erroneously assumes that because the transition is simple, it should be performed immediately.
This behaviour is not part of the API documentation:
https://github.com/zephyrproject-rtos/zephyr/blob/1f9f4c8cee02cfd6cbb45dfa5dcd3913a338917f/include/zephyr/pm/device_runtime.h#L122-L130

**Expected behavior**

PM release should run at the requested time, not immediately.

**Impact**

PM release cannot be scheduled for a future time if the device PM functions are simple enough (`PM_DEVICE_FLAG_ISR_SAFE`).

**Logs and console output**

**Environment (please complete the following information):**

 - Zephyr v4.1

JordanYates, 2025-04-07:
> Are there devices which can be immediately resumed/suspended which would justify the need for keeping them resumed while nothing is requiring them? I would be more for updating the API compared to adding back a feature I think may be unnecessary and wasteful :)

There are APIs that do not provide a user callback when an action completes (UART interrupt API), and therefore the only reasonable application level action is to schedule the power down at some point in the future.
There may also be applications where the pinctrl state toggling might be problematic (hypothetical).

bjarki-andreasen, 2025-04-07:
> There are APIs that do not provide a user callback when an action completes (UART interrupt API), and therefore the only reasonable application level action is to schedule the power down at some [...]

These would not be ISR safe then would they? If they need to block for some time before being suspended or resumed, they are not ISR safe. The application should surely not need to add "some delay" in between or after calling any APIs. APIs must either be sync or provide a "done" callback, otherwise the app can not be deterministic or portable :)

> There may also be applications where the pinctrl state toggling might be problematic (hypothetical).

Yeah, but again, that would make them not ISR safe at least as I understand it :)

JordanYates, 2025-04-07:
> These would not be ISR safe then would they? If they need to block for some time before being suspended or resumed, they are not ISR safe. The application should surely not need to add "some delay" [...]

I think you're conflating ISR safe and blocking. It is valid to want to call an ISR safe function at some point in the future, and that has nothing to do with blocking. The whole point of `pm_device_runtime_put_async` is to schedule a function to run at some point in the future, whether it is `ISR_SAFE` or both irrelevant and (deliberately?) hidden from the user.

Can you guarantee that every API exposes a callback at the appropriate time to power down the device? If yes, then the current behaviour is fine. But that is not the case in tree right now.

bjarki-andreasen, 2025-04-07:
> > These would not be ISR safe then would they? If they need to block for some time before being suspended or resumed, they are not ISR safe. The application should surely not need to add "some delay" in between or after calling any APIs. APIs must either be sync or provide a "done" callback, otherwise the app can not be deterministic or portable :)
> 
> I think you're conflating ISR safe and blocking. It is valid to want to call an ISR safe function at some point in the future, and that has nothing to do with blocking. The whole point of `pm_device_runtime_put_async` is to schedule a function to run at some point in the future without blocking, whether the function itself is `ISR_SAFE` is both irrelevant and (deliberately?) hidden from the user.

ISR SAFE, or [isr_ok](https://nordicsemi.atlassian.net/browse/NCSDK-32868?focusedCommentId=2069919) rather, is specifically tied to the blocking aspect of the call. `pm_device_runtime_put_async` is there to do two things, delegate calling pm_device_runtime_put() to a thread from ISR context if it is blocking (using K_NO_WAIT), or keeping the device alive for a short amount of time to potentially save a suspend/resume cycle if wait is used.

For K_NO_WAIT, if the call is ISR SAFE, there is no reason to delegate, just call immediately, and even if a wait as supplied, the device can be suspended immediately with no cost, so just do that, its transparent to the application anyway :)

> Can you guarantee that every API exposes a callback at the appropriate time to power down the device? If yes, then the current behaviour is fine. But that is not the case in tree right now.

Applications can inform the PM subsystem that its okay to leave a device powered for X amount of time to optimize power consumption with `pm_device_runtime_put_async`. Whether that is actually optimal depends on the driver, so if its more efficient to suspend immediately (ISR SAFE), we should do that. `pm_device_runtime_put_async` is just a hint, its not a decree to keep the device powered :)

JordanYates, 2025-04-07:
> For K_NO_WAIT, if the call is ISR SAFE, there is no reason to delegate, just call immediately, and even if a wait is supplied, the device can be suspended immediately with no cost, so just do that, its transparent to the application anyway :)

You are still conflating the two. Just because the function is `ISR_SAFE` does not mean that the device is safe to power down right now. There is no possible way it can mean that, because it is totally valid for me to say "turn off in 100ms time, because I'm starting some task that will run for less than 100ms that needs the device powered".

> pm_device_runtime_put_async is just a hint, its not a decree to keep the device powered :)

Strongly disagree, please point to where that reasoning comes from. If I request a device to be turned off in 1 seconds time, I would 100% consider it as a bug if it is turned off at any point before that. No other API that I know of arbitrarily decides that it knows best and silently overrides the user request.

JordanYates, 2025-04-07:
In my opinion nothing you have linked to supports your conclusions.
https://github.com/zephyrproject-rtos/zephyr/blob/d8cd8ddc5193a430c36b578b9f3d31b13a90faa8/include/zephyr/pm/device.h#L181-L182
Please point to the part of the above that means the PM transition *MUST* be performed immediately, regardless of the user request.
https://github.com/zephyrproject-rtos/zephyr/blob/d8cd8ddc5193a430c36b578b9f3d31b13a90faa8/include/zephyr/pm/device.h#L59-L63
Please point to the part of the above that has any bearing whatsover on the behaviour of `pm_device_runtime_async_put`. The documentation of the flag is irrelevant, given a `struct device *` an application uses `pm_device_runtime_async_put`, not an internal flag.

> The quote "suspending and resuming is short as it is executed in the critical section" specifically defines that ISR_SAFE here means the PM action is performed within the critical section

Wonderful, the transition is short. This does not mean the transition should be run sooner than requested.

>  pm_device_runtime_put_async is just a hint, its not a decree to keep the device powered :)

Again, please point to anything that supports this assertion. The documentation of the function is quite specific that this is a *MINIMUM*, not a hint to be ignored.
https://github.com/zephyrproject-rtos/zephyr/blob/798db0f777d0fc1298694ac9431cee6ce94c2103/include/zephyr/pm/device_runtime.h#L135

bjarki-andreasen, 2025-04-08:
ok, lets look at some history :)

The function which today is known as `pm_device_runtime_put_async(const struct device *dev, k_timeout_t timeout)` started out as `pm_device_put(const struct device *dev)`. It was added 4 years ago or so.
https://github.com/zephyrproject-rtos/zephyr/blame/f69759d34a4c2f06e98ddf8072acfa2fd1942cf4/include/pm/device_runtime.h#L82-L96
It was not added to delay suspend, it was just to be able to trigger suspend from ISRs. It had a counterpart `device_pm_put_sync()` which is like todays `pm_device_runtime_put().

Later, the function was renamed to `pm_device_runtime_put_async()`, still with no delay present.
https://github.com/zephyrproject-rtos/zephyr/blame/e3fece5241f66898a4f0d539533b06b485ca029a/include/pm/device_runtime.h#L79-L86 and https://github.com/zephyrproject-rtos/zephyr/blame/0ea173b7747f7d93931d496a8ca4b329edb612b1/include/zephyr/pm/device_runtime.h#L118

The timeout parameter shows up here, about 2 years ago by @ceolin :
https://github.com/zephyrproject-rtos/zephyr/commit/b5ca7a06b434055be5c71ecbc013d42c6ee4357a
The comment states:
_"This allows to delay the put operation what is useful to avoid multiple states transitions."_
The description of the timeout param states "minimum time before suspend", granted, but the point of the parameter is clearly to avoid unnecessary state transitions.

Lastly, about 1 year ago, the ISR_SAFE flag was introduced by @nordic-krch:
https://github.com/zephyrproject-rtos/zephyr/commit/25173f71cda630d4fb0c860d4e45e6f93c0995dd#diff-0fcf6e90adaa9ef2b6d938d59ee274475c698ec0b718a144eb0dce96d1390fb1

This is where `PM_DEVICE_FLAG_ISR_SAFE` is introduced, with a commit message that describes it as _"synchronous runtime PM"_, this is why ISR_SAFE in the context of PM means the device can be resumed and suspended immediately, within the isr context, since, again from the commit message: _"In many cases suspending or resuming of a device is limited to
just a few register writes"_.

So what is more likely, with the new ISR_SAFE flag, @nordic-krch simply overlooked the description of the param of

JordanYates, 2025-04-08:
> "This allows to delay the put operation what is useful to avoid multiple states transitions."
The description of the timeout param states "minimum time before suspend", granted, but the point of the parameter is clearly to avoid unnecessary state transitions.

If the point was purely to avoid unnecessary state transitions, why was `delay` added as a parameter at all? I argue again that there are APIs that don't notify you when they are done with a device, so how are you supposed to schedule the PM constraint to be released at some point in the future if `pm_device_runtime_put_async` silently powers down the device immediately?

> This is where PM_DEVICE_FLAG_ISR_SAFE is introduced, with a commit message that describes it as "synchronous runtime PM"

This is a red herring, because @nordic-krch  explicitly said that he knew the name was wrong, and the whole feature was renamed *from* synchronous *to* ISR safe. Arguing that the original (wrong & changed) name implies that the API change was intended is a stretch IMO.
https://github.com/zephyrproject-rtos/zephyr/pull/67424#issuecomment-1888560251

> So what is more likely: with the new ISR_SAFE flag, @nordic-krch simply overlooked the description of the param of pm_device_runtime_put_async(), or, ISR_SAFE really means something completely different from what was described in the commit message and implemented in code.

I think it is more likely that the behavioral change was overlooked, the PR was merged, and you are using the fact that it was merged to justify the existing documentation of `pm_device_runtime_put_async` being wrong. I don't lend much weight to anything to do with the docs of `ISR_SAFE`, since it doesn't appear that any reviewers (including myself) noticed the behavior change it was introducing for the case of `pm_device_runtime_async_put`, since there was no comments at all about it.

Since it doesn't appear that we are going to agree, there needs to be a general agreement from others on which path to go down.
1. `pm_device_runtime_put_async` should behave differently depending on whether or not the `ISR_SAFE` flag is set internally.
2. `pm_device_runtime_put_async` should behave according to the current API documentation

@ceolin what is your opinion on the two choices?

JordanYates, 2025-04-08:
> The usecase where a user uses pm_device_runtime_put_async() for anything which is not to potentially save a suspend/resume cycle is invalid, like for example, setting a callback, then calling pm_device_runtime_put_async(), and expecting it to fire. There is a reason there is no callback when a device finally changes to suspended, it is irrelevant to the user given at this time, noone has a reference to the device...

Why is that usage invalid? Consider the following interrupt driven UART implementation:
```
static void interrupt_handler(const struct device *dev, void *user_data)
{
  while (uart_irq_update(dev) && uart_irq_is_pending(dev)) {
    if(uart_irq_tx_ready(dev)) {
      buf = k_fifo_get(buf_fifo, K_NO_WAIT);
      if (buf == NULL) {
          uart_irq_tx_disable(dev);
      }
      uart_fifo_fill(dev, buf->data, buf->len);
    }
  }
}

int subsystem_send(struct net_buf *buf)
{
   k_fifo_put(buf_fifo, buf);
   uart_irq_tx_enable(dev);
}
```
Where would you put the PM calls (and which ones) to ensure that the UART is powered while it is transmitting, and powered down when it is not?

nordic-krch, 2025-04-08:
@JordanYates in your example PM calls should not be needed or should be placed in `subsystem_send` and after `uart_fifo_fill` (assuming that fifo_fill consumes whole packet). uart driver must internally call runtime PM API whenever active so explicit runtime PM calls are only to reduce number of suspend/resume.

JordanYates, 2025-04-08:
> @JordanYates in your example PM calls should not be needed or should be placed in subsystem_send and after uart_fifo_fill (assuming that fifo_fill consumes whole packet). uart driver must internally [...]

And for UART drivers that don't manage PM internally? That is still not a required behavior for drivers unless something has changed.

nordic-krch, 2025-04-08:
I thought that it is required. In that case timeout is to cover for lack of runtime PM implementation in the device but then timeout value is just a guess which is risky.

JordanYates, 2025-04-08:
> pm_device_runtime_get() should be called by subsystem, before subsystem_send, and subsystem should call pm_device_runtime_put() after sending is done (there would need to be a done callback called from interrupt_handler or the subsystem_send would need to be blocking, waiting on a sem given from interrupt_handler).

This is my point entirely, THERE IS NO DONE CALLBACK. How can it be put in something that does not exist. The API literally does not provide a way to know when transmission has finished.

JordanYates, 2025-04-08:
> I thought that it is required. In that case timeout is to cover for lack of runtime PM implementation in the device but then timeout value is just a guess which is risky.

As of now, there are exactly two drivers in tree that call `pm_device_runtime_get` on themselves, out of nine that use `PM_DEVICE_DT_INST_GET`. A hit rate of 22% isn't a requirement in my book, and I can't remember ever seeing it specified anywhere as such. There have been discussions that it might be a good idea (#75963), but nothing formalised as far as I know.

bjarki-andreasen, 2025-04-09:
> This is my point entirely, THERE IS NO DONE CALLBACK. How can it be put in something that does not exist. The API literally does not provide a way to know when transmission has finished.

Almost all driver APIs have done callbacks or are blocking, see i2c_transfer, adc_read, and most other APIs in tree. Your snippet could easily have a done callback added as well.

JordanYates, 2025-04-09:
> Almost all driver APIs have done callbacks or are blocking, see i2c_transfer, adc_read, and most other APIs in tree. Your snippet could easily have a done callback added as well.

What? This is the UART interrupt driven API, not something that I have cooked up. It does not provide a done callback. I don't know how many times I can repeat that.

bjarki-andreasen, 2025-04-09:
> What? This is the UART interrupt driven API, not something that I have cooked up. It does not provide a done callback. I don't know how many times I can repeat that.

Most APIs do :) Could this be used for the UART? https://docs.zephyrproject.org/latest/doxygen/html/group__uart__interrupt.html#ga917935a13bf6a5d1e32ef34339e96455

JordanYates, 2025-04-09:
> Most APIs do :) Could this be used for the UART? https://docs.zephyrproject.org/latest/doxygen/html/group__uart__interrupt.html#ga917935a13bf6a5d1e32ef34339e96455

Honestly, I originally discarded that function and forgot about it, since in the nRF driver it uses the same implementation as `uart_irq_tx_ready`, which does not adhere to the `tx_complete` API. I would argue that is a bug in the nRF driver however, and is not a general problem with the API. If that is fixed then yes, there would be an appropriate place to put a `pm_device_runtime_put_async(dev, K_NO_WAIT)` in my snippet. I was wrong.

That said, I still don't like `pm_device_runtime_put_async` behaving differently based on some internal black box state, especially since it is in conflict with the (current) documentation. The semantics of the function *means* would need to be significantly updated.

bjarki-andreasen, 2025-04-09:
> That said, I still don't like `pm_device_runtime_put_async` behaving differently based on some internal black box state, especially since it is in conflict with the (current) documentation. The semantics of the function _means_ would need to be significantly updated.

I agree we either update the doc or the implementation. I believe the best solution from the perspective of PM is to suspend immediately, since that saves more power, but this is a bug for users who expect the device to remain active as documented by the API.

Optimal solution is probably to update the implementation now to adhere to the API, then potentially later adjust the API doc and implementation to suspend ISR_SAFE immediately.

ceolin, 2025-04-15:
@JordanYates @bjarki-andreasen sorry for the delay, I was out on vacation. I tend to prefer the same behavior. That's said, my understanding is that `pm_device_runtime_put_async` is used by the application to avoid toggling device power unnecessary to save energy, assuming that these operations are costly (timing and energy wise). If be ISR safe means that these assumptions are false I don't see much problem in suspend the device right way. The problem though is that now we are adding one more requirement / characteristic for ISR safe devices ...

To be clear, my preference right now is to ISR-safe and non ISR-safe should have the same behavior and the only difference is that ISR-safe can be suspended from an ISR.