zephyrproject-rtos/zephyr #88227: doc: Documentation issue in 'boards/seeed/xiao_esp32c6/doc/index'

physicalit, 2025-04-07:
**Describe the bug**

First of al the example is abd, because you need to choose board for hpcore and lpcore:
xiao_esp32c6

```
CMake Error at /home/giumbai/Projects/esp32_zephyr/zephyr/cmake/modules/boards.cmake:286 (message):
  Board qualifiers `/esp32c6` for board `xiao_esp32c6` not found.  Please
  specify a valid board target.

  Valid board targets for xiao_esp32c6 are:

  xiao_esp32c6/esp32c6/hpcore

  xiao_esp32c6/esp32c6/lpcore

Call Stack (most recent call first):
  /home/user/Projects/esp32_zephyr/zephyr/cmake/modules/zephyr_default.cmake:131 (include)
  /home/user/Projects/esp32_zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/user/Projects/esp32_zephyr/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:4 (find_package)
```
Even after this modification there are still some issues:

```
(base) [user@lab-workstation zephyr]$ west build -b xiao_esp32c6/esp32c6/hpcore samples/basic/blinky
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/user/Projects/esp32_zephyr/zephyr/samples/basic/blinky
-- CMake version: 4.0.0
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 4.1.99 (/home/user/Projects/esp32_zephyr/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- Board: xiao_esp32c6, qualifiers: esp32c6/hpcore
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.17.0 (/home/user/zephyr-sdk-0.17.0)
-- Found toolchain: zephyr 0.17.0 (/home/user/zephyr-sdk-0.17.0)
-- Found Dtc: /home/user/zephyr-sdk-0.17.0/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/user/Projects/esp32_zephyr/zephyr/boards/common/stub.dts
-- Generated zephyr.dts: /home/user/Projects/esp32_zephyr/zephyr/build/zephyr/zephyr.dts
-- Generated pickled edt: /home/user/Projects/esp32_zephyr/zephyr/build/zephyr/edt.pickle
-- Generated devicetree_generated.h: /home/user/Projects/esp32_zephyr/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h
-- Including generated dts.cmake file: /home/user/Projects/esp32_zephyr/zephyr/build/zephyr/dts.cmake
Parsing /home/user/Projects/esp32_zephyr/zephyr/Kconfig
Loaded configuration '/home/user/Projects/esp32_zephyr/zephyr/samples/basic/blinky/prj.conf'
Configuration saved to '/home/user/Projects/esp32_zephyr/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/user/Projects/esp32_zephyr/zephyr/build/zephyr/include/generated/zephyr/autoconf.h'
-- Found GnuLd: /home/user/zephyr-sdk-0.17.0/riscv64-zephyr-elf/riscv64-zephyr-elf/bin/ld.bfd (found version "2.38")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/user/zephyr-sdk-0.17.0/riscv64-zephyr-elf/bin/riscv64-zephyr-elf-gcc
-- Espressif HAL path: /home/user/Projects/esp32_zephyr/modules/hal/espressif
-- Use the esptool.py: /home/user/Projects/esp32_zephyr/modules/hal/espressif/tools/esptool_py/esptool.py
CMake Error at /home/user/Projects/esp32_zephyr/zephyr/cmake/modules/extensions.cmake:4241 (message):
  dt_reg_addr(image_off ...) missing required argument: PATH
Call Stack (most recent call first):
  /home/user/Projects/esp32_zephyr/zephyr/soc/espressif/common/CMakeLists.txt:55 (dt_reg_addr)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/bin/cmake -DWEST_PYTHON=/usr/bin/python -B/home/user/Projects/esp32_zephyr/zephyr/build -GNinja -DBOARD=xiao_esp32c6/esp32c6/hpcore -S/home/user/Projects/esp32_zephyr/zephyr/samples/basic/blinky
```
**Environment**

* Page: `boards/seeed/xiao_esp32c6/doc/index`
* Version: main
* SHA-1: de13103f0b67296b045978fd51836c9101514a0f

sylvioalves, 2025-04-07:
Ok, last ESP32-C6 missed `xiao_esp32c6` board target. Will fix it.

physicalit, 2025-04-07:
@sylvioalves Awesome, Thank you!

I tried to troubleshoot it myself.. unfortunately i run from one issue to another. Hopefully, after I will understand the project better will be able to help more.

PR #88254 (2025-04-09): boards: xiao_esp32c6: add multi-core support
ESP32-C6 now supports both HP and LP core app.
This fixes and updates XIAO board to support that as well.

This PR also updates `cpp/helllo_world` sample so that it defines the `cpp` tag. This fixes the `cpp` ignore filter present in those boards.

Fixes #88227
closed 2025-04-09 by kartben