zephyrproject-rtos/zephyr #88229: mbedtls: update to 3.6.3 in 3.7 branch

aescolar, 2025-04-07:
From https://github.com/zephyrproject-rtos/zephyr/pull/87946#issuecomment-2768963833:
Mbed TLS 3.6 is LTS, Zephyr 3.7 is LTS, Zephyr 3.7.1 had Mbed TLS 3.6.2 so Zephyr 3.7.2 should have Mbed TLS 3.6.3 (fix version + 1) :)
plus Mbed TLS 3.6.3 fixes two CVEs, see https://github.com/Mbed-TLS/mbedtls/releases/tag/mbedtls-3.6.3

mkaranki, 2025-04-14:
Hi,

I noticed build failure on my application after the update while building TF-M:

```
.../modules/crypto/mbedtls/library/md.c:251: undefined reference to `psa_can_do_hash'
```

TF-M 2.1.x upstream branch (https://git.trustedfirmware.org/plugins/gitiles/TF-M/trusted-firmware-m.git/+log/refs/heads/release/2.1.x/) has a fix https://git.trustedfirmware.org/plugins/gitiles/TF-M/trusted-firmware-m.git/+/bdffde9c634c9277c128e19510653be3c136d1a2 which I believe should at least be taken into Zephyr's 3.7 TF-M branch. With the change included, I can get TF-M to pass the build.

There at least two mbedTLS update related changes as well. So maybe it makes most sense to update the whole TF-M to the latest 2.1.x version?

tomi-font, 2025-04-14:
Thanks for the report. Indeed this upstream fix was made after I brought up and discussed this issue with TF-M.
I put up a PR to fix issues that came up in TF-M due to the Mbed TLS update in Zephyr's `main` branch: https://github.com/zephyrproject-rtos/zephyr/pull/88505

Can you test cherry picking [this PR](https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/127) and confirm that it solves the issues for you as well with Zephyr 3.7?

tomi-font, 2025-04-15:
And here is the TF-M 2.1.2 update PR for Zephyr 3.7: https://github.com/zephyrproject-rtos/zephyr/pull/88634
It will supersede https://github.com/zephyrproject-rtos/zephyr/pull/88582.
@mkaranki can you confirm that it still works for you?

PR #87946 (2025-04-13): (backport v3.7) modules: mbedtls: update to 3.6.3
Backport the Mbed TLS 3.6.3 update to the v3.7 branch.

Fixes #88229
PR #88634 (2025-04-30): (backport 3.7) modules: tf-m: update to 2.1.2
3.7 version of https://github.com/zephyrproject-rtos/zephyr/pull/88593

Fixes #88229
closed 2026-09-26 by github-actions[bot]