zephyrproject-rtos/zephyr #88265: Logging using "%.*s" is not using the provided length for buffer size

stefansobol, 2025-04-07:
**Describe the bug**

When using "%.*s" in a log message, one can end up with a warning like:
`<wrn> log: Message dropped because it exceeds size limitation (2047)`

**To Reproduce**

1. Allocate a large buffer and fill the buffer with non null characters.
`ptr = k_malloc(4000);
memset(ptr, 'a', 4000);`
3. Place '\0' at the end of the buffer.
`ptr[3999] = '\0';`
5. Use LOG_INF to print a limited number ob characters
`LOG_INF("test %.*s", 3, ptr);`

**Expected behavior**

The length of the string in the original buffer should not influence the necessary buffer when using "%.*s".
**Impact**

**Logs and console output**

`[00:00:00.440,000] <wrn> log: Message dropped because it exceeds size limitation (2047)`
**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK
 - Version used: v4.1.0-920-g1dd9308e2f78

**Additional context**

I'm using
`CONFIG_LOG_MODE_IMMEDIATE=y`

nordic-krch, 2025-04-08:
Unfortunately, this is not supported by the logging. It is because logging is using only argument types to build a logging message and does not examine content of the format string. This limitation was not mentioned in the documentation, I've added that in #88280.

stefansobol, 2025-04-08:
This feature is useful when working with strings that are not null terminated. By the way, if the required buffer size is less than configured size, logging works just fine.