zephyrproject-rtos/zephyr #88293: tests: drivers: gpio: gpio_basic_api: drivers.gpio.2pin fails on nrf54h20dk@0.9.0/nrf54h20/cpuapp and cpuard

nordic-piks, 2025-04-08:
**Describe the bug**
Tests 
- tests/drivers/gpio/gpio_basic_api/drivers.gpio.2pin

Observed for `nrf54h20dk@0.9.0/nrf54h20/cpuapp` and `nrf54h20dk@0.9.0/nrf54h20/cpurad`.

**To Reproduce**
Steps to reproduce the behavior:
1. haven rf54h20dk@0.9.0/nrf54h20/cpuapp connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/drivers/gpio/gpio_basic_api -p nrf54h20dk@0.9.0/nrf54h20/cpuapp --device-testing --device-serial /dev/ttyACM0 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v4.1.0-2034-g8f2f97ce320a ***
Running TESTSUITE after_flash_gpio_config_trigger
===================================================================
START - test_gpio_config_trigger
 PASS - test_gpio_config_trigger in 0.011 seconds
===================================================================
START - test_gpio_config_twice_trigger
 PASS - test_gpio_config_twice_trigger in 0.011 seconds
===================================================================
TESTSUITE after_flash_gpio_config_trigger succeeded
Running TESTSUITE gpio_port
===================================================================
START - test_gpio_port
Validate device gpio@938000 and gpio@938000
Check gpio@938000 output 6 connected to gpio@938000 input 7
OUT 6 to IN 7 linkage works
- bits_physical
- pin_physical
- check_raw_output_levels
- check_logic_output_levels
- check_input_levels
- bits_logical
- check_pulls

    Assertion failed at ../../../../../../../../../zephyr/tests/drivers/gpio/gpio_basic_api/src/test_gpio_port.c:660: pin_get_config: (flags_get not equal to flags_set)
flags are different
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.17.0
 - Commit SHA or Version used: `v4.1.0-2034-g8f2f97ce320a`

nordic-piks, 2025-04-11:
Fails the same at nrf54l15dk/nrf54l15/cpuapp

nordic-piks, 2025-04-11:
Additionally, similar failure at nrf52840dk/nrf52840 for:
**tests/drivers/gpio/gpio_basic_api/drivers.gpio.nrf_sense_edge**
and **tests/drivers/gpio/gpio_basic_api/drivers.gpio.2pin_arduino**
```
START - test_gpio_port
Validate device gpio@50000300 and gpio@50000300
Check gpio@50000300 output 1 connected to gpio@50000300 input 2
OUT 1 to IN 2 linkage works
- bits_physical
- pin_physical
- check_raw_output_levels
- check_logic_output_levels
- check_input_levels
- bits_logical
- check_pulls

    Assertion failed at ../../../../../../../../zephyr/tests/drivers/gpio/gpio_basic_api/src/test_gpio_port.c:660: pin_get_config: (flags_get not equal to flags_set)
flags are different
 FAIL - test_gpio_port in 0.043 seconds
```

and
**tests/drivers/gpio/gpio_hogs/drivers.gpio.hogs**
```
*** Booting Zephyr OS build v4.1.0-2223-g7abda99fe38c ***
Running TESTSUITE gpio_hogs
===================================================================
START - test_gpio_hog_input_config
 PASS - test_gpio_hog_input_config in 0.001 seconds
===================================================================
START - test_gpio_hog_input_direction
 PASS - test_gpio_hog_input_direction in 0.001 seconds
===================================================================
START - test_gpio_hog_output_high_config

    Assertion failed at ../../../../../../../../zephyr/tests/drivers/gpio/gpio_hogs/src/main.c:73: assert_gpio_hog_config: (actual & expected not equal to expected)
GPIO hog gpio@50000000, pin 13 flags not set (0x00020000 vs. 0x00060000)
 FAIL - test_gpio_hog_output_high_config in 0.021 seconds
===================================================================
START - test_gpio_hog_output_high_direction
 PASS - test_gpio_hog_output_high_direction in 0.001 seconds
===================================================================
START - test_gpio_hog_output_low_config

    Assertion failed at ../../../../../../../../zephyr/tests/drivers/gpio/gpio_hogs/src/main.c:73: assert_gpio_hog_config: (actual & expected not equal to expected)
GPIO hog gpio@50000000, pin 14 flags not set (0x00020000 vs. 0x00060000)
 FAIL - test_gpio_hog_output_low_config in 0.021 seconds
```

katgiadla, 2025-05-06:
Issues occurs on

```
- scenarios:
    - drivers.gpio.2pin
  platforms:
    - nrf54h20dk@0.9.0/nrf54h20/cpuapp
    - nrf54h20dk@0.9.0/nrf54h20/cpurad
    - nrf54l15dk/nrf54l15/cpuflpr
    - nrf54l15dk/nrf54l15/cpuapp

- scenarios:
    - drivers.gpio.nrf_sense_edge
    - drivers.gpio.hogs
    - drivers.gpio.2pin_arduino
  platforms:
    - nrf52840dk/nrf52840
```
what is status of this issue?

wearyzen, 2025-05-06:
2 more instances blocking https://github.com/zephyrproject-rtos/zephyr/pull/89370:
https://github.com/zephyrproject-rtos/zephyr/actions/runs/14856947600/job/41712752094?pr=89370
https://github.com/zephyrproject-rtos/zephyr/actions/runs/14856947600/job/41712751562?pr=89370

wearyzen, 2025-05-06:
@kartben @aescolar could we disable the test for bsim boards till the issue is fixed?

bjarki-andreasen, 2025-05-07:
I will work on it now :)

aescolar, 2025-05-07:
tests/drivers/gpio/gpio_hogs failure bisected to bea0a068e1b8bc8e9850c6e6281bede2fa65c258

bjarki-andreasen, 2025-05-07:
> tests/drivers/gpio/gpio_hogs failure bisected to [bea0a06](https://github.com/zephyrproject-rtos/zephyr/commit/bea0a068e1b8bc8e9850c6e6281bede2fa65c258)

should be fixed by #89595

wearyzen, 2025-05-07:
Hi @bjarki-andreasen , thank you for looking into this. I tried #89595 and still see the gpio hogs issue with bsim. Could you please confirm if the commands mentioned here https://github.com/zephyrproject-rtos/zephyr/issues/88293#issuecomment-2854322667 ran successfully for you?

aescolar, 2025-05-07:
> Hi @bjarki-andreasen , thank you for looking into this. I tried https://github.com/zephyrproject-rtos/zephyr/pull/89595 and still see the gpio hogs issue with bsim. Could you please confirm if the [...]

At least to me the whole
`twister -T tests/drivers/gpio -p nrf52_bsim  -p nrf5340bsim/nrf5340/cpuapp -p nrf54l15bsim/nrf54l15/cpuapp`
works with that PR.

wearyzen, 2025-05-07:
> > Hi [@bjarki-andreasen](https://github.com/bjarki-andreasen) , thank you for looking into this. I tried [#89595](https://github.com/zephyrproject-rtos/zephyr/pull/89595) and still see the gpio hogs issue with bsim. Could you please confirm if the commands mentioned here [#88293 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/88293#issuecomment-2854322667) ran successfully for you?
> 
> At least to me the whole `twister -T tests/drivers/gpio -p nrf52_bsim -p nrf5340bsim/nrf5340/cpuapp -p nrf54l15bsim/nrf54l15/cpuapp` works with that PR.

Must be something wrong with my workspace then, thanks for confirming though.

PR #89595 (2025-05-09): nRF GPIO: Fix get_config not returning initial output setting
The nRF GPIO hardware does not store the initial output value set resulting from gpio_pin_configure() and thus, when gpio_get_config() is used, the initial value is not returned.

This commit reads the output value, and converts this to the corresponding INIT flag in the config

fixes: #88293
closed 2025-05-09 by kartben