zephyrproject-rtos/zephyr #88303: CMake cache variable not propagated correctly in sysbuild context

PragatiGarg-eaton, 2025-04-08:
**Description**: There appears to be an issue with the propagation of CMake cache variables when using sysbuild. Specifically, changing the value of a cache variable from the west build command does not have any impact, and the default value from CMakeLists.txt is always propagated.

**Steps to Reproduce:**

- Define a cache variable in CMakeLists.txt:
- option(FOO "Enable Application Validation")
- Attempt to override this variable using the west build command:
- west build --sysbuild ... -- -Dmy_sample_FOO=ON (Ref: https://docs.zephyrproject.org/latest/build/sysbuild/index.html#cmake-variable-namespacing)
- Observe that the value of FOO remains as "default_value" instead of "ON".

**Expected Behavior:** The value of FOO should be overridden to "ON" when specified in the west build command.

**Actual Behavior:** The value of FOO remains as "default_value" regardless of the override attempt.

**Additional Information:**
The issue persists even when using the zephyr_get function.
The CMake cache variable is not recognized by CMakeLists.txt when set via the west build command.

**Sample Code:**
CMakeLists.txt
option(FOO "Enable Application Validation")

Attempt to use the variable
message(STATUS "FOO is set to ${FOO}")

**Build Command:**
west build --sysbuild ... -- -Dmy_sample_FOO=ON

**Environment:**

Zephyr version: 3.6.0
CMake version: 3.31.6
West version: 1.2.0
Operating System: Windows

PragatiGarg-eaton, 2025-04-14:
Hi @henrikbrixandersen, @tejlmand do we have any updates on this issue?

PragatiGarg-eaton, 2025-04-28:
@nashif, @henrikbrixandersen , @tejlmand, it has been quite a while since I opened this issue, do you have any update on it? Are there any other workarounds I can consider?

boaks, 2025-05-11:
Seems to be the same on my side.
NCS: 2.9.1
Zephyr: 3.7.99
Pretty nasty.

"medium" only for those, who either don't rely on sysbuild, or don't pass in C-defines.
For those who use sysbuild and pass in C-defines it's pretty nasty.

PragatiGarg-eaton, 2025-05-11:
@nashif @erwango @jukkar  @kartben @tejlmand, this issue has been reported for over than a month now. Could someone please address it?

boaks, 2025-05-13:
See answers in #82678  .

Please test to add

```
...
zephyr_get(FOO SYSBUILD)
...
```

to your CMakeLists.txt.

PragatiGarg-eaton, 2025-05-28:
@boaks, thanks for the suggestion. Unfortunately, this configuration isn't working for me. 
I want to set my custom Kconfig root file based on the value of a CMake cache variable from west build command, but this approach isn't effective. 

Despite setting the variable through the west build command and later passing it to the zephyr_get function, the Kconfig root file isn't being set. By the time the zephyr_get function is called, the Zephyr system has already parsed the default Kconfig root file if it isn't found.
cc: @nordicjm

closed 2025-08-13 by github-actions[bot]