zephyrproject-rtos/zephyr #88319: i2c_mcux_lpi2c_rtio does not build for MCXN947

ubieda, 2025-04-08:
**Describe the bug**

Building for MCXN947 and I2C_RTIO fails due to this driver not being updated with latest MCUX SDK changes (#70866). As a result, applications using RTIO won't work (e.g: Sensors using Model V2).

**To Reproduce**

Build command:
```
west build -b frdm_mcxn947/mcxn947/cpu0 samples/subsys/shell/shell_module -- -DCONFIG_I2C=y -DCONFIG_I2C_SHELL=y -DCONFIG_I2C_RTIO=y
```

Ouptut:
```
[2/247] Generating include/generated/zephyr/version.h
-- Zephyr version: 4.1.99 (/home/ubieda/zephyrproject/zephyr), build: v4.1.0-2109-g4925e22a6798
[240/247] Building C object zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_mcux_lpi2c_rtio.c.obj
FAILED: zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_mcux_lpi2c_rtio.c.obj 
ccache /home/ubieda/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc -DCPU_MCXN947VDF_cm33_core0 -DKERNEL -DK_HEAP_MEM_POOL_SIZE=1024 -DLPFLEXCOMM_INIT_NOT_USED_IN_DRIVER=1 -DNDEBUG -DPICOLIBC_LONG_LONG_PRINTF_SCANF -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR_SUPERVISOR__ -D__ZEPHYR__=1 -I/home/ubieda/zephyrproject/zephyr/build/zephyr/include/generated/zephyr -I/home/ubieda/zephyrproject/zephyr/include -I/home/ubieda/zephyrproject/zephyr/build/zephyr/include/generated -I/home/ubieda/zephyrproject/zephyr/soc/nxp/mcx -I/home/ubieda/zephyrproject/zephyr/include/zephyr/posix -I/home/ubieda/zephyrproject/zephyr/lib/posix/options/getopt -I/home/ubieda/zephyrproject/zephyr/soc/nxp/mcx/mcxn/. -I/home/ubieda/zephyrproject/zephyr/subsys/rtio -I/home/ubieda/zephyrproject/zephyr/subsys/shell/modules/kernel_service/thread/.. -I/home/ubieda/zephyrproject/modules/hal/cmsis/CMSIS/Core/Include -I/home/ubieda/zephyrproject/zephyr/modules/cmsis/. -I/home/ubieda/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/drivers/common -I/home/ubieda/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/drivers/lpflexcomm -I/home/ubieda/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/drivers/mcx_spc -I/home/ubieda/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/drivers/port -I/home/ubieda/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/drivers/cache/cache64 -I/home/ubieda/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/drivers/lpflexcomm/lpi2c -I/home/ubieda/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/drivers/lpflexcomm/lpuart -I/home/ubieda/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/devices/MCX/MCXN/MCXN947 -I/home/ubieda/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/devices/MCX/MCXN/periph -I/home/ubieda/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/devices/MCX/MCXN/MCXN947/drivers -isystem /home/ubieda/zephyrproject/zephyr/lib/libc/common/include -Wshadow -fno-strict-aliasing -Os -imacros /home/ubieda/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/autoconf.h -fno-printf-return-value -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m33 -mthumb -mabi=aapcs -mfp16-format=ieee -mtp=soft --sysroot=/home/ubieda/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi -imacros /home/ubieda/zephyrproject/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -ftls-model=local-exec -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=/home/ubieda/zephyrproject/zephyr/samples/subsys/shell/shell_module=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/ubieda/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/ubieda/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -mcmse -specs=picolibc.specs -D_POSIX_THREADS -std=c99 -MD -MT zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_mcux_lpi2c_rtio.c.obj -MF zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_mcux_lpi2c_rtio.c.obj.d -o zephyr/drivers/i2c/CMakeFiles/drivers__i2c.dir/i2c_mcux_lpi2c_rtio.c.obj -c /home/ubieda/zephyrproject/zephyr/drivers/i2c/i2c_mcux_lpi2c_rtio.c
/home/ubieda/zephyrproject/zephyr/drivers/i2c/i2c_mcux_lpi2c_rtio.c: In function 'mcux_lpi2c_isr':
/home/ubieda/zephyrproject/zephyr/drivers/i2c/i2c_mcux_lpi2c_rtio.c:283:39: warning: passing argument 1 of 'LPI2C_MasterTransferHandleIRQ' makes integer from pointer without a cast [-Wint-conversion]
  283 |         LPI2C_MasterTransferHandleIRQ(base, &data->handle);
      |                                       ^~~~
      |                                       |
      |                                       LPI2C_Type *
In file included from /home/ubieda/zephyrproject/zephyr/drivers/i2c/i2c_mcux_lpi2c_rtio.c:18:
/home/ubieda/zephyrproject/modules/hal/nxp/mcux/mcux-sdk-ng/drivers/lpflexcomm/lpi2c/fsl_lpi2c.h:953:45: note: expected 'uint32_t' {aka 'unsigned int'} but argument is of type 'LPI2C_Type *'
  953 | void LPI2C_MasterTransferHandleIRQ(uint32_t instance, void *lpi2cMasterHandle);
      |                                    ~~~~~~~~~^~~~~~~~
In file included from /home/ubieda/zephyrproject/zephyr/include/zephyr/sys/util_macro.h:34,
                 from /home/ubieda/zephyrproject/zephyr/include/zephyr/irq_multilevel.h:16,
                 from /home/ubieda/zephyrproject/zephyr/include/zephyr/devicetree.h:20,
                 from /home/ubieda/zephyrproject/zephyr/include/zephyr/device.h:12,
                 from /home/ubieda/zephyrproject/zephyr/include/zephyr/drivers/i2c.h:27,
                 from /home/ubieda/zephyrproject/zephyr/drivers/i2c/i2c_mcux_lpi2c_rtio.c:13:
/home/ubieda/zephyrproject/zephyr/drivers/i2c/i2c_mcux_lpi2c_rtio.c: In function 'mcux_lpi2c_config_func_0':
/home/ubieda/zephyrproject/zephyr/build/zephyr/include/generated/zephyr/devicetree_generated.h:17176:49: error: 'DT_N_S_soc_S_peripheral_50000000_S_flexcomm_94000_S_lpi2c_94800_IRQ_IDX_0_VAL_priority' undeclared (first use in this function); did you mean 'DT_N_S_soc_S_peripheral_50000000_S_flexcomm_94000_S_lpi2c_94800_REG_IDX_0_VAL_ADDRESS'?
17176 | #define DT_N_INST_0_nxp_lpi2c                   DT_N_S_soc_S_peripheral_50000000_S_flexcomm_94000_S_lpi2c_94800
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ubieda/zephyrproject/zephyr/include/zephyr/sys/util_internal.h:72:26: note: in definition of macro '__DEBRACKET'
   72 | #define __DEBRACKET(...) __VA_ARGS__
      |                          ^~~~~~~~~~~
/home/ubieda/zephyrproject/zephyr/include/zephyr/sys/util_internal.h:64:9: note: in expansion of macro '__GET_ARG2_DEBRACKET'
   64 |         __GET_ARG2_DEBRACKET(one_or_two_args _if_code, _else_code)
      |         ^~~~~~~~~~~~~~~~~~~~
/home/ubieda/zephyrproject/zephyr/include/zephyr/sys/util_internal.h:59:9: note: in expansion of macro '__COND_CODE'
   59 |         __COND_CODE(_XXXX##_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~
/home/ubieda/zephyrproject/zephyr/include/zephyr/sys/util_macro.h:204:9: note: in expansion of macro 'Z_COND_CODE_1'
  204 |         Z_COND_CODE_1(_flag, _if_1_code, _else_code)
      |         ^~~~~~~~~~~~~
/home/ubieda/zephyrproject/zephyr/include/zephyr/devicetree.h:4986:9: note: in expansion of macro 'COND_CODE_1'
 4986 |         COND_CODE_1(DT_HAS_COMPAT_STATUS_OKAY(DT_DRV_COMPAT),   \
      |         ^~~~~~~~~~~
/home/ubieda/zephyrproject/zephyr/include/zephyr/arch/arm/irq.h:100:9: note: in expansion of macro 'BUILD_ASSERT'
  100 |         BUILD_ASSERT(((flags_p & IRQ_ZERO_LATENCY) && \
      |         ^~~~~~~~~~~~
/home/ubieda/zephyrproject/zephyr/include/zephyr/arch/arm/irq.h:123:9: note: in expansion of macro '_CHECK_PRIO'
  123 |         _CHECK_PRIO(priority_p, flags_p) \
      |         ^~~~~~~~~~~
/home/ubieda/zephyrproject/zephyr/include/zephyr/irq.h:49:9: note: in expansion of macro 'ARCH_IRQ_CONNECT'
   49 |         ARCH_IRQ_CONNECT(irq_p, priority_p, isr_p, isr_param_p, flags_p)
      |         ^~~~~~~~~~~~~~~~
/home/ubieda/zephyrproject/zephyr/drivers/i2c/i2c_mcux_lpi2c_rtio.c:385:17: note: in expansion of macro 'IRQ_CONNECT'
  385 |                 IRQ_CONNECT(DT_INST_IRQN(n),                            \
      |                 ^~~~~~~~~~~
/home/ubieda/zephyrproject/zephyr/include/zephyr/devicetree.h:2679:9: note: in expansion of macro 'DT_CAT5'
 2679 |         DT_CAT5(node_id, _IRQ_IDX_, idx, _VAL_, cell)
      |         ^~~~~~~
/home/ubieda/zephyrproject/zephyr/include/zephyr/devicetree.h:4595:9: note: in expansion of macro 'DT_IRQ_BY_IDX'
 4595 |         DT_IRQ_BY_IDX(DT_DRV_INST(inst), idx, cell)
[... 384 lines cut ...]
[... cut ...]
/home/ubieda/zephyrproject/zephyr/include/zephyr/sys/util_internal.h:105:36: note: in expansion of macro 'DT_FOREACH_OKAY_INST_nxp_lpi2c'
  105 | #define UTIL_PRIMITIVE_CAT(a, ...) a##__VA_ARGS__
      |                                    ^
/home/ubieda/zephyrproject/zephyr/drivers/i2c/i2c_mcux_lpi2c_rtio.c:393:1: note: in expansion of macro 'DT_INST_FOREACH_STATUS_OKAY'
  393 | DT_INST_FOREACH_STATUS_OKAY(I2C_MCUX_LPI2C_INIT)
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.
```

**Expected behavior**
Driver compiles and works.

ubieda, 2025-04-08:
Marking as low as it seems this platform did not ever have support for this driver(#70866 was introducing such support). @dleach02 feel free to override if you disagree.

teburd, 2025-04-14:
Thanks for fixing this, just ran into it

PR #88315 (2025-04-14): drivers: Update MCUX I2C RTIO driver with SDK changes
### Description
Update the driver to account for variations in the SDK driver
when it uses the instance number instead of the base address.

Changes come from #70866 and #83704 applied to I2C RTIO.

Fixes #88319.

### Testing

Built for MCXN947 and tested I2C commands work.

```
west build -b frdm_mcxn947/mcxn947/cpu0 samples/subsys/shell/shell_module -- -DCONFIG_I2C=y -DCONFIG_I2C_SHELL=y -DCONFIG_I2C_RTIO=y
```

Output:
```
uart:~$ i2c scan lpi2c@
  lpi2c@b7800  lpi2c@94800  
uart:~$ i2c scan lpi2c@94800 
     0  1  2  3  4  5  6  7  8  9 [... cut]
PR #88615 (2025-07-04): [Backport v4.1-branch] drivers: Update MCUX I2C RTIO driver with SDK changes
Backport d7ebf672bd481164b81919b78c338c2563c7d144~4..d7ebf672bd481164b81919b78c338c2563c7d144 from #88315.

Fixes #88319.
closed 2025-04-14 by kartben