zephyrproject-rtos/zephyr #88324: build_info.yml not properly regenerated on non-pristine builds

edersondisouza, 2025-04-09:
**Describe the bug**
`build_info.yml` is not properly regenerated on non-pristine builds (those without `-p`). Essentially, this causes it to contain, instead of expected data, commented-out cmake yaml lists (used to handle cmake generator expressions[genexes]). It seems like, after `build_info.yml` exists, target `build_info_yaml_saved` is never run again. However, new builds may update its contents, possibly with genexes, but they won't be properly converted.

This incomplete `build_info.yml` then breaks EDK generation, as it depends on it.

**To Reproduce**

```
west build -p -b qemu_cortex_r5 samples/subsys/llext/edk/app
touch samples/subsys/llext/edk/app/CMakeLists.txt 
west build -b qemu_cortex_r5 samples/subsys/llext/edk/app
```
Note that in my tests, the changes need to involve the build system. Not sure if there are other changes that can trigger this.

See that the generated `build_info.yml` has something like this:

```
(...)
  llext-edk:
#   cflags: @YAML-LIST@;-D$<JOIN:KERNEL$<SEMICOLON>__ZEPHYR__=1$<SEMICOLON>(...)
    file: /home/ederson/work/zephyr/zephyr-west/zephyr/build/zephyr/llext-edk.tar.xz
#   include-dirs: @YAML-LIST@;$<TARGET_PROPERTY:zephyr_interface,INTERFACE_INCLUDE_DIRECTORIES>
  toolchain:
    name: zephyr
(...)
```

**Expected behavior**
A completely generated `build_info.yml`.

**Impact**
Annoyance. Time wasted. Higher energy costs? Warmer planet?
One can workaround that by 1) sticking to pristine builds or 2) removing `build_info.yml` before a new build. Still annoying.

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain Zephyr SDK 0.17.0
 - Latest main (ddedff71557552ed6c4cb88ce14299217111f482)

edersondisouza, 2025-04-09:
@pillo79 Have you seen this? My impression is that we may have to remove `build_info.yml` to ensure it's completely generated on every build - at least, build that touches build files.

pillo79, 2025-04-10:
I routinely do `-p` so I didn't, but thanks for noticing! In fact the issue is that `file(GENERATE ...)` helpfully updates the timestamp only when the contents change, so on a re-run the timestamp of the intermediate file is older than the output and nothing gets rebuilt. 

Adding 
```
    FILE(TOUCH ${json_file})
```
just after the `FILE(GENERATE` in `cmake/modules/yaml.cmake` fixes the issue AFAICS.

Will submit a fix as soon as #87752 is merged, as the var name is about to change. 🙂

PR #88410 (2025-04-14): cmake: yaml: update timestamp on intermediate file save
`file(GENERATE ...)` does not update the output file if the content is unchanged. Since the metadata in `build_info.yml` mostly depends on the build configuration, the timestamp of the intermediate file does not get updated on most rebuilds, while the final file does, due to immediate `file(WRITE ...)` calls. Since the latter is newer, no post-process step is executed and the file is left with commented genexes.

Touching the intermediate file ensures that the post-process step is performed every time, even if the content is unchanged, re [... cut]
closed 2025-04-14 by kartben