zephyrproject-rtos/zephyr #88328: [Backport v3.7-branch] soc: npcx: scfg: select host interface type in global

TomChang19, 2025-04-09:
**Is your enhancement proposal related to a problem? Please describe.**
The Host Interface Type in the DEVCNT register sets the HIF type
(either eSPI or LPC).
Currently, it is configured in the host-interface-related drivers like
eSPI or SHI. However, some I/O pads sourced from VHIF in the other
modules such as GPIO and I3C also rely on this field. It might be
problematic when using those I/Os without enabling eSPI or SHI drivers.

**Describe the solution you'd like**
Move the setting from the specific drivers to the global system initialization function scfg_init().

dkalowsk, 2025-04-15:
@TomChang19 can we close this issue?  Looks like the PR has been merged.

TomChang19, 2025-04-16:
Close the issue. The PR has been merged.

PR #88219 (2025-04-13): [Backport v3.7-branch] soc: npcx: scfg: select host interface type in global
Backport 1aff275642297daf6ac50c5f378ccaa06ba4abfb from #76874.

Fixes #88328
closed 2025-04-16 by TomChang19