zephyrproject-rtos/zephyr #88333: Default runner support for rpi_pico2 (OpeOCD) is missing/incomplete.

gonzolively, 2025-04-09:
The [Programming and Debugging](https://docs.zephyrproject.org/latest/boards/raspberrypi/rpi_pico2/doc/index.html#programming-and-debugging) section of the `rpi_pico2` support page states that a user should be able to use OpenOcd as the default runner (**e.g.** `west flash`) in conjunction with the [Raspberry Pi Debug Probe](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html) — however, that is not the case.

## Steps to Reproduce

Here are the steps I followed, and the resulting error:

1. Physically hooked up my Pico Pi Debug Probe to the appropriate pins on my Pico 2W board.
2. Built my Zephyr application and attempted to flash via `west flash`.
3. I received the following error:

```
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: /Users/knox/Repos/zephyrproject/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
set_adapter_speed_if_not_set
Can't find target/rp2350.cfg
FATAL ERROR: command exited with status 1: /opt/homebrew/bin/openocd -s /Users/knox/Repos/zephyrproject/zephyr/boards/raspberrypi/rpi_pico2/support -s /Users/knox/zephyr-sdk-0.16.8/sysroots/arm64-pokysdk-linux/usr/share/openocd/scripts -f /Users/knox/Repos/zephyrproject/zephyr/boards/raspberrypi/rpi_pico2/support/openocd.cfg -c 'source [find interface/cmsis-dap.cfg]' -c 'source [find target/rp2350.cfg]' -c 'set_adapter_speed_if_not_set 5000' '-c init' '-c targets' -c 'reset init' -c 'flash write_image erase /Users/knox/Repos/zephyrproject/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```

After quite a bit of digging around, I was able to find Raspberry Pi's OpenOCD fork, and I copied the missing config, [rp2350.cfg](https://github.com/raspberrypi/openocd/blob/sdk-2.0.0/tcl/target/rp2350.cfg), from that repo and placed it in the appropriate place in my current OpenOCD install. Though I got further, I still was unable to build as I received this error:

```
checking adapter speed...
Error: BUG: unknown adapter clock mode
adapter speed: 5000 kHz

Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E663B03597613F26
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x4c013477
Info : [rp2350.dap.core0] Cortex-M33 r1p0 processor detected
Info : [rp2350.dap.core0] target has 8 breakpoints, 4 watchpoints
Info : [rp2350.dap.core1] Cortex-M33 r1p0 processor detected
Info : [rp2350.dap.core1] target has 8 breakpoints, 4 watchpoints
Info : starting gdb server for rp2350.dap.core0 on 3333
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* rp2350.dap.core0   cortex_m   little rp2350.swd         running
 1  rp2350.dap.core1   cortex_m   little rp2350.swd         running

Warn : [rp2350.dap.core1] VECTRESET is not supported on this Cortex-M core, using SYSRESETREQ instead.
Warn : [rp2350.dap.core1] Set 'cortex_m reset_config sysresetreq'.
[rp2350.dap.core0] halted due to debug-request, current mode: Thread
xPSR: 0xf9000000 pc: 0x00000088 msp: 0xf0000000
[rp2350.dap.core1] halted due to debug-request, current mode: Thread
xPSR: 0xf9000000 pc: 0x00000088 msp: 0xf0000000
Error: Incorrect RP2040 BOOT ROM version
Error: Debug trampoline not found in RP2040 ROM.
Error: auto_probe failed
auto erase enabled
FATAL ERROR: command exited with status 1: /opt/homebrew/bin/openocd -s /Users/knox/Repos/zephyrproject/zephyr/boards/raspberrypi/rpi_pico2/support -s /Users/knox/zephyr-sdk-0.16.8/sysroots/arm64-pokysdk-linux/usr/share/openocd/scripts -f /Users/knox/Repos/zephyrproject/zephyr/boards/raspberrypi/rpi_pico2/support/openocd.cfg -c 'source [find interface/cmsis-dap.cfg]' -c 'source [find target/rp2350.cfg]' -c 'set_adapter_speed_if_not_set 5000' '-c init' '-c targets' -c 'reset init' -c 'flash write_image erase /Users/knox/Repos/zephyrproject/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```

## Expected Behavior

1. Set up the physical debugger
2. Build my Zephyr application
3. Flash using `west flash`
4. The application is successfully flashed to my board.

## Closing Thoughts

Seeing as the [board.cmake](https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/raspberrypi/rpi_pico2/board.cmake) calls for (`rp2350.cfg` ) in the OpenOCD arguments you'd expect it to be in the OpenOCD code-base already as no other requirements/documentation says otherwise. I'm not sure if this is an oversight where you should have mentioned grabbing the RPI OpenOCD fork, or if it worked at one time and has been pulled from the OpenOCD repos, but either way the documentation seems to be lacking.

soburi, 2025-04-09:
@gonzolively 

Please also see https://docs.zephyrproject.org/latest/boards/raspberrypi/rpi_pico/doc/index.html#using-openocd.
I know the lack of a link to this information is something that needs to be fixed.

soburi, 2025-04-25:
Fix https://github.com/zephyrproject-rtos/zephyr/pull/88827

ajf58, 2025-04-27:
>  I'm not sure if this is an oversight where you should have mentioned grabbing the RPI OpenOCD fork,

It's documented in the commit message that introduces this support. https://github.com/zephyrproject-rtos/zephyr/commit/dae7787542a2550098f361af927fb0bf5fcd584f . I appreciate that not everyone goes digging into commit messages, however. Yes it would be nice to update Zephyr's version of OpenOCD and also get these changes upstreamed.

PR #88827 (2025-05-21): boards: raspberrypi: Aggregate debugging info to rpi_pico docs
I've consolidated the debugging procedures for rpi_pico derived boards into the rpi_pico documentation.

I've written about the blackmagicprobe and uf2 runner.
I hope that this will be linked from the automatically generated table, but this cannot be confirmed as it is not generated in local builds.

- Added information about UF2 and BlackMagic Probe.
- Centralize flash/debug instructions under rpi_pico docs.

Fix #88333
closed 2025-05-21 by kartben