zephyrproject-rtos/zephyr #88336: NXP frdm-rw612 is not using factory MAC address for Ethernet

jukkar, 2025-04-09:
**Describe the bug**
* The `frdm-rw612` does not use the factory set MAC address for the Ethernet interface when it starts. It by default uses random MAC address for the Ethernet interface. The factory MAC address is printed on top of the Ethernet socket in the board.
* The Wi-Fi interface properly uses the factory set MAC for its interface.

**Expected behavior**
* User gets by default the factory set MAC address for the Ethernet interface

**Impact**
* Using the Ethernet interface is annoying as the DHCP address changes constantly because DHCP server gives new IP address to the device after every boot
* As a workaround it is possible to set the `local-mac-address` property but this requires manual work and is rather pointless as the device has a dedicated MAC address for the Ethernet interface so it should use it automatically.

theadib, 2025-04-14:
Hi @jukkar as a workaround have you tried nxp,unique-mac? That should calculate MAC from chip UUID at least for my MCXN947. That gives stable MAC address for each device.
I don't know if RW612 uses that driver :-O
My MCXN947 frdm board has also printed MAC address but there is no NXP MCU default address in the onchip memory for the MAC address. 
Regards, Adib.

jukkar, 2025-04-15:
@theadib Unfortunately that also creates a somewhat random MAC. The first 3 bytes remain the same, but the last 3 are randomized.

jukkar, 2025-04-15:
Just to clarify, what I meant that the last 3 bytes in MAC are random (they change) in every boot in this board, so basically the MAC is still random.

dleach02, 2025-04-18:
@jukkar, it also appears to be a bug for this particular target where we are not using this property.

closed 2025-08-04 by decsny