zephyrproject-rtos/zephyr #88396: renode: `hifive_unleashed/fu540/e51 tests/drivers/console/line_splitting -T drivers.console.line_splitting.plus.some_option` failing

JordanYates, 2025-04-10:
**Describe the bug**

The following test is failing on PRs that do not touch related code, causing CI failures.
```
west build -p -b hifive_unleashed/fu540/e51 tests/drivers/console/line_splitting -T drivers.console.line_splitting.plus.some_option
```

For example:
https://github.com/zephyrproject-rtos/zephyr/actions/runs/14368472709/job/40286906722?pr=87988
https://github.com/zephyrproject-rtos/zephyr/actions/runs/14301219029/job/40076053547?pr=88221

**Expected behavior**

Test should pass.

**Environment (please complete the following information):**

 - main

fkokosinski, 2025-04-11:
Hi @JordanYates, thanks for reporting this!

Were you perhaps successful in recreating this behavior locally? I've tested this behavior on the `zephyrprojectrtos/ci:latest` Docker image and couldn't reproduce this behavior:

```
% ~ west twister -p hifive_unleashed/fu540/e51 --test tests/drivers/console/line_splitting/drivers.console.line_splitting.plus.some_option

[...]
INFO    - 1/1 hifive_unleashed/fu540/e51 tests/drivers/console/line_splitting/drivers.console.line_splitting.plus.some_option PASSED (renode 4.012s <zephyr>)
[...]
```

Or by running the whole root like this:

```
% ~ west twister --testsuite-root tests/drivers/console/line_splitting -v

[...]
INFO    - 1/4 hifive_unleashed/fu540/e51 tests/drivers/console/line_splitting/drivers.console.line_splitting.plus.some_option PASSED (renode 6.289s <zephyr>)
INFO    - 2/4 hifive_unleashed/fu540/e51 tests/drivers/console/line_splitting/drivers.console.line_splitting PASSED (renode 5.098s <zephyr>)
INFO    - 3/4 hifive1@A/fe310           tests/drivers/console/line_splitting/drivers.console.line_splitting.plus.some_option PASSED (renode 5.123s <zephyr>)
INFO    - 4/4 hifive1@A/fe310           tests/drivers/console/line_splitting/drivers.console.line_splitting PASSED (renode 5.143s <zephyr>)
[...]
```

nashif, 2025-04-14:
@msierszulski can you please take a look?

PiotrZierhoffer, 2025-04-14:
Redirected to @fkokosinski , we're actively looking into this

PR #88707 (2025-04-16): tests: robot: renode: Use precise pause for UART keywords
Remove the source of non-determinism during emulation by keeping terminal events reproducible between runs. By using the mechanism of precise pause in Renode emulator every interaction with terminal happens at the same virtual time from the perspective of an emulated system, hence makes it independent of the host. Otherwise the result of a test is susceptible to the load on the host and robot process keywords' timing.

Renode emulator supports precise pausing when specified sequence of characters is received on a terminal. Besides that, [... cut]
closed 2025-04-16 by kartben