zephyrproject-rtos/zephyr #88398: mbedtls: `thingy53/nrf5340/cpuapp/ns samples/net/sockets/http_server -T sample.net.sockets.https.server` error

JordanYates, 2025-04-10:
**Describe the bug**

MbedTLS has a build error in some of the in-tree samples, leading to CI failing on unrelated changes:
See:
https://github.com/zephyrproject-rtos/zephyr/actions/runs/14370996793/job/40294070100?pr=88397

```
/home/jordan/code/zephyr/modules/crypto/mbedtls/library/ssl_tls.c:1497:12: error: 'mbedtls_ssl_context' has no member named 'badmac_seen_or_in_hsfraglen'
 1497 |         ssl->badmac_seen_or_in_hsfraglen = 0;
```

**To Reproduce**

`west build -p -b thingy53/nrf5340/cpuapp/ns samples/net/sockets/http_server -T sample.net.sockets.https.server`

**Expected behavior**

Sample should either build successfully or be excluded from the tests if it is not expected to work.

**Impact**

CI failures

**Environment (please complete the following information):**

 - main

katgiadla, 2025-04-11:
The same issue occurs on `tests/net/socket/register/net.socket.register.tls` on `nrf5340dk/nrf5340/cpuapp/ns` and `nrf9160dk@0.14.0/nrf9160/ns`.

tomi-font, 2025-04-11:
https://github.com/zephyrproject-rtos/zephyr/pull/88505

tomi-font, 2025-04-14:
https://github.com/zephyrproject-rtos/zephyr/pull/88593

closed 2025-04-22 by kartben