zephyrproject-rtos/zephyr #88400: nrf_wifi: FPU configuration is hardcoded

JordanYates, 2025-04-10:
**Describe the bug**

The FPU configuration of the `nrf_fpu` is hardcoded based on the CPU, ignoring whether `CONFIG_FPU` is set or not.
The following commit recognised that there was a problem, but didn't apply a robust fix:
https://github.com/zephyrproject-rtos/nrf_wifi/commit/e59d42046057cdae7b30c4b21e43d5996978b7c7

**Expected behavior**

Compiler FP arguments should be correctly set depending on the values of `CONFIG_FPU`, `CONFIG_FP_HARDABI` and `CONFIG_FP_SOFTABI`.

**Impact**

Build failures unless the application is configured the way that `nrf_wifi` is hardcoded.

**Logs and console output**

```
/home/jordan/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: error: zephyr/zephyr_pre0.elf uses VFP register arguments, modules/nrf_wifi/os/nrf_wifi_osal/libnrf-wifi-osal.a(rx.c.obj) does not
```

**Environment (please complete the following information):**

 - Zephyr v4.1

krish2718, 2025-04-10:
Thanks, but this is already fixed, I am pulling the flags from Zephyr interface to handle all platforms: https://github.com/zephyrproject-rtos/nrf_wifi/commit/f8dbe23c2af8eb06b7a4bd9aa928e93d032a8578

krish2718, 2025-04-15:
@JordanYates  can this be closed now?

JordanYates, 2025-04-15:
Yes, I misread the link and thought it was pending

closed 2025-04-15 by JordanYates