zephyrproject-rtos/zephyr #88443: drivers: ethernet: phy: API add ability to read, whether auto-negotiation is enabled or whether a fixed link is used

clamattia, 2025-04-10:
**Is your enhancement proposal related to a problem? Please describe.**
Make it possible to query the phy of whether auto-negotiation is used or whether the link is fixed. This is necessary for some interfaces.

**Describe the solution you'd like**
Add a function to `Ethernet PHY Interface` to query this information.

**Describe alternatives you've considered**
Caller could directly look in the devicetree. But this seems very rigid and fragile at the same time.

**Additional context**
TBD

clamattia, 2025-04-10:
See also https://github.com/zephyrproject-rtos/zephyr/issues/88355

maass-hamburg, 2025-04-17:
> > In general the ethernet mac should not need to know this. In the stm32 eth driver it had to be done, because the hal was also communicating with the phy in its init function, but that will be removed with [#88483](https://github.com/zephyrproject-rtos/zephyr/pull/88483).
> 
> Some protocols require this information. So the application needs to read it. An example would be to use:
> 
>     ok = net_eth_get_hw_config(iface, ETHERNET_CONFIG_TYPE_AUTO_NEG, &cfg);
> 
> But it is not implemented in a lot of drivers. And to implement it, I suspect, the `mac` needs to know this.

thats to get the info from the eth driver not the phy. 

It might be a good idea to redirect ETHERNET_CONFIG_TYPE_AUTO_NEG ETHERNET_CONFIG_TYPE_LINK and ETHERNET_CONFIG_TYPE_DUPLEX to the phy, which will then after the link changes inform the ethernet driver via the callback about the change.
Maybe this can be done in a general way to not have duplicated code in every ethernet driver.

clamattia, 2025-04-17:
> Maybe this can be done in a general way to not have duplicated code in every ethernet driver.

That was my idea behind this Ticket.

Also an option would be to read it directly from the phy via `net_eth_get_phy` -> the new funcitonality to be defined.

maass-hamburg, 2025-04-17:
> Also an option would be to read it directly from the phy via `net_eth_get_phy` -> the new funcitonality to be defined.

I like that option more, because than they could directly use `phy_configure_link()`. then we could even remove or deprecate ETHERNET_CONFIG_TYPE_AUTO_NEG ETHERNET_CONFIG_TYPE_LINK and ETHERNET_CONFIG_TYPE_DUPLEX. 
We could just have to make it possible to activate and deactivate auto negation via phy_configure_link(). Ether we add another parameter to that function (breaking api change) or add a option like NO_AUTONEG to enum phy_link_speed.

tpambor, 2025-06-19:
Since #91354 there is support to disable auto-negotiation and with https://github.com/zephyrproject-rtos/zephyr/pull/90652 `ETHERNET_CONFIG_TYPE_AUTO_NEG`, `ETHERNET_CONFIG_TYPE_LINK` and `ETHERNET_CONFIG_TYPE_DUPLEX` have been dropped.

I propose adding a new function to the Ethernet PHY API. This function would expose the advertised speeds, with an empty set indicating that auto-negotiation is disabled.