zephyrproject-rtos/zephyr #88484: net: phy: Add entries to `enum phy_link_speed`

clamattia, 2025-04-11:
**Is your enhancement proposal related to a problem? Please describe.**
There is no way to express different standards in the enum [phy_link_speed](https://docs.zephyrproject.org/apidoc/latest/group__ethernet__phy.html#ga9b97fff9fcd6823c9b564b3e86b8da68) today.

**Describe the solution you'd like**
Add entries for enum [phy_link_speed](https://docs.zephyrproject.org/apidoc/latest/group__ethernet__phy.html#ga9b97fff9fcd6823c9b564b3e86b8da68). Or find alternative to give more information about the phy.
Examples missing:
- 10Base-FL
- 100Base-FX
- 100Base-X
- 1000Base-SX
- 1000Base-LX

**Describe alternatives you've considered**
Handling the cable standard separately.

**Additional context**
TBD

maass-hamburg, 2025-04-16:
@clamattia please look into https://github.com/zephyrproject-rtos/zephyr/pull/87194 
The enum is mainly used in the communication between the phy and the mac and therefore it should stay media independent as it is intended by the ethernet specification. the cable standard should only be relevant for the phy, so we need a new api for that.

clamattia, 2025-04-16:
> [@clamattia](https://github.com/clamattia) please look into [#87194](https://github.com/zephyrproject-rtos/zephyr/pull/87194) The enum is mainly used in the communication between the phy and the mac [...]

Thank you. Is there some other way of retrieve the information about the medium/standard?

clamattia, 2025-04-16:
> > > [@clamattia](https://github.com/clamattia) please look into [#87194](https://github.com/zephyrproject-rtos/zephyr/pull/87194) The enum is mainly used in the communication between the phy and the mac and therefore it should stay media independent as it is intended by the ethernet specification. the cable standard should only be relevant for the phy, so we need a new api for that.
> > 
> > 
> > Thank you. Is there some other way of retrieve the information about the medium/standard?
> 
> Currently there is nothing implemented in zephyr. if you want take a look into clause 22 of the ethernet standard, to get some infos how to get infos from the phy in a generic way. But if I remember correctly at least for 100Base phys (at least in the standard and generic registers) don't differentiate between using fiber or copper as it uses the same pcs.

Ok, thanks. Shall I close this then?

closed 2025-04-22 by maass-hamburg