zephyrproject-rtos/zephyr #88488: MCUboot app fails to compile for Raspberry Pi Pico2 (RP2350)

hanan619, 2025-04-11:
**Description:**
When trying to build an MCUboot-based application for the Raspberry Pi Pico2 board (RP2350), the compilation fails with the following error:
`FLASH_MAP (defined at subsys/storage/flash_map/Kconfig:10) has direct dependencies FLASH_HAS_DRIVER_ENABLED with value n, but is currently being y-selected by the following symbols:
 - MCUBOOT_DEVICE_SETTINGS`

**Investigation details:**
The flash controller (qmi) is disabled in the rp2350.dtsi file:
`qmi: flash-controller@400d0000 {
    status = "disabled";
};`
If I try to enable it in an overlay:
`/ {
    chosen {
        zephyr,flash-controller = &qmi;
    };
};

&qmi {
    status = "okay";
};`
then new build errors appear from flash_rpi_pico.h:
`#include <hardware/structs/ssi.h>: No such file or directory`
It seems that flash_rpi_pico.h is designed only for the RP2040 (original Pico/Pico W) flash controller and does not support the RP2350 (Pico2) flash controller.
Pico2 has different hardware, and ssi.h does not exist for it.
**Summary:**
- MCUboot cannot currently be compiled for Pico2 (RP2350).
- The flash controller (qmi) support is missing or incomplete for RP2350.
- The existing flash driver (flash_rpi_pico.h) assumes RP2040 hardware and needs to be updated for RP2350.

de-nordic, 2025-04-11:
Here https://github.com/zephyrproject-rtos/zephyr/pull/86292
Can you also change the title of the issue, because this is not MCUboot problem.

hanan619, 2025-04-15:
@de-nordic
I have checked the PR #86292 
The PR does not work with mcuboot.

for mcuboot, flash_controller needs to be enabled and image definition block should be included in bootloader only and not in the actual app.

Should I create another PR after this gets submitted?

de-nordic, 2025-04-15:
> [@de-nordic](https://github.com/de-nordic) I have checked the PR [#86292](https://github.com/zephyrproject-rtos/zephyr/pull/86292) The PR does not work with mcuboot.
>
> for mcuboot, flash_controller [...]

If you have a fix, that is independent fix, or something you consider the right fix, then you may provide PR to fix it.
But, if the change is required in MCUboot, rather then here, then you need to post that to https://github.com/mcu-tools/mcuboot

hanan619, 2025-04-15:
No the changes will be made in this repo.
Basically my changes are based on this PR #86292 
Once the PR gets merged, then I can create a new pull request