zephyrproject-rtos/zephyr #88534: net_mgmt: event handling is broken

JordanYates, 2025-04-12:
**Describe the bug**

The current mechanism for subscribing to net_mgmt events is currently broken, in that code that register for callbacks (`net_mgmt_init_event_callback`) on certain events will also result in callbacks being run on other, seemingly random events. This also affects the event waiting functions, `net_mgmt_event_wait` and `net_mgmt_event_wait_on_iface`.

The core of the problem is that these APIs are defined at the bitmask level, i.e. the internal code is checking bitmasks to determine whether an event matches the requested events:

https://github.com/zephyrproject-rtos/zephyr/blob/781011b911736ad94a8d7ebde24ceae3998bc03f/subsys/net/ip/net_mgmt.c#L203-L204

The problem is that *ALL* of the events are defined as integer values, not bitmasks. The following example is from the Wi-Fi layer, but it is the same for all the others:
https://github.com/zephyrproject-rtos/zephyr/blob/781011b911736ad94a8d7ebde24ceae3998bc03f/include/zephyr/net/wifi_mgmt.h#L320-L325

https://github.com/zephyrproject-rtos/zephyr/blob/781011b911736ad94a8d7ebde24ceae3998bc03f/include/zephyr/net/wifi_mgmt.h#L360-L366

As a result, if you register for callbacks like so, `net_mgmt_init_event_callback(&cb, wifi_mgmt_event_handler, NET_EVENT_WIFI_CMD_AP_STA_CONNECTED);`, where `NET_EVENT_WIFI_CMD_AP_STA_CONNECTED` == 15), you end up being notified when any of the following events occur, since they all share bits with 15.
```
NET_EVENT_WIFI_CMD_SCAN_RESULT
NET_EVENT_WIFI_CMD_SCAN_DONE
NET_EVENT_WIFI_CMD_CONNECT_RESULT
NET_EVENT_WIFI_CMD_DISCONNECT_RESULT
NET_EVENT_WIFI_CMD_IFACE_STATUS
NET_EVENT_WIFI_CMD_TWT
NET_EVENT_WIFI_CMD_TWT_SLEEP_STATE
NET_EVENT_WIFI_CMD_RAW_SCAN_RESULT
NET_EVENT_WIFI_CMD_DISCONNECT_COMPLETE
NET_EVENT_WIFI_CMD_SIGNAL_CHANGE
NET_EVENT_WIFI_CMD_NEIGHBOR_REP_RECEIVED
NET_EVENT_WIFI_CMD_NEIGHBOR_REP_COMPLETE
NET_EVENT_WIFI_CMD_AP_ENABLE_RESULT
NET_EVENT_WIFI_CMD_AP_DISABLE_RESULT
NET_EVENT_WIFI_CMD_AP_STA_CONNECTED
```

**Potential Solutions**

1.  Convert events to bitmasks

Converting events from integers to bitmasks (as implied by the API), would resolve the issue.
Unfortunately, the command mask only has space for 16 unique events per layer:
https://github.com/zephyrproject-rtos/zephyr/blob/781011b911736ad94a8d7ebde24ceae3998bc03f/include/zephyr/net/net_mgmt.h#L44
The Wi-Fi management layer already defines 17 unique events, and the IPv6 layer defines over 20.

2. Convert management event mask to 64 bits, convert events to bitmasks 

Converting the event mask to 64 bits would give up to 32 additional possible masks to each layer.
This would be a rather intrusive change, requiring updates to every callback definition to change the API signature to accept the 64bit parameter.
Concerns have also been raise that we will eventually run out of events again: https://github.com/zephyrproject-rtos/zephyr/pull/88495#issuecomment-2796903007

3. Convert `net_mgmt_init_event_callback` to register for layers, not events

Update the semantics of `net_mgmt_init_event_callback` so that instead of registering for individual events, users register for the base layer and perform their own event filtering in the callback. This would allow the event IDs to remain integers, but it a significant API change.

It additionally would make the waiting functions (`net_mgmt_event_wait`, `net_mgmt_event_wait_on_iface`) not possible to implement for more than a single event (not that they currently work properly).

**Additional context**

Initial attempt to fix: #88495

jukkar, 2025-04-14:
> you end up being notified when any of the following events occur, since they all share bits with 15.

Yes, but that is why in the event callback one needs to check what event if being fired. This is documented in https://docs.zephyrproject.org/latest/connectivity/networking/api/net_mgmt.html#listening-to-network-events

JordanYates, 2025-04-14:
> Yes, but that is why in the event callback one needs to check what event if being fired. This is documented in [...]

It may be documented there, but it isn't mentioned anywhere on the function [doxygen](https://docs.zephyrproject.org/latest/doxygen/html/group__net__mgmt.html#gaa1d086dcdeb54412073e287129bc50e0). If you can't properly differentiate between different events as part of the API, why pretend that you can? It should just be subscribing for events at the "layer code" level.

There is also the question of what the semantics of `net_mgmt_event_wait` are, if the current behaviour is expected. Instead of `Used to wait synchronously on an event mask` it should be `Used to wait synchronously on any event with the same layer code`. The current usage in tree certainly seems to be assuming that exactly the event requested is being blocked on.
https://github.com/zephyrproject-rtos/zephyr/blob/d453fb94075c5d107f521f8a1ebaf85f3ead32d7/samples/net/cellular_modem/src/main.c#L298-L313

jukkar, 2025-04-16:
Robert and I pondered this a bit offline and we need to overhaul this as Jordan noticed.
* It seems that mapping events to bit is a possible solution. How do we make this extensible (perhaps bitarray? https://github.com/zephyrproject-rtos/zephyr/blob/main/include/zephyr/sys/bitarray.h)
* What would the user API look in this case, do we need a new API for making this work properly. In which case we should probably deprecate the old API unless it could be converted to use this new API somehow.
* We should probably have one central header file that is used to allocate the different layer codes and which could act as a registry. Now the layer codes are scattered around which makes it difficult to assure that there are no duplicate layers. Also we could allocate some layers for out-of-tree code

Comments / ideas?

JordanYates, 2025-04-16:
> Comments / ideas?

My 2 cents. Callbacks registered at the "layer code" level, with individual events filtered out inside the callbacks.
Since there would be no longer be a need for everything to fit in a `uint32_t`, more bits could be allocated to the layers to optimize the internal processing, which right now is a bit of a mess (`global_event_mask` is a bit of a travesty atm).
`event_wait` (or a new function name) would take the layer code and event ID as individual parameters, and loop internally with a `k_timepoint_t` (the looping could be transitioned to something more optimal if there is a need).

jukkar, 2025-04-16:
@JordanYates IMHO your proposal sounds reasonable. Looking at the PR #88495, it would need some tweaks, are you able to work on this?

carlescufi, 2025-04-23:
@JordanYates gentle ping, will you update the PR?

JordanYates, 2025-04-23:
> @JordanYates IMHO your proposal sounds reasonable. Looking at the PR https://github.com/zephyrproject-rtos/zephyr/pull/88495, it would need some tweaks, are you able to work on this?

Sorry, I don't really have the time at the moment to make changes with that sort of scope, there are almost 100 instances of `net_mgmt_add_event_callback` in tree.

jukkar, 2025-04-24:
@JordanYates no worries, I can take a look at this.