zephyrproject-rtos/zephyr #88546: Support for POSIX/Linux local hci_vhci for testing

dsseng, 2025-04-12:
**Is your feature request related to a problem? Please describe.**
There is a need for testing facilities for Bluetooth profiles

**Describe the solution you'd like**

**Describe alternatives you've considered**
There's Bumble support, [like for Android Emulator](https://docs.zephyrproject.org/latest/connectivity/bluetooth/bluetooth-tools.html#android-emulator), so perhaps Bumble VHCI backend could be used, I have not tried this yet. *However*, not having to use an extra layer but so that Zephyr-POSIX app could directly talk to /dev/vhci could be a great option.

**Additional context**
I could try myself at implementing this if approved

jhedberg, 2025-04-13:
I don't think anything new is needed in Zephyr for this. The BlueZ `btvirt` tool already supports creating multiple vhci controllers, and you can hook up Zephyr to one of them using the `native_posix` target and the `--bt-dev=hciN` command line parameter.

dsseng, 2025-04-28:
I've figured out how to run this both with btvirt and Bumble. Managed to even make LE Audio Unicast sample work with some work atop btvirt.

Will post patches to btvirt (and maybe try to make Bumble HCI emulation greater, it currently can run health thermometer or sth like that, but not BAP yet), and also document this more in Zephyr.

Please keep this open and possibly relabel to docs request. Thanks

jhedberg, 2026-09-17:
@dsseng should this still be kept open?

dsseng, 2026-09-24:
I believe it's still relevant, but I do not have the time to document it myself, sorry