zephyrproject-rtos/zephyr #88586: Add support for programmable clock generator devices

javad123javad, 2025-04-14:
**Is your enhancement proposal related to a problem? Please describe.**
There are some commonly used programmable clock generators, such as the SI5351, which can be configured via a bus subsystem (e.g., I²C). Is there currently support for such devices in Zephyr? If not, would it be feasible to add support and corresponding API calls for these types of devices?

**Describe the solution you'd like**
Support for such devices in Zephyr will be added via a standard interface.

rruuaanng, 2025-04-14:
For the clock, we provided https://docs.zephyrproject.org/latest/doxygen/html/group__clock__control__interface.html. Of course, the protocol for interfacing with an external clock is provided by the hardware and depends on the specific implementation chosen by the developer.

javad123javad, 2025-04-15:
Ok, Thanks. So, I can put the drivers in this directory.

closed 2025-04-15 by javad123javad