zephyrproject-rtos/zephyr #88601: k_poll on a pipe does not wait for data to become available

andrew-gillan, 2025-04-14:
**Describe the bug**
k_poll on a pipe no longer waits for data to be available. This results in k_poll always returning immediately on a pipe. This was introduced in 4.1.0 in the following commit:

https://github.com/zephyrproject-rtos/zephyr/commit/a9ab8cb7797d9b2783ce273f022161aea66f52a0

See the change in the `is_condition_met()` function.

**To Reproduce**
k_poll on an empty pipe

**Expected behavior**
The k_poll waits until data is available to return

**Impact**
Can cause unexpected behaviour like tasks consuming much more cpu than designed for or not allowing the cpu to sleep and save power.

**Logs and console output**
n/a

**Environment (please complete the following information):**
Zephyr 4.1.0

**Additional context**
n/a

maass-hamburg, 2025-04-15:
@andrew-gillan the old pipe api (`CONFIG_PIPES`) already is deprecated (will be removed in 4.3.0), maybe already move to the new pipe api.

andrew-gillan, 2025-04-15:
@maass-hamburg I am currently using the new api. That said, this bug is about the k_poll api. The changes to pipes broke k_poll.

maass-hamburg, 2025-04-15:
@Mattemagikern FYI.

PR #88743 (2025-04-21): Fix poll support for pipes
- **tests: poll: add pipe coverage**
- **kernel/pipe: fix poll support**

Fixes #88601
closed 2025-04-21 by kartben