zephyrproject-rtos/zephyr #88603: X-NUCLEO-WB05KN1 UART does not work with STM32 nucleo board

dd1337, 2025-04-14:
**Describe the bug**
After connecting X-NUCLEO-WB05KN1 out of the box to Nucleo h755zi-q via UART (usart2 or lpuart1) BLE cannot be initialised due to receiving unknown command errors, as such:

```
bt_hci_core: opcode 0x1003 status 0x01 BT_HCI_ERR_UNKNOWN_CMD
```

**To Reproduce**

**Expected behavior**
Should initialise BLE

**Impact**
showstopper

**Logs and console output**
```
16:33:59.803 > *** Booting Zephyr OS build v4.1.0 ***
16:33:59.818 > start
16:33:59.818 > [00:00:00.004,000] ␛[0m<dbg> bt_hci_core: bt_hci_cmd_create: opcode 0x0c03 param_len 0␛[0m
16:33:59.818 > [00:00:00.011,000] ␛[0m<dbg> bt_hci_core: bt_hci_cmd_create: buf 0x2402e7c8␛[0m
16:33:59.818 > [00:00:00.019,000] ␛[0m<dbg> bt_hci_core: bt_hci_cmd_send_sync: buf 0x2402e7c8 opcode 0x0c03 len 3␛[0m
16:33:59.819 > [00:00:00.028,000] ␛[0m<dbg> bt_hci_core: bt_tx_irq_raise: kick TX␛[0m
16:33:59.823 > [00:00:00.034,000] ␛[0m<dbg> bt_hci_core: tx_processor: TX process start␛[0m
16:33:59.830 > [00:00:00.041,000] ␛[0m<dbg> bt_hci_core: hci_core_send_cmd: fetch cmd␛[0m
16:33:59.836 > [00:00:00.047,000] ␛[0m<dbg> bt_hci_core: hci_core_send_cmd: Sending command 0x0c03 (buf 0x2402e7c8) to driver␛[0m
16:33:59.846 > [00:00:00.057,000] ␛[0m<dbg> bt_hci_core: bt_send: buf 0x2402e7c8 len 3 type 1␛[0m
16:33:59.854 > [00:00:00.065,000] ␛[0m<dbg> bt_hci_core: bt_tx_irq_raise: kick TX␛[0m
16:33:59.860 > [00:00:00.071,000] ␛[0m<dbg> bt_hci_core: tx_processor: TX process start␛[0[00:00:00.077,000] ␛[1;31m<err> bt_driver: Unknown H:4 type 0x82␛[0m
16:33:59.873 > [00:00:00.083,000] ␛[1;31m<err> bt_driver: Unknown H:4 type 0xff␛[0m
16:33:59.879 > m
16:33:59.879 > [00:00:00.090,000] ␛[0m<dbg> bt_hci_core: bt_recv_unsafe: buf 0x2402a4e4 len 6␛[0m
16:33:59.886 > [00:00:00.097,000] ␛[0m<dbg> bt_hci_core: hci_cmd_complete: opcode 0x0c03␛[0m
16:33:59.893 > [00:00:00.104,000] ␛[0m<dbg> bt_hci_core: hci_cmd_done: opcode 0x0c03 status 0x00 BT_HCI_ERR_SUCCESS buf 0x2402a4e4␛[0m
16:33:59.903 > [00:00:00.114,000] ␛[0m<dbg> bt_hci_core: hci_cmd_done: sync cmd released␛[0m
16:33:59.910 > [00:00:00.121,000] ␛[0m<dbg> bt_hci_core: bt_tx_irq_raise: kick TX␛[0m
16:33:59.917 > [00:00:00.128,000] ␛[0m<dbg> bt_hci_core: tx_processor: TX process start␛[0m
16:33:59.923 > [00:00:00.134,000] ␛[0m<dbg> bt_hci_core: bt_hci_cmd_send_sync: rsp 0x2402e7c8 opcode 0x0c03 len 1␛[0m
16:33:59.932 > [00:00:00.143,000] ␛[0m<dbg> bt_hci_core: hci_reset_complete: status 0x00 BT_HCI_ERR_SUCCESS␛[0m
16:33:59.941 > [00:00:00.152,000] ␛[0m<dbg> bt_hci_core: bt_hci_cmd_create: opcode 0x1003 param_len 0␛[0m
16:33:59.949 > [00:00:00.160,000] ␛[0m<dbg> bt_hci_core: bt_hci_cmd_create: buf 0x2402e7c8␛[0m
16:33:59.956 > [00:00:00.167,000] ␛[0m<dbg> bt_hci_core: bt_hci_cmd_send_sync: buf 0x2402e7c8 opcode 0x1003 len 3␛[0m
16:33:59.965 > [00:00:00.176,000] ␛[0m<dbg> bt_hci_core: bt_tx_irq_raise: kick TX␛[0m
16:33:59.971 > [00:00:00.182,000] ␛[0m<dbg> bt_hci_core: tx_processor: TX process start␛[0m
16:33:59.978 > [00:00:00.189,000] ␛[0m<dbg> bt_hci_core: hci_core_send_cmd: fetch cmd␛[0m
16:33:59.984 > [00:00:00.195,000] ␛[0m<dbg> bt_hci_core: hci_core_send_cmd: Sending command 0x1003 (buf 0x2402e7c8) to driver␛[0m
16:33:59.995 > [00:00:00.206,000] ␛[0m<dbg> bt_hci_core: bt_send: buf 0x2402e7c8 len 3 type 1␛[0m
16:34:00.002 > [00:00:00.213,000] ␛[0m<dbg> bt_hci_core: bt_tx_irq_raise: kick TX␛[0m
16:34:00.008 > [00:00:00.219,000] ␛[0m<dbg> bt_hci_core: bt_recv_unsafe: buf 0x2402a4e4 len 6␛[0m
16:34:00.016 > [00:00:00.226,000] ␛[0m<dbg> bt_hci_core: hci_cmd_status: opcode 0x1003␛[0m
16:34:00.022 > [00:00:00.233,000] ␛[0m<dbg> bt_hci_core: hci_cmd_done: opcode 0x1003 status 0x01 BT_HCI_ERR_UNKNOWN_CMD buf 0x2402a4e4␛[0m
16:34:00.033 > [00:00:00.244,000] ␛[0m<dbg> bt_hci_core: hci_cmd_done: sync cmd released␛[0m
16:34:00.040 > [00:00:00.251,000] ␛[0m<dbg> bt_hci_core: bt_tx_irq_raise: kick TX␛[0m
16:34:00.046 > [00:00:00.257,000] ␛[0m<dbg> bt_hci_core: tx_processor: TX process start␛[0m
16:34:00.053 > [00:00:00.264,000] ␛[1;33m<wrn> bt_hci_core: opcode 0x1003 status 0x01 BT_HCI_ERR_UNKNOWN_CMD␛[0m
16:34:00.061 > bluetooth init failed (err -5)

```

**Environment (please complete the following information):**

Tried with: \
Zephyr 3.7.1 ,4.0.0, 4.1.0

**Additional context**
overlay dts

```
&uart8 {
	status = "disabled";
};

&arduino_serial {
	status = "disabled";
};

/ {
	chosen {
		zephyr,bt-c2h-uart = &usart2;
		zephyr,bt-hci = &bt_hci_uart;
	};
};

&usart2 {
	pinctrl-0 = <&usart2_tx_pd5 &usart2_rx_pd6>;
	pinctrl-names = "default";
	status = "okay";
	current-speed = <921600>;

	bt_hci_uart: bt_hci_uart {
		compatible = "zephyr,bt-hci-uart";
		status = "okay";
	};
};
```

I have also tried to use lpuart1 with the same results.

HoZHel, 2025-04-15:
Hi @dd1337, regarding your issue there are some points described below:
1. X-Nucleo-WB05KN1 comes with a controller+host firmware out of the box which is not compatible with Zephyr. So, you need to change the firmware to controller-only version [(x-cube-wb05n)](https://www.st.com/en/embedded-software/x-cube-wb05n.html). I'll update the documentation on Zephyr regarding this point.

2. The `arduino_serial` node should be enabled on your Nucleo board. I see that it's disabled in your overlay file. Moreover, the `arduino_serial` pins (D1 and D0) on your board are PB6(TX) and PB7(RX) respectively; therefore, `arduino_serial` node should refer to `usart1`. So, there is a mistake in the `arduino_r3_connector.dtsi` file available in the board directory, which I'll fix by a PR.

3. If you want to use other USARTs, you can't attach the board on top and you need to use jumper cables.

With the solution I provided you, I tested `samples/bluetooth/beacon` on Nucleo-H723ZG and it works fine.

[image]

dd1337, 2025-04-15:
@HoZHel thanks for getting back 🙏  

1. Time to get STLink tool then.

2. Thanks, will be looking forward for it. I disabled `arduino_serial` because it is on UART8, which is assigned to D0 and D1, but UART8 runs on the m4 core. I need to flash m7 core so I replaced it with LPUART1. The connection worked, but also resulted in the same error code as reported initially.  Here's the overlay in case someone stumbles upon this issue.
```
&uart8 {
	status = "disabled";
};

&arduino_serial {
	status = "disabled";
};

/ {
	chosen {
		zephyr,bt-c2h-uart = &lpuart1;
		zephyr,bt-hci = &bt_hci_uart;
	};
};

&lpuart1 {
	pinctrl-0 = <&lpuart1_tx_pb6 &lpuart1_rx_pb7>;
	pinctrl-names = "default";
	status = "okay";
	current-speed = <921600>;

	bt_hci_uart: bt_hci_uart {
		compatible = "zephyr,bt-hci-uart";
		status = "okay";
	};
};
```

Thanks again for your support. I will report back in a couple weeks when I receive STLink and flash the shield.

cgbdevelco, 2025-04-22:
@dd1337 I have same nucleo board and since SPI to this board doesn't currently work, I tested UART as backup solution and it works based on your overlay file with shield flashed with the "DTM_UART_WITH_UPDATER_CONTROLLER" firmware.

dd1337, 2025-05-07:
@cgbdevelco thanks for testing.  I can also confirm that it works. Important note that Kconfig must have `CONFIG_BT_EXT_ADV=y` otherwise connection will not work.

closed 2025-05-07 by dd1337