zephyrproject-rtos/zephyr #88621: STM32L4x rng entropy driver fails to compile

mihnen, 2025-04-15:
**Describe the bug**
Target is stm32l4p5. Zephyr version is release 4.1.0.

Enabling the entropy driver with the following config causes the compile to fail:

```
CONFIG_ENTROPY_GENERATOR=y
CONFIG_ENTROPY_STM32_RNG=y
````

This appears to be because the hal drivers are using a different name for 3 different functions than the driver expects. Not sure if other stm32's are affected or not but searching all the hal files the only ones I see that are different are for the stm32l4x.

Patch to fix this is attached.

**Impact**

Cannot use stm32's RNG for entropy which breaks a lot of things.

**Logs and console output**

```
[245/301] Building C object zephyr/drivers/entropy/CMakeFiles/drivers__entropy.dir/entropy_stm32.c.obj
/home/mihnen/develop/personal/zephyr/sensometer-workspace/zephyr/drivers/entropy/entropy_stm32.c: In function 'configure_rng':
/home/mihnen/develop/personal/zephyr/sensometer-workspace/zephyr/drivers/entropy/entropy_stm32.c:227:17: warning: implicit declaration of function 'LL_RNG_DisableCondReset' [-Wimplicit-function-declaration]
  227 |                 LL_RNG_DisableCondReset(rng);
      |                 ^~~~~~~~~~~~~~~~~~~~~~~
/home/mihnen/develop/personal/zephyr/sensometer-workspace/zephyr/drivers/entropy/entropy_stm32.c:229:24: warning: implicit declaration of function 'LL_RNG_IsEnabledCondReset'; did you mean 'LL_RNG_IsEnabledIT'? [-Wimplicit-function-declaration]
  229 |                 while (LL_RNG_IsEnabledCondReset(rng) == 1) {
      |                        ^~~~~~~~~~~~~~~~~~~~~~~~~
      |                        LL_RNG_IsEnabledIT
/home/mihnen/develop/personal/zephyr/sensometer-workspace/zephyr/drivers/entropy/entropy_stm32.c: In function 'recover_seed_error':
/home/mihnen/develop/personal/zephyr/sensometer-workspace/zephyr/drivers/entropy/entropy_stm32.c:280:9: warning: implicit declaration of function 'LL_RNG_EnableCondReset' [-Wimplicit-function-declaration]
  280 |         LL_RNG_EnableCondReset(rng);
      |         ^~~~~~~~~~~~~~~~~~~~~~
[296/301] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map 
```

**Environment (please complete the following information):**

 - OS: Linux
 - Toolchain: arm-zephyr-eabi
 - Commit SHA or Version used: v4.1.0

** Patch **

diff --git a/drivers/entropy/entropy_stm32.c b/drivers/entropy/entropy_stm32.c
index 602daf19277..8dcd2416af5 100644
--- a/drivers/entropy/entropy_stm32.c
+++ b/drivers/entropy/entropy_stm32.c
@@ -224,10 +224,17 @@ static void configure_rng(void)
 		LL_RNG_SetHealthConfig(rng, desired_htcr);
 #endif /* health_test_config */
 
+#if defined(CONFIG_SOC_SERIES_STM32L4X)
+		LL_RNG_ResetConditioningResetBit(rng);
+		/* Wait for conditioning reset process to be completed */
+		while (LL_RNG_IsResetConditioningBitSet(rng) == 1) {
+		}
+#else
 		LL_RNG_DisableCondReset(rng);
 		/* Wait for conditioning reset process to be completed */
 		while (LL_RNG_IsEnabledCondReset(rng) == 1) {
 		}
+#endif /* CONFIG_SOC_SERIES_STM32L4X */
 	}
 #endif /* STM32_CONDRST_SUPPORT */
 
@@ -277,13 +284,23 @@ static int recover_seed_error(RNG_TypeDef *rng)
 {
 	uint32_t count_timeout = 0;
 
-	LL_RNG_EnableCondReset(rng);
-	LL_RNG_DisableCondReset(rng);
+#if defined(CONFIG_SOC_SERIES_STM32L4X)
+		LL_RNG_SetConditioningResetBit(rng);
+		LL_RNG_ResetConditioningResetBit(rng);
+#else
+		LL_RNG_EnableCondReset(rng);
+		LL_RNG_DisableCondReset(rng);
+#endif /* CONFIG_SOC_SERIES_STM32L4X */
+
 	/* When reset process is done cond reset bit is read 0
 	 * This typically takes: 2 AHB clock cycles + 2 RNG clock cycles.
 	 */
 
+#if defined(CONFIG_SOC_SERIES_STM32L4X)
+	while (LL_RNG_IsResetConditioningBitSet(rng) ||
+#else
 	while (LL_RNG_IsEnabledCondReset(rng) ||
+#endif /* CONFIG_SOC_SERIES_STM32L4X */
 		ll_rng_is_active_seis(rng) ||
 		ll_rng_is_active_secs(rng)) {
 		count_timeout++;

FRASTM, 2025-04-16:
Thanks @mihnen for detecting and fixing the problem.

Do you intend to put your patch in a PR ?

mathieuchopstm, 2025-04-16:
Thanks for the report.

Indeed, there is a misalignment between the STM32CubeL4 package and other series', and this wasn't caught before (at the moment, I think the RNG driver is not build as part of CI...)

This issue has been reported internally (ST Reference Number: 207828).

As suggested by @FRASTM , could you submit your patch as a PR?

mihnen, 2025-04-16:
Yes I can open a PR, thanks for confirming.

PR #88719 (2025-04-22): drivers: stm32_hal: stm32_rng Fix compile errors for stm32l4 rng.
The low level function names in the stm32l4xx low level driver are different than all the other stm32's even though the functionality is the same. This breaks the entropy module for these parts. This patch fixes the build and has been tested on a custom stm32l4p5 board.

Fixes: #88621
closed 2025-04-22 by kartben