zephyrproject-rtos/zephyr #88636: doc: tracing: Backport Percepio View section

eriktamlin, 2025-04-15:
This will be used for backport PRs to update tracing documentation regarding the new Percepio View,

Main PR #88635

kartben, 2025-04-15:
Backports are for bug fixes, here you're adding a new "feature" to the documentation so I wouldn't be comfortable having this backported (even more so as the tool was not even available when e.g 3.7 was released).

eriktamlin, 2025-04-16:
It's not really a feature for zephyr, it's just a documentation update to mention a tool that supports e.g. 3.7. There are no code changes in zephyr or submodules. While Percepio View should support everything back to 3.5, I was thinking of only backporting to 4.0 and 4.1 so it's a clean break.