zephyrproject-rtos/zephyr #88654: boards: nxp: frdm_mcxn947: w25q64jvssiq flash support is broken

butok, 2025-04-15:
**Describe the bug**

Support of the w25q64jvssiq flash is broken for frdm_mcxn947.

Flash test is FAILED (tests/drivers/flash/common)

**To Reproduce**
Steps to reproduce the behavior:
1. `west build -p always -b frdm_mcxn947/mcxn947/cpu0 tests/drivers/flash/common`
2. `west flash`
3. See `PROJECT EXECUTION FAILED`
```
*** Booting Zephyr OS build v4.1.0-1332-g7ed7cd191afb ***
Running TESTSUITE flash_driver
===================================================================
START - test_flash_copy
Test will run on device w25q64jvssiq@0
 PASS - test_flash_copy in 0.203 seconds
===================================================================
START - test_flash_erase
Test will run on device w25q64jvssiq@0
i=0:    read_buf[i]=0   expected=255
i=1:    read_buf[i]=1   expected=255
i=2:    read_buf[i]=2   expected=255
i=3:    read_buf[i]=3   expected=255
i=4:    read_buf[i]=4   expected=255
i=5:    read_buf[i]=5   expected=255
i=6:    read_buf[i]=6   expected=255
i=7:    read_buf[i]=7   expected=255
i=8:    read_buf[i]=8   expected=255
i=9:    read_buf[i]=9   expected=255
i=10:   read_buf[i]=10  expected=255
i=11:   read_buf[i]=11  expected=255
i=12:   read_buf[i]=12  expected=255
i=13:   read_buf[i]=13  expected=255
i=14:   read_buf[i]=14  expected=255
i=15:   read_buf[i]=15  expected=255
i=16:   read_buf[i]=16  expected=255
i=17:   read_buf[i]=17  expected=255
i=18:   read_buf[i]=18  expected=255
i=19:   read_buf[i]=19  expected=255
i=20:   read_buf[i]=20  expected=255
i=21:   read_buf[i]=21  expected=255
i=22:   read_buf[i]=22  expected=255
i=23:   read_buf[i]=23  expected=255
i=24:   read_buf[i]=24  expected=255
i=25:   read_buf[i]=25  expected=255
i=26:   read_buf[i]=26  expected=255
i=27:   read_buf[i]=27  expected=255
i=28:   read_buf[i]=28  expected=255
i=29:   read_buf[i]=29  expected=255
i=30:   read_buf[i]=30  expected=255
i=31:   read_buf[i]=31  expected=255
i=32:   read_buf[i]=32  expected=255
i=33:   read_buf[i]=33  expected=255
i=34:   read_buf[i]=34  expected=255
i=35:   read_buf[i]=35  expected=255
i=36:   read_buf[i]=36  expected=255
i=37:   read_buf[i]=37  expected=255
i=38:   read_buf[i]=38  expected=255
i=39:   read_buf[i]=39  expected=255
i=40:   read_buf[i]=40  expected=255
i=41:   read_buf[i]=41  expected=255
i=42:   read_buf[i]=42  expected=255
i=43:   read_buf[i]=43  expected=255
i=44:   read_buf[i]=44  expected=255
i=45:   read_buf[i]=45  expected=255
i=46:   read_buf[i]=46  expected=255
i=47:   read_buf[i]=47  expected=255
i=48:   read_buf[i]=48  expected=255
[... 562 lines cut ...]
 - PASS - [flash_driver.test_flash_page_layout] duration = 0.071 seconds
 - SKIP - [flash_driver.test_get_size] duration = 0.004 seconds
 - PASS - [flash_driver.test_read_unaligned_address] duration = 0.070 seconds

------ TESTSUITE SUMMARY END ------

===================================================================
PROJECT EXECUTION SUCCESSFUL
```

**Impact**
All samples and tests for frdm_mcxn947 writing to the storage partition, located at the w25q64jvssiq flash, are failed.

**Environment:**

 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 -  SHA-1: 8d6ab28ff772a1cb10b53b1db1902d9716edaa13

**Additional context**

**The issue is caused by the following commit**: 7ed7cd191afbfa00b9eb6ad2f40b372170260e7e
Author: Zhaoxiang Jin <Zhaoxiang.Jin_1@nxp.com>
Date: 3/26/2025
Message:  **modules: hal_nxp: Move hal_nxp glue layer to zephyr repo**

ZhaoxiangJin, 2025-04-16:
fixed in hal_nxp side. https://github.com/zephyrproject-rtos/hal_nxp/pull/527

ZhaoxiangJin, 2025-04-16:
Hello @hakehuang, this run failure is caused by hal_nxp update, could you check why we didn't detect this issue during our test cycle? Thanks.

hakehuang, 2025-04-16:
> Hello [@hakehuang](https://github.com/hakehuang), this run failure is caused by hal_nxp update, could you check why we didn't detect this issue during our test cycle? Thanks.

@ZhaoxiangJin this has been reported in https://github.com/zephyrproject-rtos/zephyr/issues/88080 , for hal upgrading, I am sorry, I did not catch this for this platform.

dleach02, 2025-04-23:
Duplicate of #88080

closed 2025-04-23 by dleach02