zephyrproject-rtos/zephyr #88660: drivers: Peripheral to peripheral interconnect API

carlescufi, 2025-04-15:
**Is your enhancement proposal related to a problem? Please describe.**

This issue describes the (far-fetched at this point) possibility of having a common driver API for peripheral to peripheral interconnect functionality.

Today there are many microcontrollers that offer some kind of mechanism to achieve the following:

- Direct peripheral to peripheral connection
- No CPU intervention
- Ability to trigger actions on one peripheral based on the output of another one
- Configurable at runtime

**Describe the solution you'd like**

Ideally we would need a new driver API class that encapsulates this functionality in a vendor-agnostic fashion. Whether this is actually possible at all remains to be seen, given the ample differences between vendors.

**Describe alternatives you've considered**

The simplest alternative is to have vendor-specific APIs for each vendor individually. This may end up being the case if we do not manage to find enough commonality among the different hardware.

aescolar, 2025-04-15:
My 2 cents:
Apart from something like this being probably difficult to abstract without lots of complexity,
I wonder what would be the benefit: By definition connecting HW peripherals together is going to be really HW specific and be done either in drivers for some very specific set of HW IPs, or in an APP that is relying on very specific HW. So what would the benefit be of providing an abstraction over a HAL/vendor driver for this?

carlescufi, 2025-04-15:
> My 2 cents: Apart from something like this being probably difficult to abstract without lots of complexity, I wonder what would be the benefit: By definition connecting HW peripherals together is [...]

It's a very good point @aescolar. In fact the purpose of this issue is to try and find out whether it makes sense at all to abstract this type of functionality (à la pinctrl) or not at all. So your input is valuable because it gives us one point of view already. At the very least though, we should define a driver "folder" (i.e. not driver class but folder) where to place all of these driver implementations.

henrikbrixandersen, 2025-04-15:
This is sort of like DMA, but even less generic across platforms. Giving the complexity of the DMA API and considering that this will likely be even more complex giving that we're not just talking data transfers, but also potentially signal routing and events, I too think this is too broad to benefit from a genetic API.

I agree, however, that such implementations could live in the same folder, but this folder could also just be drivers/misc/ for now?

teburd, 2025-04-15:
Right, there's just not a huge benefit to building an abstract API like this.

That said, upleveling functionality to say something like spi/dma/i2c/adc/dac/sensors kind of makes sense maybe? I had it in mind when creating rtio.

E.g.

```c
rtio_prep_await_trigger(sqe, /* Some hardware specific trigger sourcing here */)
rtio_prep_write(...)
rtio_prep_read(...)
rtio_submit(...)
```

Now if you were to do some scan of this chain of operations you could setup the hardware to say... await a timer event to trigger of bus actions to read a sensor or adc or whatever perhaps.

Going a step further for sensors in particular, where right now what we do is in each driver setup a driver specific GPIO callback which is.. well it is what it is..

but then yeah, doing dataready/fifo reads could *maybe* be done with something like NXP's XBAR or ST's interconnect stuff? Clearly other vendors have similar features.

But doing this API at the hardware level in some abstract manner? Yeah I agree with @henrikbrixandersen in that regard, DMA does this and it has more or less become a union of all functionality. The only time its abstract across hardware today is whenever SOF is involved because SOF wants all DMAs to operate in a particular manner so you get things like pseudo DMA drivers https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/dma/dma_nxp_sof_host_dma.c

And trying to add in software shims for what might be a union of DMA features doesn't probably make much sense at this sort of device scale imho.

decsny, 2025-05-27:
I think there is value in a generic hardware abstraction to control the wide varieties of muxing hardware. I don't think that it needs to be expected to solve complex system and software use cases on it's own. That's for other abstraction layers and components to handle, which can make use of a generic mux API.

FelixWang47831, 2026-01-22:
Recently, while working on an encoder driver for NXP devices on Zephyr, I noticed that for configuring encoder input signals, some devices use XBAR, while others use inputmux. These are two different peripherals, but both are capable of configuring signal routing. I think it would be best to have a unified interconnect API that can select and operate on the appropriate peripheral based on the information provided in the devicetree.

And I also noticed at samples/boards/nordic/nrfx/src/main.c,
the existing Nordic nrfx sample already shows application‑level use of interconnect hardware.
This is a very simple application—when a button is pressed, an LED pin is toggled.

However, for more complex applications, such as motor control:
1. PWM needs to be synchronized with ADC sampling
2.  Comparator output needs to be used as a PWM fault input to provide overcurrent protection. 

These specific logic functions are implemented through interconnect hardware and do not require CPU involvement, which greatly reduces latency and also lowers power consumption. If this logic is written directly into drivers, it will make the drivers messy. If it is configured at the application layer without a common interface, then applications cannot be adapted to different devices.

For me, the greatest value of Zephyr is that the same application code can run across different SoCs. Without support from this kind of driver, the application is no longer really a “Zephyr application,” but rather a “vendor HAL application wrapped with Zephyr.”

For my current needs, all I really need is a way to associate an input signal with an output signal — or put differently, to set up an event producer and an event consumer. I don't expected a complicated “one‑size‑fits‑all” generic driver that tries to cover every vendor’s interconnect features
As for defining additional APIs beyond that, but still staying within a just‑right support boundary, that’s something need to be explored a bit more.

stemschmidt, 2026-09-18:
**A proposal for a peripheral interconnect AP as a starting point for discussionsI:**

All peripheral interconnect mechanisms I came across are basically publish-subscribe interconnections in hardware.

In order to encapsulate the vendor specifics, opaque handles and a mechanism to get those handles via device
tree aliases have to be used.

A client can then get references to peripherals and use them as events or listener. The client can then combine
them in a handle and request a connection from the peripheral interconnect driver on a interconnect lane.

If e.g. I2C or SPI buses are used as listeners the peripheral interconnect driver has to ensure that they are
exclusively used by the peripheral interconnect driver. There are no synchronization mechanisms on the hardware
level.

```c
/* Opaque handle identifying an event for a peripheral interconnect lane */
typedef void * peripheral_interconnect_event_t;
/* Opaque handle identifying a listener for a peripheral interconnect lane */
typedef void * peripheral_interconnect_listener_t;

/* Structure assigning one or several listeners to an event in a peripheral interconnect lane */
struct peripheral_interconnect_handle {
    peripheral_interconnect_event_t event;
    uint32_t num_listener;
    peripipheral_interconnect_listener_t *listeners;
};

/* Get a peripheral_interconnect_event_t from a DT alias (to make it vendor independent) */
peripheral_interconnect_event_t * dt_get_peripheral_interconnect_event(DT_ALIAS_MAGIC());

/* Get a peripheral_interconnect_listener_t from a DT alias (to make it vendor independent) */
peripheral_interconnect_listener_t * dt_get_peripheral_interconnect_listener(DT_ALIAS_MAGIC());

/* Allocate and configure a peripheral interconnect lane with the event and listeners given in handle.
   The peripheral interconnect driver verifies if 
   - a peripheral interconnect lane is available
   - the events and listeners can be assigned to the same peripheral interconnect lane
   When all conditions are fulfilled the event and the listeners are registered and the lane is enabled.
 */
int peripheral_interconnect_set_handle(peripherial_interconnect_handle * handle);

/* Disables the peripheral interconnect lane and removes events and listeners from it. Deallocates/releases
 * the peripheral interconnect lane.
 */
int peripheral_interconnect_release_handle(peripherial_interconnect_handle * handle);
```