zephyrproject-rtos/zephyr #88665: net_if: misaligned access

clamattia, 2025-04-15:
**Describe the bug**
UB sanitizers finds misaligned reads/writes of `struct in_addr` when running in `native_sim`.

**To Reproduce**
1. Build for native_sim with UB-sanitizer enabled
2. Run the sim
3. Network Ping
4. ->UB sanitizer finds `misaligned accesses`

**Expected behavior**
UB sanitizer should not find `misaligned access` under those circumstances.

**Impact**
- Either those are a real problem
- Or they are false positives, and the UB sanitizer can not be used to test an application in above circumstances to find actual problems

**Logs and console output**
```
zephyr/subsys/net/ip/net_if.c:3878:8: runtime error: member access within misaligned address 0xXXXXXXXX for type 'const struct in_addr', which requires 4 byte alignment
zephyr/subsys/net/ip/net_if.c:3583:23: runtime error: member access within misaligned address 0xXXXXXXXX for type 'const struct in_addr', which requires 4 byte alignment
zephyr/include/zephyr/net/net_ip.h:926:9: runtime error: member access within misaligned address 0xXXXXXXXX for type 'const struct in_addr', which requires 4 byte alignment
zephyr/include/zephyr/net/net_ip.h:813:22: runtime error: member access within misaligned address 0xXXXXXXXX for type 'struct in_addr', which requires 4 byte alignment
zephyr/include/zephyr/net/net_ip.h:837:10: runtime error: member access within misaligned address 0xXXXXXXXX for type 'const struct in_addr', which requires 4 byte alignment
zephyr/include/zephyr/net/net_ip.h:926:9: runtime error: member access within misaligned address 0xXXXXXXXX for type 'const struct in_addr', which requires 4 byte alignment
zephyr/include/zephyr/net/net_ip.h:825:9: runtime error: member access within misaligned address 0xXXXXXXXX for type 'const struct in_addr', which requires 4 byte alignment
zephyr/include/zephyr/net/net_ip.h:837:10: runtime error: member access within misaligned address 0xXXXXXXXX for type 'const struct in_addr', which requires 4 byte alignment
```

**Environment (please complete the following information):**
TBD

**Additional context**
TBD

tpambor, 2025-04-15:
These are false-positive, as `UNALIGNED_GET` can be optimized away on architectures that allow unaligned memory access. UBSan does not see that we are aware of the potential unaligned access by using the `UNALIGNED_GET` macro. It only sees that we want to dereference a struct element with unaligned access. To solve this, UBSan alignment checks should be suppressed for accesses with the `UNALIGNED_GET` macro by using `__attribute__((no_sanitize("alignment")))`.

jukkar, 2025-04-15:
There was issues before we added `UNALIGNED_GET()` calls, But these were fixed many years go.

tpambor, 2025-04-15:
A reproducing example can be found here: https://godbolt.org/z/nvf9dGGYM

Cortex-M0 does not support unaligned memory accesses. The function `test_no_unaligned_get` demonstrates that without `UNALIGNED_GET` the compiler generates a normal 32-bit memory access which would cause a fault for unaligned addresses.

In comparison, `test_unaligned_get_broken` correctly generates multiple 8-bit accesses to avoid unaligned memory accesses, but UBSan would trap if the passed address is unaligned, although the code would execute correctly.

`test_unaligned_get_safe` is marked with `__attribute__((no_sanitize("alignment")))` and UBSan does not generate traps here for unaligned accesses.

jukkar, 2025-04-17:
Closing as there is no issue here.

clamattia, 2025-04-17:
> Closing as there is no issue here.

Agree, that it is a false triggering of the UBsan. But it would be helpful to avoid the false positive. Would it be ok to add this `__attribute__((no_sanitize("alignment")))` to the respective function?

jukkar, 2025-04-17:
Perhaps the unaligned macro could be modified to have the extra attribute?

clamattia, 2025-04-17:
> Perhaps the unaligned macro could be modified to have the extra attribute?

That would be great.
I can try if it would be fixed.

closed 2025-04-19 by jukkar